OE4T/meta-tegra #126: Statx syscall causes kernel oops on TX2

w0land, 2019-04-09:
Hi, 

I'm having problems with TX2 and Qt5; Qt internal filesystem engine is making a call to **statx** syscall which leads to kernel oops: 
```
Apr 09 08:05:28 jetson-tx2 kernel: Qt5_CinematicEx[8481]: syscall 291
Apr 09 08:05:28 jetson-tx2 kernel: Code: d2802468 a90153f3 aa0403f3 d4000001 (b140041f) 
Apr 09 08:05:28 jetson-tx2 kernel: CPU: 1 PID: 8481 Comm: Qt5_CinematicEx Not tainted 4.4.38-l4t-r28.2+g2657cd8 #1
Apr 09 08:05:28 jetson-tx2 kernel: Hardware name: quill (DT)
Apr 09 08:05:28 jetson-tx2 kernel: task: ffffffc07582b200 ti: ffffffc1e1420000 task.ti: ffffffc1e1420000
Apr 09 08:05:28 jetson-tx2 kernel: PC is at 0x7fb6ce7b20
Apr 09 08:05:28 jetson-tx2 kernel: LR is at 0x7fb7170b00
Apr 09 08:05:28 jetson-tx2 kernel: pc : [<0000007fb6ce7b20>] lr : [<0000007fb7170b00>] pstate: 00000000
Apr 09 08:05:28 jetson-tx2 kernel: sp : 0000007fffff9c50
Apr 09 08:05:28 jetson-tx2 kernel: x29: 0000007fffff9c50 x28: 0000007fffffab88 
Apr 09 08:05:28 jetson-tx2 kernel: x27: 0000000000000000 x26: 0000000065584966 
Apr 09 08:05:28 jetson-tx2 kernel: x25: 000000555b495a40 x24: 0000007fb7ffe718 
Apr 09 08:05:28 jetson-tx2 kernel: x23: 000000555b47dbe0 x22: 000000000000001f 
Apr 09 08:05:28 jetson-tx2 kernel: x21: 0000007fffff9db8 x20: 0000007fb7557000 
Apr 09 08:05:28 jetson-tx2 kernel: x19: 0000007fffff9db8 x18: 0000000000007cfe 
Apr 09 08:05:28 jetson-tx2 kernel: x17: 0000007fb6ce7af0 x16: 0000007fb7555420 
Apr 09 08:05:28 jetson-tx2 kernel: x15: 000000000000b3f6 x14: 0000000000000000 
Apr 09 08:05:28 jetson-tx2 kernel: x13: 00ffffff00ffffff x12: 0000000000000001 
Apr 09 08:05:28 jetson-tx2 kernel: x11: 000000005cac44fe x10: 000000000001f52c 
Apr 09 08:05:28 jetson-tx2 kernel: x9 : 00000000000007ff x8 : 0000000000000123 
Apr 09 08:05:28 jetson-tx2 kernel: x7 : 0000007fb73c7130 x6 : 000000000000001f 
Apr 09 08:05:28 jetson-tx2 kernel: x5 : 0000000000001000 x4 : 0000007fffff9db8 
Apr 09 08:05:28 jetson-tx2 kernel: x3 : 0000000000000fff x2 : 0000000000001000 
Apr 09 08:05:28 jetson-tx2 kernel: x1 : 0000007fb73c7130 x0 : 000000000000001f 
```

Board is TX2 with kernel: 
```
root@jetson-tx2:~# uname -a
Linux jetson-tx2 4.4.38-l4t-r28.2+g2657cd8 #1 SMP PREEMPT Sun Mar 31 08:28:25 UTC 2019 aarch64 aarch64 aarch64 GNU/Linux
```

My layers are: 
```
meta
meta-poky
meta-yocto-bsp       = "thud:faeb366bc3eb322f5f203cfe08dc4cf529a822e9"
meta-qt5             = "5.12:e413185212c68e0c3d895c0d7aff508ef297f848"
meta-tegra           = "thud:da90b42297f5d14d362015be1c5a888f5cac946a"
meta-oe
meta-python          = "thud:4cd3a39f22a2712bfa8fc657d09fe2c7765a4005"
````

What helps is disabling **statx** support from Qt, but my question is: 
Is this a know issue with Tegra Linux kernel? I've tried googling but nothing comes up. 

Thanks, 
B

madisongh, 2019-04-09:
Any syscall that isn't supported by the kernel (at least on arm64) will log a message about it.  It's not an "oops" per se, since the kernel itself isn't in trouble; it's just letting you know that some userland code tried to use a syscall that isn't there. I've seen that a lot with systemd... newer versions tend to be built on the latest-and-greatest upstream kernel, so those of us stuck on older kernel versions have to to put up with a slew of such traces when it starts.  What it does when it wants to use a brand-new syscall is to try it, then fall back to another implementation of whatever it's trying to do if it gets the ENOSYS error back. I've sometimes patched the code to just use the fallback path and skip the syscall check, just to keep my message logs clean.

If I'm reading the kernel sources right, you could also try using `sysctl debug.exception-trace=0` to disable the messages, but that might turn off other messages that could be useful, too.

w0land, 2019-04-09:
@madisongh Yeah I've just found out that **statx** was introduced in _4.11._ so naturally it's not present in 4.4.38. 
So I guess the problem is that glibc actually implements this on systems with older kernels.
Do you think a glibc_%.bbappend is needed here?

madisongh, 2019-04-10:
First check to see if Qt is calling glibc's statx or whether it is directly making the syscall. OLDEST_KERNEL should already be set such that if glibc can emulate statx on older kernels, in which case patching Qt to avoid the direct syscall and just use the function exported by glibc should work.

closed 2019-09-01 by madisongh