OE4T/meta-tegra #127: recipe for target 'drivers' failed

MartinBagge, 2019-05-01:
Hi,

I'm trying to build an image for a Jetson Xavier.
At some point in the build process i get this error:
build/tmp-glibc/work-shared/jetson-xavier/kernel-source/Makefile:1102: recipe for target 'drivers' failed
Do I need to download something to get the drivers?
Or set some variable?
I have already set the NVIDIA-DEVNET-MIRROR and other variables specified.

madisongh, 2019-05-02:
Which gcc version are you using? I'll bet there are some issues compiling NVIDIA's kernel with gcc 8, and I've been using gcc 7 so far in my testing.

MartinBagge, 2019-05-03:
I just tried to create docker a container for the building and made sure that the environment had gcc 7.4 only and output of gcc --version is also 7.4
I tried specifying the GCCVERSION in local.conf but as I think you mention on the wiki the package containing gcc 7 can't be found.

I am still getitng the same drivers error. I have attached the log file for further information on it if it helps.

[log.do_compile.10447.txt](https://github.com/madisongh/meta-tegra/files/3141792/log.do_compile.10447.txt)

Thanks for spending the time to help out :)

commit 104e9cff (2019-05-04): linux-tegra: bump SRCREV

to pick up the following changes to make the kernel buildable
with gcc 8:

5563471b8615cc77f60cfc9bdefc985edfaf34e1 kernel: fix gcc 8 compiler warnings
c069635f63ff30ed98645d9d9a8102b536ee3f6a Kbuild: suppress packed-not-aligned warning for default setting only
5c32b385c35226c4b28e952d9fd3dd4626ab350f disable stringop truncation warnings for now
87633eaaf26d38375eef77058b656db91ac7356e Compiler Attributes: enable -Wstringop-truncation on W=1 (gcc >= 8)
a2c4e01c480c28ff2af37e41b50804b6c7938606 sched: fix format specifiers for util_avg_walt
0b8faa2e40d53e0e0c850041bf17c [... cut]
commit f2ed64d8 (2019-05-05): linux-tegra: bump SRCREV

to pick up the following changes to make the kernel buildable
with gcc 8:

5563471b8615cc77f60cfc9bdefc985edfaf34e1 kernel: fix gcc 8 compiler warnings
c069635f63ff30ed98645d9d9a8102b536ee3f6a Kbuild: suppress packed-not-aligned warning for default setting only
5c32b385c35226c4b28e952d9fd3dd4626ab350f disable stringop truncation warnings for now
87633eaaf26d38375eef77058b656db91ac7356e Compiler Attributes: enable -Wstringop-truncation on W=1 (gcc >= 8)
a2c4e01c480c28ff2af37e41b50804b6c7938606 sched: fix format specifiers for util_avg_walt
0b8faa2e40d53e0e0c850041bf17c [... cut]
closed 2019-05-04 by madisongh