OE4T/meta-tegra #132: Jetson Nano - warrior - core-image-sato - Postinstall scriptlets of ['tegra-libraries-argus'] have failed.

brianleePTL, 2019-05-24:
Hi All,

I get a build error when trying to build core-image-sato for the Jetson Nano board on the warrior branch. Setup and error is as follows. Note that core-image-minimal does succeed.

Is this a genuine bug or something I am doing wrong?

Thanks!

```
mkdir ~/yocto/ && cd ~/yocto/
git clone git://git.yoctoproject.org/poky -b warrior ~/yocto/sources/poky
git clone https://github.com/madisongh/meta-tegra -b warrior ~/yocto/sources/meta-tegra
. sources/poky/oe-init-build-env build
```

bblayers.conf:
```
# POKY_BBLAYERS_CONF_VERSION is increased each time build/conf/bblayers.conf
# changes incompatibly
POKY_BBLAYERS_CONF_VERSION = "2"

BBPATH = "${TOPDIR}"
BBFILES ?= ""

BBLAYERS ?= " \
  ${TOPDIR}/../sources/poky/meta \
  ${TOPDIR}/../sources/poky/meta-poky \
  ${TOPDIR}/../sources/poky/meta-yocto-bsp \
  ${TOPDIR}/../sources/meta-tegra \
  "
```

```
MACHINE=jetson-nano bitbake core-image-sato
```

```
Loading cache: 100% |###################################################################################################################################| Time: 0:00:00
Loaded 1393 entries from dependency cache.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.42.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-poky-linux"
MACHINE              = "jetson-nano"
DISTRO               = "poky"
DISTRO_VERSION       = "2.7"
TUNE_FEATURES        = "aarch64 armv8a crc"
TARGET_FPU           = ""
meta                 
meta-poky            
meta-yocto-bsp       = "HEAD:f2b442c5c8e3925a4d3fc9693f1e47863dade2a9"
meta-tegra           = "HEAD:255deac20fe56c63a6f73cbf4d468f3796ca56c8"

Initialising tasks: 100% |##############################################################################################################################| Time: 0:00:08
Sstate summary: Wanted 1446 Found 0 Missed 1446 Current 761 (0% match, 34% complete)
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
ERROR: core-image-sato-1.0-r0 do_rootfs: Postinstall scriptlets of ['tegra-libraries-argus'] have failed. If the intention is to defer them to first boot,
then please place them into pkg_postinst_ontarget_${PN} ().
Deferring to first boot via 'exit 1' is no longer supported.
Details of the failure are in /home/brian/yocto/build/tmp/work/jetson_nano-poky-linux/core-image-sato/1.0-r0/temp/log.do_rootfs.
ERROR: core-image-sato-1.0-r0 do_rootfs: 
ERROR: core-image-sato-1.0-r0 do_rootfs: Function failed: do_rootfs
ERROR: Logfile of failure stored in: /home/brian/yocto/build/tmp/work/jetson_nano-poky-linux/core-image-sato/1.0-r0/temp/log.do_rootfs.11449
ERROR: Task (/home/brian/yocto/sources/poky/meta/recipes-sato/images/core-image-sato.bb:do_rootfs) failed with exit code '1'
NOTE: Tasks Summary: Attempted 6050 tasks of which 2434 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/brian/yocto/sources/poky/meta/recipes-sato/images/core-image-sato.bb:do_rootfs
Summary: There were 3 ERROR messages shown, returning a non-zero exit code.
```

madisongh, 2019-05-24:
Can you include an extract from the `log.do_rootfs` log file where it was running the postinstall scriptlets?

madisongh, 2019-05-24:
Never mind, I think I've figured it out.  Fix coming shortly.

madisongh, 2019-05-24:
Should be fixed with 8b921f5d45e91f03ac2f6f9f9e6543cc4e1d285f - try updating and see if that helps.

brianleePTL, 2019-05-24:
```
+ [ -n /home/brian/yocto/build/tmp/work/jetson_nano-poky-linux/core-image-sato/1.0-r0/rootfs ]
+ OPT=-f -r /home/brian/yocto/build/tmp/work/jetson_nano-poky-linux/core-image-sato/1.0-r0/rootfs
+ update-rc.d -f -r /home/brian/yocto/build/tmp/work/jetson_nano-poky-linux/core-image-sato/1.0-r0/rootfs nvargus-daemon remove
 Removing any system startup links for nvargus-daemon ...
%prein(tegra-libraries-argus-32.1.0-r0.armv8a_tegra210): waitpid(23192) rc 23192 status 0

  Installing       : tegra-libraries-argus-32.1.0-r0.armv8a_tegra210    698/740 
  Running scriptlet: tegra-libraries-argus-32.1.0-r0.armv8a_tegra210    698/740 
%post(tegra-libraries-argus-32.1.0-r0.armv8a_tegra210): scriptlet start
%post(tegra-libraries-argus-32.1.0-r0.armv8a_tegra210): execv(/bin/sh) pid 23194
+ set -e
+ true
+ type update-rc.d
+ [ -n /home/brian/yocto/build/tmp/work/jetson_nano-poky-linux/core-image-sato/1.0-r0/rootfs ]
+ OPT=-f -r /home/brian/yocto/build/tmp/work/jetson_nano-poky-linux/core-image-sato/1.0-r0/rootfs
+ update-rc.d -f -r /home/brian/yocto/build/tmp/work/jetson_nano-poky-linux/core-image-sato/1.0-r0/rootfs nvargus-daemon defaults
update-rc.d: /home/brian/yocto/build/tmp/work/jetson_nano-poky-linux/core-image-sato/1.0-r0/rootfs/etc/init.d/nvargus-daemon: file does not exist
%post(tegra-libraries-argus-32.1.0-r0.armv8a_tegra210): waitpid(23194) rc 23194 status 100
warning: %post(tegra-libraries-argus-32.1.0-r0.armv8a_tegra210) scriptlet failed, exit status 1

Error in POSTIN scriptlet in rpm package tegra-libraries-argus
  Installing       : tegra-libraries-32.1.0-r0.armv8a_tegra210          699/740 
  Running scriptlet: tegra-libraries-32.1.0-r0.armv8a_tegra210          699/740 
%post(tegra-libraries-32.1.0-r0.armv8a_tegra210): scriptlet start
%post(tegra-libraries-32.1.0-r0.armv8a_tegra210): execv(/bin/sh) pid 23196
+ set -e
+ [ x/home/brian/yocto/build/tmp/work/jetson_nano-poky-linux/core-image-sato/1.0-r0/rootfs = x ]
%post(tegra-libraries-32.1.0-r0.armv8a_tegra210): waitpid(23196) rc 23196 status 0

  Installing       : libdrm-nvdc-32.1.0-r0.armv8a_tegra210              700/740 
  Installing       : libdrm2-2.4+git0+4b909926c5-r0.armv8a_tegra210     701/740 
  Running scriptlet: libdrm2-2.4+git0+4b909926c5-r0.armv8a_tegra210     701/740 
%post(libdrm2-2.4+git0+4b909926c5-r0.armv8a_tegra210): scriptlet start
%post(libdrm2-2.4+git0+4b909926c5-r0.armv8a_tegra210): execv(/bin/sh) pid 23197
+ set -e
+ [ x/home/brian/yocto/build/tmp/work/jetson_nano-poky-linux/core-image-sato/1.0-r0/rootfs = x ]
%post(libdrm2-2.4+git0+4b909926c5-r0.armv8a_tegra210): waitpid(23197) rc 23197 status 0
```

I think this is what you're looking for

brianleePTL, 2019-05-24:
oops - I see you have maybe fixed it. I'll give it a go, thanks!

brianleePTL, 2019-05-24:
I get a successful build now. Thanks!

closed 2019-05-24 by brianleePTL