OE4T/meta-tegra #133: 'flashvars' file not found for TX2

ordellz, 2019-06-07:
Hi,
I'm trying to build warrior for TX2.
U-boot fails to build with message:

> ERR: flash variable set not defined

These vars are defined in the flashvars file of the corresponding machine (in recipes-bsp/tegra-binaries). For TX2, I think it should be found in directory 'files/tegra186' directory but this is named 'files/jetson-tx2' instead. So it doesn't find it.
Renaming 'jetson-tx2' dir to 'tegra186' in in meta-tegra fixes the problem indeed.

Is this the right fix or am I wrong about something?

Regards
Eric

madisongh, 2019-06-07:
What's your MACHINE setting?  If it's `jetson-tx2`, then that file should still be found - certainly works for my builds. The flash variables include settings that are carrier-specific (pinmux config file, etc.), so if you're building for a different MACHINE (e.g., for a custom board or a different carrier for your TX2), you should create a flashvars file that points to the correct config files for your system - typically through having a bbappend in another layer that prepends to FILESEXTRAPATHS to point to the location of your file.

ordellz, 2019-06-11:
You are right: I created my own MACHINE based on tegra186 so 'jetson-tx2' directory was not searched in.
I had even created in my own layer bbappend+FILESEXTRAPATHS_prepend with flashvars placed in files/tegra186, just to rename from jetson-tx2 to tegra186... so I just need to rename it with the name of my MACHINE.

Thanks for your fast answer and for your so good job on meta-tegra.

closed 2019-06-11 by ordellz