OE4T/meta-tegra #134: tx2 mandatory partition sizes on Jetpack 4.2

acostach, 2019-06-07:
Hey Matt I have a question which I couldn't find an answer for on devtalk yet. Maybe you have an ideea about this.

I need to shrink down mandatory partitions on sdmmc_user for tx2 on jetpack 4.2, for instance kernel and kernel_b which are like 60mb and contain a 400kb image of uboot. I've tried changing LNXSIZE  to half of the original value but uboot isn't loaded anymore.

Do you know if this is the right approach or maybe where I should replace partition sizes further?

Thank you

madisongh, 2019-06-07:
I just tried it myself - using 1MiB and 4MiB sizes, and can confirm that it doesn't work - cboot spits out errors about reading past the end of the partition when trying to load the "kernel".  

I went back to thud/L4T R28.2.1 and LNXSIZE to 4MiB, and my jetson-tx2 booted up just fine. Something must have changed with L4T R32.1, and unfortunately NV hasn't yet published the cboot source for that release, so it will be hard to debug.

The flash.sh script for both L4T releases hard-codes LNXSIZE to 64MiB, but I have adjusted the size both downwards and upwards (for having cboot boot Linux directly without U-Boot) in the past for a custom board, and never had any problems with it.  That was with R28.2 and R28.2.1, though.

acostach, 2019-06-08:
I see.. Thank you very much for looking into this and sharing your feedback!

closed 2019-06-08 by acostach

acostach, 2019-06-11:
In case you ever need this feature at some point, it's a bug in the bootloader and will be fixed in the next l4t release. Thanks again for your time!