OE4T/meta-tegra #136: jetson nano mesa issue configure: error: unrecognized option: `-Dglvnd=true'

bsdwatch, 2019-06-11:
WARNING: /home/dingo/yocto/Yocto/meta-tegra/recipes-graphics/wayland/egl-wayland_1.1.3.bb: Unable to get checksum for egl-wayland SRC_URI entry nvidia_wayland.json: file could not be found
Parsing recipes: 100% |#########################################################################################################################| Time: 0:00:39
Parsing of 2449 .bb files complete (0 cached, 2449 parsed). 3578 targets, 153 skipped, 0 masked, 0 errors.
Removing 1 recipes from the x86_64 sysroot: 100% |##############################################################################################| Time: 0:00:01
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.42.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-poky-linux"
MACHINE              = "jetson-nano"
DISTRO               = "poky"
DISTRO_VERSION       = "2.7"
TUNE_FEATURES        = "aarch64 armv8a crc"
TARGET_FPU           = ""
meta                 
meta-poky            
meta-yocto-bsp       = "warrior:254bef1d42174b8fd7d5c9ca6baa97a5ec7533be"
meta-oe              
meta-multimedia      
meta-networking      
meta-perl            
meta-python          = "warrior:6fa72d587aab8e9d56d67a2552eb9cfbe25c86d2"
meta-qt5             = "warrior:2a5f9415e48a38f6f4eb0cf375737a2d8438e2fe"
meta-tegra           = "master:e85721894e369f2eef092beecf72274fad997fdd"

Initialising tasks: 100% |######################################################################################################################| Time: 0:00:04
Sstate summary: Wanted 1550 Found 0 Missed 1550 Current 413 (0% match, 21% complete)
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
ERROR: mesa-2_19.0.1-r0 do_configure: configure failed
ERROR: mesa-2_19.0.1-r0 do_configure: Function failed: do_configure (log file is located at /home/dingo/yocto/Yocto/build/tmp/work/armv8a_tegra210-poky-linux/mesa/2_19.0.1-r0/temp/log.do_configure.85619)
ERROR: Logfile of failure stored in: /home/dingo/yocto/Yocto/build/tmp/work/armv8a_tegra210-poky-linux/mesa/2_19.0.1-r0/temp/log.do_configure.85619
Log data follows:
| DEBUG: SITE files ['endian-little', 'bit-64', 'arm-common', 'arm-64', 'common-linux', 'common-glibc', 'aarch64-linux', 'common']
| DEBUG: Executing shell function autotools_preconfigure
| DEBUG: Shell function autotools_preconfigure finished
| DEBUG: Executing python function autotools_aclocals
| DEBUG: SITE files ['endian-little', 'bit-64', 'arm-common', 'arm-64', 'common-linux', 'common-glibc', 'aarch64-linux', 'common']
| DEBUG: Python function autotools_aclocals finished
| DEBUG: Executing shell function do_configure
| automake (GNU automake) 1.16.1
| Copyright (C) 2018 Free Software Foundation, Inc.
| License GPLv2+: GNU GPL version 2 or later <https://gnu.org/licenses/gpl-2.0.html>
| This is free software: you are free to change and redistribute it.
| There is NO WARRANTY, to the extent permitted by law.
| 
| Written by Tom Tromey <tromey@redhat.com>
|        and Alexandre Duret-Lutz <adl@gnu.org>.
| AUTOV is 1.16
| NOTE: Executing ACLOCAL="aclocal --system-acdir=/home/dingo/yocto/Yocto/build/tmp/work/armv8a_tegra210-poky-linux/mesa/2_19.0.1-r0/recipe-sysroot/usr/share/aclocal/ --automake-acdir=/home/dingo/yocto/Yocto/build/tmp/work/armv8a_tegra210-poky-linux/mesa/2_19.0.1-r0/recipe-sysroot-native/usr/share/aclocal-1.16" autoreconf -Wcross --verbose --install --force --exclude=autopoint -I /home/dingo/yocto/Yocto/build/tmp/work/armv8a_tegra210-poky-linux/mesa/2_19.0.1-r0/mesa-19.0.1/m4/ -I /home/dingo/yocto/Yocto/build/tmp/work/armv8a_tegra210-poky-linux/mesa/2_19.0.1-r0/recipe-sysroot-native/usr/share/aclocal/
| autoreconf: Entering directory `.'
| autoreconf: configure.ac: not using Gettext
| autoreconf: running: aclocal --system-acdir=/home/dingo/yocto/Yocto/build/tmp/work/armv8a_tegra210-poky-linux/mesa/2_19.0.1-r0/recipe-sysroot/usr/share/aclocal/ --automake-acdir=/home/dingo/yocto/Yocto/build/tmp/work/armv8a_tegra210-poky-linux/mesa/2_19.0.1-r0/recipe-sysroot-native/usr/share/aclocal-1.16 -I /home/dingo/yocto/Yocto/build/tmp/work/armv8a_tegra210-poky-linux/mesa/2_19.0.1-r0/mesa-19.0.1/m4/ -I /home/dingo/yocto/Yocto/build/tmp/work/armv8a_tegra210-poky-linux/mesa/2_19.0.1-r0/recipe-sysroot-native/usr/share/aclocal/ --force -I m4
| autoreconf: configure.ac: tracing
| autoreconf: running: libtoolize --copy --force
| libtoolize: putting auxiliary files in AC_CONFIG_AUX_DIR, 'bin'.
| libtoolize: copying file 'bin/ltmain.sh'
| libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
| libtoolize: copying file 'm4/libtool.m4'
| libtoolize: copying file 'm4/ltoptions.m4'
| libtoolize: copying file 'm4/ltsugar.m4'
| libtoolize: copying file 'm4/ltversion.m4'
| libtoolize: copying file 'm4/lt~obsolete.m4'
| autoreconf: running: /home/dingo/yocto/Yocto/build/tmp/work/armv8a_tegra210-poky-linux/mesa/2_19.0.1-r0/recipe-sysroot-native/usr/bin/autoconf --include=/home/dingo/yocto/Yocto/build/tmp/work/armv8a_tegra210-poky-linux/mesa/2_19.0.1-r0/mesa-19.0.1/m4/ --include=/home/dingo/yocto/Yocto/build/tmp/work/armv8a_tegra210-poky-linux/mesa/2_19.0.1-r0/recipe-sysroot-native/usr/share/aclocal/ --force
| autoreconf: configure.ac: not using Autoheader
| autoreconf: running: automake --add-missing --copy --force-missing
| configure.ac:74: installing 'bin/compile'
| configure.ac:46: installing 'bin/missing'
| src/Makefile.am: installing 'bin/depcomp'
| autoreconf: running: gnu-configize
| autoreconf: Leaving directory `.'
| NOTE: Running ../mesa-19.0.1/configure  --build=x86_64-linux 		  --host=aarch64-poky-linux 		  --target=aarch64-poky-linux 		  --prefix=/usr 		  --exec_prefix=/usr 		  --bindir=/usr/bin 		  --sbindir=/usr/sbin 		  --libexecdir=/usr/libexec 		  --datadir=/usr/share 		  --sysconfdir=/etc 		  --sharedstatedir=/com 		  --localstatedir=/var 		  --libdir=/usr/lib 	  --includedir=/usr/include 		  --oldincludedir=/usr/include 		  --infodir=/usr/share/info 		  --mandir=/usr/share/man 		  --disable-silent-rules 		  --disable-dependency-tracking 		  --with-libtool-sysroot=/home/dingo/yocto/Yocto/build/tmp/work/armv8a_tegra210-poky-linux/mesa/2_19.0.1-r0/recipe-sysroot --enable-shared-glapi                 --disable-opencl                 --enable-glx-read-only-text                 PYTHON2=python2                 --with-llvm-prefix=/home/dingo/yocto/Yocto/build/tmp/work/armv8a_tegra210-poky-linux/mesa/2_19.0.1-r0/recipe-sysroot/usr/lib/llvm8.0.0                 --with-platforms='wayland x11                drm                surfaceless'                 --enable-autotools  --disable-static  --enable-dri --with-dri-drivers= --enable-dri3 --enable-egl --with-gallium-drivers=swrast,virgl  --disable-llvm --enable-gbm --enable-gles1 --enable-gles2 -Dglvnd=true --enable-opengl --disable-gallium-osmesa --disable-libunwind --with-vulkan-drivers= --enable-glx-tls  --disable-xa --disable-xvmc --enable-nls
| configure: error: unrecognized option: `-Dglvnd=true'
| Try `../mesa-19.0.1/configure --help' for more information
| NOTE: The following config.log files may provide further information.
| NOTE:
| ERROR: configure failed
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_configure (log file is located at /home/dingo/yocto/Yocto/build/tmp/work/armv8a_tegra210-poky-linux/mesa/2_19.0.1-r0/temp/log.do_configure.85619)
ERROR: Task (/home/dingo/yocto/Yocto/poky-thud/meta/recipes-graphics/mesa/mesa_19.0.1.bb:do_configure) failed with exit code '1'
NOTE: Tasks Summary: Attempted 1554 tasks of which 1521 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/dingo/yocto/Yocto/poky-thud/meta/recipes-graphics/mesa/mesa_19.0.1.bb:do_configure
Summary: There was 1 WARNING message shown.
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
dingo@debian:~/yocto/Yocto/build$

madisongh, 2019-06-11:
Looks like you've got meta-tegra on its master branch, but are using warrior for the other layers.  Try checking out warrior in meta-tegra, that should work better.

bsdwatch, 2019-06-11:
Yupp - fixed

closed 2019-06-11 by bsdwatch