OE4T/meta-tegra #137: core-image-weston - [14:23:43.461] no drm device found

bsdwatch, 2019-06-11:
So i got an core-image-weston built however weston-start fails... any hints ?
[14:23:43.458] weston 5.0.0
               https://wayland.freedesktop.org
               Bug reports to: https://gitlab.freedesktop.org/wayland/weston/issues/
               Build: unknown (not built from git or tarball)
[14:23:43.458] Command line: /usr/bin/weston --modules=xwayland.so --log=/var/log/weston.log
[14:23:43.458] OS: Linux, 4.9.140-l4t-r32.1+g3c02a65d9170, #1 SMP PREEMPT Tue Jun 11 12:03:09 UTC 2019, aarch64
[14:23:43.458] Starting with no config file.
[14:23:43.458] Output repaint window is 7 ms maximum.
[14:23:43.458] Loading module '/usr/lib/libweston-5/drm-backend.so'
[14:23:43.460] initializing drm backend
[14:23:43.461] no drm device found
[14:23:43.461] fatal: failed to create compositor backend

madisongh, 2019-06-12:
Looks like the `--use-egldevice` option is missing.  Here's the command line from my run:
```
Command line: /usr/bin/weston --use-egldevice --modules=xwayland.so --log=/var/log/weston.log
```
That option gets set via the OPTARGS variable in `/etc/default/weston`, which should have been installed  through the bbappend in meta-tegra for the weston recipe.

bsdwatch, 2019-06-12:
Yupp cleaner the whole environment and rebuilt boots to the weston UI, I can also confirm core-image-x11 works and boots to ui also... on to cameras and streaming

closed 2019-06-12 by bsdwatch