OE4T/meta-tegra #138: CUDA Compiler unable to compile a simple test program

reyapo, 2019-06-14:
Hi,

I'm trying to add a CUDA project to our yocto build but have some trouble getting CMAKE to correctly detect CUDA. My recipe is the following:
```
inherit cmake cuda

SRCREV = "${AUTOREV}"
SRC_URI = "git://uri/to/git;protocol=ssh;"
S = "${WORKDIR}/git"

export CUDACXX="${STAGING_DIR_NATIVE}/usr/local/cuda-${CUDA_VERSION}/bin/nvcc"
export CUDAHOSTCXX="${STAGING_DIR_NATIVE}/usr/bin/aarch64-poky-linux/aarch64-poky-linux-g++"

DEPENDS = " gstreamer1.0 gstreamer1.0-plugins-base boost tbb "
```

CMAKE fails on enabling the language:

```
project(${ROOT_NAME} VERSION ${ROOT_VERSION} LANGUAGES CXX C CUDA)`
```
```
Log data follows:
| DEBUG: Executing python function externalsrc_configure_prefunc
| DEBUG: Python function externalsrc_configure_prefunc finished
| DEBUG: Executing shell function do_configure
| -- The CXX compiler identification is GNU 7.2.1
| -- The C compiler identification is GNU 7.2.1
| -- The CUDA compiler identification is unknown
| -- Check for working CXX compiler: /yocto/tmp/work/armv8a_tegra194-poky-linux/vr360/git-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-g++
| -- Check for working CXX compiler: /yocto/tmp/work/armv8a_tegra194-poky-linux/vr360/git-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-g++ -- works
| -- Detecting CXX compiler ABI info
| -- Detecting CXX compiler ABI info - done
| -- Detecting CXX compile features
| -- Detecting CXX compile features - done
| -- Check for working C compiler: /yocto/tmp/work/armv8a_tegra194-poky-linux/vr360/git-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-gcc
| -- Check for working C compiler: /yocto/tmp/work/armv8a_tegra194-poky-linux/vr360/git-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-gcc -- works
| -- Detecting C compiler ABI info
| -- Detecting C compiler ABI info - done
| -- Detecting C compile features
| -- Detecting C compile features - done
| -- Check for working CUDA compiler: /yocto/tmp/work/armv8a_tegra194-poky-linux/vr360/git-r0/recipe-sysroot-native/usr/local/cuda-10.0/bin/nvcc
| -- Check for working CUDA compiler: /yocto/tmp/work/armv8a_tegra194-poky-linux/vr360/git-r0/recipe-sysroot-native/usr/local/cuda-10.0/bin/nvcc -- broken
| CMake Error at /yocto/tmp/work/armv8a_tegra194-poky-linux/vr360/git-r0/recipe-sysroot-native/usr/share/cmake-3.12/Modules/CMakeTestCUDACompiler.cmake:46 (message):
|   The CUDA compiler
|
|     "/yocto/tmp/work/armv8a_tegra194-poky-linux/vr360/git-r0/recipe-sysroot-native/usr/local/cuda-10.0/bin/nvcc"
|
|   is not able to compile a simple test program.
|
|   It fails with the following output:
|
|     Change Dir: /yocto/tmp/work/armv8a_tegra194-poky-linux/vr360/git-r0/vr360-git/CMakeFiles/CMakeTmp
|
|     Run Build Command:"ninja" "cmTC_c3978"
|     [1/3] Building CUDA object CMakeFiles/cmTC_c3978.dir/main.cu.o
|     FAILED: CMakeFiles/cmTC_c3978.dir/main.cu.o
|     /yocto/tmp/work/armv8a_tegra194-poky-linux/vr360/git-r0/recipe-sysroot-native/usr/local/cuda-10.0/bin/nvcc -ccbin=/yocto/tmp/work/armv8a_tegra194-poky-linux/vr360/git-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-g++    -x cu -c main.cu -o CMakeFiles/cmTC_c3978.dir/main.cu.o && /yocto/tmp/work/armv8a_tegra194-poky-linux/vr360/git-r0/recipe-sysroot-native/usr/local/cuda-10.0/bin/nvcc -ccbin=/yocto/tmp/work/armv8a_tegra194-poky-linux/vr360/git-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-g++    -x cu -M main.cu -MT CMakeFiles/cmTC_c3978.dir/main.cu.o -o CMakeFiles/cmTC_c3978.dir/main.cu.o.d
|     cc1plus: fatal error: cuda_runtime.h: No such file or directory
|     compilation terminated.
|     ninja: build stopped: subcommand failed.
```

Without the two exports CUDA is not detected at all. I tried settings `EXTRA_OECMAKE` with various extra defines found in other issues (CMAKE_CUDA_TOOLKIT_INCLUDE_DIRECTORIES for example) but none of it helps.

I'm using the `linaro-7.2` toolchain to compile.

Is there something obvious I'm doing wrong?

Thanks in advance!

madisongh, 2019-06-15:
There have been other issues opened on this topic, such as #83.  I don't do any CUDA development myself, and I'm not a big cmake user, so `cuda.bbclass` has been pretty minimal in terms of what it supports -- mainly just enough to get opencv and the CUDA samples to compile.  I didn't even know about cmake's additions to treat CUDA as a first-class language.

So I took a small CUDA test program, converted it to build with cmake, and have made some updates to `cuda.bbclass` to try and make it "just work" for that use case.  The changes are on both the master and warrior branches; try the latest there and see if it works better.  You should remove the `CUDACXX` and `CUDAHOSTCXX` environment variables from your recipe when you do; they shouldn't be needed.

reyapo, 2019-06-18:
For me its working as well, thank you for the quick reply and fix!

closed 2019-06-18 by reyapo