OE4T/meta-tegra #140: Jetson Nano - Weston - resolution

jan-moravec, 2019-07-01:
Hi,

I am not able to change the resolution of HDMI output under the Weston. 

I have tried using the weston.ini configuration, xrandr application and the "/sys/class/graphics/fb0/modes", but with no success.

Is there a way to change the display resolution under Weston?

Thank you.

madisongh, 2019-07-01:
If you can you provide more details on what you tried, I'll do some digging and see what I can find.

jan-moravec, 2019-07-02:
Of course, thank you. I have the Jetson Nano. I did try both the core-image-weston and core-image-x11.

With the core-image-x11, the way to change the HDMI display resolution is the xrandr command. And it works.

With the core-image-weston, xrandr returns just:
```
root@jetson-nano:~# xrandr
Screen 0: minimum 320 x 200, current 1920 x 1080, maximum 8192 x 8192
XWAYLAND0 connected 1920x1080+0+0 (normal left inverted right x axis y axis) 0mm x 0mm
   1920x1080     59.96*+
```
There is no other resolution than 1920x1080 and I was not able to change the resolution.

I also tried to write to the graphics driver:
```
root@jetson-nano:~# cat /sys/class/graphics/fb0/modes
D:1920x1080p-60
U:720x400p-70
V:640x480p-60
U:640x480p-67
V:640x480p-72
V:640x480p-75
V:800x600p-56
V:800x600p-60
V:800x600p-72
V:800x600p-75
U:832x624p-75
V:1024x768p-60
V:1024x768p-70
V:1024x768p-75
V:1280x1024p-75
S:1920x1080p-60
S:1680x1050p-60
S:1440x900p-60
S:1280x1024p-60
S:1280x960p-60
S:1280x720p-60
S:1152x864p-75
D:720x480p-59
D:1280x720p-60
D:1280x720p-50
D:720x576p-50
U:640x480p-59
U:720x480p-59
U:720x576p-50
U:1280x720p-50
U:1280x720p-60
U:1440x480p-59
U:1440x576p-50
U:1920x1080p-50
U:1920x1080p-60
U:1280x720p-60
U:1920x1080p-60
root@jetson-nano:~# echo U:1280x720p-60 > /sys/class/graphics/fb0/mode
```
But the HDMI output stopped working and weston needed to be resetted.

The last option was trying the configuration file weston.ini.
- https://wiki.archlinux.org/index.php/Wayland#Configuration

I did manage to adjust the resolution of Weston. But the display resolution stayed 1920x1080 and the Weston desktop was not fullscreen, but just covered the 1280x720 area.

Thank you for any help.

madisongh, 2019-07-04:
It looks like better DRM/KMS support is coming with the next L4T/JetPack release, from [this NVIDIA forum post](https://devtalk.nvidia.com/default/topic/1048881/enable-wayland-on-tx2-with-jetpack-4-2/), and some support appears to be there now, but I'll have to rework the weston patches and libdrm shim library to see if it will work.

In the meantime, I'm not finding any way to set the mode dynamically - whatever mode gets set for the display at boot time is kept.

madisongh, 2019-07-20:
I just noticed that the L4T kit includes a script called `nv-enable-hard-coded-kernel-boot-display-mode.sh` in the `Linux_for_Tegra/kernel` directory that adds settings to the device tree to set a fixed resolution on the display at boot time.

madisongh, 2020-12-11:
No activity on this for well over a year, so closing this now.

closed 2020-12-11 by madisongh