OE4T/meta-tegra #144: core-image-weston based build errors: mesa/libglvnd/cairo installing the same files

mwcoop, 2019-07-06:
Hello,

I've created an image definition that extends core-image-weston a fair amount. My distro is largely based on madisongh/test-distro. Thanks!

Unfortunately, I'm getting severall errors about files like /usr/lib/libglesv1_cm.la being installed by combinations of mesa, cairo, and libglvnd. Any help is greatly appreciated. Image definition below.

SUMMARY = "Base image for robot master nodes"
HOMEPAGE = "https://www.build2c.org"
LICENSE = "MIT"

IMAGE_FEATURES += "package-management splash ssh-server-openssh hwcodecs"
IMAGE_LINGUAS = "en-us"

inherit core-image distro_features_check

# core-image-weston stuff
REQUIRED_DISTRO_FEATURES = "wayland"
CORE_IMAGE_BASE_INSTALL += "weston weston-init weston-examples gtk+3-demo clutter-1.0-examples"
CORE_IMAGE_BASE_INSTALL += "${@bb.utils.contains('DISTRO_FEATURES', 'x11', 'weston-xwayland matchbox-terminal', '', d)}"

CORE_OS = " \
    openssh openssh-keygen openssh-sftp-server \
    rsync  \
    bash  \
    tzdata \
    pulseaudio  \
    v4l-utils  \
"

WIFI_SUPPORT = " \
    crda \
    iw \
    wpa-supplicant \
"

DEV_SDK_INSTALL = " \
    binutils \
    binutils-symlinks \
    coreutils \
    cpp \
    cpp-symlinks \
    diffutils \
    file \
    g++ \
    g++-symlinks \
    gcc \
    gcc-symlinks \
    gdb \
    gdbserver \
    gettext \
    git \
    ldd \
    libstdc++ \
    libstdc++-dev \
    libtool \
    make \
    cmake \
    pkgconfig \
    python3-modules \
"

PREFERRED_VERSION_gcc-cross-${TARGET_ARCH} = "7.%"
PREFERRED_VERSION_g++-cross-${TARGET_ARCH} = "7.%"

DEV_EXTRAS = " \
    ntp \
    ntp-tickadj \
"

EXTRA_TOOLS_INSTALL = " \
    bzip2 \
    devmem2 \
    dosfstools \
    ethtool \
    fbset \
    findutils \
    i2c-tools \
    iperf3 \
    iproute2 \
    iptables \
    less \
    nano \
    netcat \
    procps \
    sysfsutils \
    tcpdump \
    unzip \
    util-linux \
    vim \
    wget \
    zip \
"

ROBOT_STUFF = "  \
    mdns  \
    can-utils libsocketcan-dev  \
    python3-pip  \
    python3-pyserial  \
    python3-pyyaml  \
    python3-pyopenssl  \
    python3-pexpect  \
    python3-docopt  \
    python3-pybonjour  \
    opencv opencv-dev  \
    openssl-dev  \
    boost boost-dev  \
    bonefish  \
    python3-autobahn python3-umsgpack  \
    autobahn-cpp autobahn-cpp-dev  \
    msgpack-c msgpack-c-dev  \
    spdlog spdlog-dev  \
    websocketpp websocketpp-dev  \
    gstreamer1.0 gstreamer1.0-plugins-good gstreamer1.0-plugins-bad  \
"

ROBOT_STUFF_append_tegra = "  \
    cuda-core-dev cuda-cublas-dev cuda-cudart-dev cuda-cufft-dev  \
    cuda-curand-dev cuda-cusparse-dev cuda-driver cuda-misc-headers  \
    cuda-npp-dev cuda-nvrtc-dev cuda-command-line-tools cuda-samples  \
    cudnn-dev cudnn-samples  \
    tensorrt-dev tensorrt-samples  \
    libvisionworks-dev libvisionworks-tracking-dev libvisionworks-tracking-samples  \
    libvisionworks-tracking-doc  \
    tegra-mmapi-samples  \
    gstreamer1.0-plugins-tegra  \
    gstreamer1.0-omx-tegra gstreamer1.0-plugins-nveglgles  \
    gstreamer1.0-plugins-nvvideo4linux2 gstreamer1.0-plugins-nvvideosinks  \
    tegra-nvphs  \
    tegra-nvpmodel  \
    tegra-nvs-service  \
    tegra-tools-tegrastats tegra-tools-jetson-clocks  \
    tegra-libraries-libv4l-plugins tegra-libraries-argus tegra-libraries-libnvosd tegra-libraries-dev  \
    gpu-burn  \
    nl80211-wlan  \
    kernel-modules  \
"
ROBOT_STUFF_append_tegra210 = " sdcard-layout"

EXTRA_APPS = "gstreamer-tests ifupdown rng-tools"

ROBOT_KERNEL_MODULES = "  \
    kernel-module-can-dev kernel-module-can-raw kernel-module-mttcan  \
"

KERNEL_MODULE_AUTOLOAD += " ${ROBOT_KERNEL_MODULES}"

IMAGE_INSTALL += " \
    ${CORE_OS} \
    ${DEV_SDK_INSTALL} \
    ${DEV_EXTRAS} \
    ${EXTRA_TOOLS_INSTALL} \
    ${WIFI_SUPPORT} \
    ${ROBOT_STUFF}  \
"

set_local_timezone() {
    ln -sf /usr/share/zoneinfo/EST5EDT ${IMAGE_ROOTFS}/etc/localtime
}

disable_bootlogd() {
    echo BOOTLOGD_ENABLE=no > ${IMAGE_ROOTFS}/etc/default/bootlogd
}

ROOTFS_POSTPROCESS_COMMAND += " \
    set_local_timezone ; \
    disable_bootlogd ; \
"

export IMAGE_BASENAME = "robot-image-master"

mwcoop, 2019-07-06:
Hmm. Seeing the issues with core-image-weston and core-image-minimal also.

madisongh, 2019-07-07:
The `.la` files shouldn't even be getting packaged. Are you inheriting the `remove-libtool` bbclass? You can check by looking at the output of
```
$ bitbake -e core-image-minimal | grep '^INHERIT='
```
If it's not there, try adding it to the `INHERIT_DISTRO` setting your distro config file.

If that's not the issue, can you include the specific errors being logged?

mwcoop, 2019-07-08:
Adding remove-libtool seems to be helping. However, now I'm getting errors compiling gcc-sanitizers_7.3 because it cannot find the header sys/ustat.h.

Any thoughts on this one?

mwcoop, 2019-07-08:
Ok. Removing tools-sdk from the local.conf removed gcc-sanitizers. Remaining issue is postinstall for busybox. Log attached. Any thoughts?

closed 2019-07-09 by mwcoop