OE4T/meta-tegra #145: Cannot install crda/wireless-regdb for WiFi support

mwcoop, 2019-07-09:
Hello,

I'm trying to install crda to get WiFi support on my Jetson. The build fails because it isn't installing wireless-regdb. The current recipe for wireless-regdb states that it works by default with kernel >= 4.15. It does state that the kernel_wireless_regdb.bbclass file should be inherited by older kernel versions. Where should this be added? I tried adding it in meta-tegra's linux-tegra_4.9.bb recipe but then wireless-regdb gives a conflict error with wireless-regdb-static.?? Help greatly appreciated.

madisongh, 2019-07-10:
That comment in the wireless-regdb recipe only applies if you're compiling a static database into the kernel - in which case you would include `wireless-regdb-static` in your image rather than `wireless-regdb` (and you would have to change the kernel config to set `CONFIG_CFG80211_INTERNAL_REGDB` and inherit that bbclass so the database file you provide gets copied into the kernel source tree during the build).

The Tegra kernel config comes with `CONFIG_CFG80211_CRDA_SUPPORT` enabled, so you should be able to just add `crda` to your build and be good to go.

mwcoop, 2019-07-11:
That's what I've done before. Attached is the log with the error during do_rootfs.

madisongh, 2019-07-11:
The log file attachment didn't come through - I suspect Github filters out attachments in its e-mail interface.  Could you post it here?

mwcoop, 2019-07-11:
Yep. Sorry about that. :) I'm actually blind and find the web UI a pain. Anyway, here it is.
[do_rootfs.log](https://github.com/madisongh/meta-tegra/files/3382055/do_rootfs.log)

madisongh, 2019-07-11:
Hmm. Worked for me in my test build on warrior.  
The default settings OE uses for apt don't provide much useful information, but the "is not going to be installed" error is often due to either a non-existent deb package (not created because of empty contents), or due to conflict... like if in one of your builds you tried adding `wireless-regdb-static`, which conflicts with `wireless-regdb`, and that setting is still hanging around.

Adding
```
APT_ARGS = "-o Debug::pkgProblemResolver=true"
```
to your local.conf file should get you better output from apt-get to help in tracking down the cause of the failure.

closed 2019-08-04 by mwcoop