OE4T/meta-tegra #152: L4T R32.1 for Yocto/OE release Thud

leon-anavi, 2019-07-18:
Hi @madisongh,

Are you interested in accepting a GitHub PR that backports L4T R32.1 with CUDA 10 from branch Warrior to a new branch compatible with Yocto/OE release Thud?

I have already done this and I interested in contributing it to the upstream. The benefit is that CUDA 10 requires GCC 7 and Poky for Yocto/OE release offers GCC 7 which allows using CUDA 10 without an external toolchain. If you are interested in this contribution could you please create a new branch with appropriate name, for example thud-l4t-r32.1, and I will make a GitHub pull request with the required changes for backporting?

Best regards,
Leon

madisongh, 2019-07-20:
I thought I had run into some compatibility issues between OE-Core thud and some the L4T R32.1 binaries, but I don't recall now what those were.  If you have things working, great!  I've created a `thud-l4t-r32.1` branch as requested.

leon-anavi, 2019-07-22:
Hi @madisongh,

I have created a GitHub Pull Request https://github.com/madisongh/meta-tegra/pull/153. I have copied what we have working in Warrior and adapted it for Thud therefore there are numerous changes. However I haven't added the machine definitions for Xavier and Nano yet. Probably it will work for them too but I don't have this hardware at the moment to test.

Using the current version from the GiHub PR I have built and successfully tested on NVIDIA Jetson TX2 core-image-minimal (based on Poky Thud) with cuda-samples, Linux kernel 4.9.140-l4t-r32.1+g3c02a65, GCC 7 and CUDA 10.

Best regards,
Leon

madisongh, 2019-07-28:
I've merged your PR, started running some test builds, and have pushed a couple of fixes for them.  Still not getting successful builds yet across all the platforms, but as I have time I'll look into the failures.

leon-anavi, 2019-07-28:
Thanks for merging my GitHub PR and for the improvements. I will close the issue. Best regards, Leon

closed 2019-07-28 by leon-anavi