OE4T/meta-tegra #154: Using wic to create secondary RFS Partition for Jetson Nano SD Card

phishstang65, 2019-07-22:
I am trying to have Yocto create a secondary RFS Partition using the "wic" tool so that I can have the Jetson perform full and safe Firmware updates. I have meta-tegra creating my image with a working *.sdcard image ( without the second rfs partition ), and now I need to use wic to add the second partition.

I've added the following to my image:

```
# Create multi rootfs with the wic tool
IMAGE_FSTYPES+="wic"
WKS_FILE = "custom-wic.wks"

# Allow wics to partition this image by providing boot files
IMAGE_BOOT_FILES = "Image tegra210-p3448-0000-p3449-0000-a02.dtb u-boot-jetson-nano.bin u-boot-dtb.bin"
```
My wic ( .wks ) file looks like this:

```
# Boot partition
part /boot --source bootimg-partition --ondisk mmcblk0 --fstype=vfat --label boot --active --align 4 --size 16

# Rootfs partition
part / --source rootfs --ondisk mmcblk0 --fstype=ext4 --label partA --align 4 --size 4096M

# Rootfs backup partition
part /rootfsB --source rootfs --ondisk mmcblk0 --fstype=ext4 --label partB --align 4 --size 4096M

# User data partition
part /userdata --ondisk mmcblk0 --fstype=vfat --label userdata --align 4 --size 2048M

# Additional U-Boot
bootloader --append="console=ttyS0,115200"
```

The build completes and the resulting *.wic file shows the contents:

```
ubuntu@ip-172-31-23-71:~/Desktop/jetson-yocto/build/tmp/deploy/images/jetson-nano$ wic ls jetson-nano-core-image-iot-jetson-nano-20190722124859.rootfs.wic
Num     Start        End          Size      Fstype
 1          4096     59738111     59734016  fat16
 2      59740160   5643198463   5583458304  ext4
 3    5643198464  11226656767   5583458304  ext4
 4   11226656768  13374140415   2147483648  fat32

ubuntu@ip-172-31-23-71:~/Desktop/jetson-yocto/build/tmp/deploy/images/jetson-nano$ wic ls jetson-nano-core-image-iot-jetson-nano-20190722124859.rootfs.wic:1
Volume in drive : is boot       
 Volume Serial Number is 7039-B7A7
Directory for ::/

IMAGE         34295816 2019-07-22  12:53  Image
TEGRA2~1 DTB    200920 2019-07-22  12:53  tegra210-p3448-0000-p3449-0000-a02.dtb
U-BOOT~1 BIN    481280 2019-07-22  12:53  u-boot-dtb.bin
U-BOOT~2 BIN    481280 2019-07-22  12:53  u-boot-jetson-nano.bin
        4 files          35 459 296 bytes
                         24 133 632 bytes free

ubuntu@ip-172-31-23-71:~/Desktop/jetson-yocto/build/tmp/deploy/images/jetson-nano$ wic ls jetson-nano-core-image-iot-jetson-nano-20190722124859.rootfs.wic:2
debugfs 1.44.1 (24-Mar-2018)
      2   40755 (2)      0      0    4096 22-Jul-2019 12:53 .
      2   40755 (2)      0      0    4096 22-Jul-2019 12:53 ..
     11   40700 (2)      0      0   16384 22-Jul-2019 12:53 lost+found
     12   40555 (2)      0      0    4096 18-Jul-2019 15:51 sys
     13   40755 (2)      0      0    4096 22-Jul-2019 12:53 etc
    573   40555 (2)      0      0    4096 18-Jul-2019 15:51 proc
    574   40755 (2)      0      0    4096 22-Jul-2019 12:52 var
    633   41777 (2)      0      0    4096 18-Jul-2019 15:51 tmp
    634   40755 (2)      0      0    4096 22-Jul-2019 12:51 mnt
    636   40755 (2)      0      0    4096 18-Jul-2019 15:51 dev
    637   40755 (2)      0      0    4096 18-Jul-2019 15:59 home
   1820   40755 (2)      0      0    4096 22-Jul-2019 12:52 bin
   1976   40755 (2)      0      0    4096 18-Jul-2019 15:59 lib
   3370   40755 (2)      0      0    4096 22-Jul-2019 12:51 sbin
   3495   40755 (2)      0      0    4096 18-Jul-2019 15:51 run
   3496   40755 (2)      0      0    4096 18-Jul-2019 15:51 media
   3497   40755 (2)      0      0    4096 18-Jul-2019 16:09 usr
  38743   40755 (2)      0      0    4096 22-Jul-2019 12:51 boot
```

Comparing the resulting images in Ubuntu looks like:

```
file jetson-nano-core-image-iot-jetson-nano-20190722124859.sdcard
jetson-nano-core-image-iot-jetson-nano-20190722124859.sdcard: DOS/MBR boot sector; partition 1 : ID=0xee, start-CHS (0x0,0,2), end-CHS (0x3ff,255,63), startsector 1, 33554431 sectors, extended partition table (last)

file jetson-nano-core-image-iot-jetson-nano-20190722124859.rootfs.wic
jetson-nano-core-image-iot-jetson-nano-20190722124859.rootfs.wic: DOS/MBR boot sector; partition 1 : ID=0xc, active, start-CHS (0x0,0,9), end-CHS (0x38f,2,4), startsector 8, 116668 sectors; partition 2 : ID=0x83, start-CHS (0x38f,2,9), end-CHS (0x3ff,3,32), startsector 116680, 10905192 sectors; partition 3 : ID=0x83, start-CHS (0x3ff,3,32), end-CHS (0x3ff,3,32), startsector 11021872, 10905192 sectors; partition 4 : ID=0xc, start-CHS (0x3ff,3,32), end-CHS (0x3ff,3,32), startsector 21927064, 4194304 sectors

fdisk -lu jetson-nano-core-image-iot-jetson-nano-20190722124859.rootfs.wic
Disk jetson-nano-core-image-iot-jetson-nano-20190722124859.rootfs.wic: 12.5 GiB, 13374140416 bytes, 26121368 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0xe7bef76a

Device                                                            Boot    Start      End  Sectors  Size Id Type
jetson-nano-core-image-iot-jetson-nano-20190722124859.rootfs.wic1 *           8   116675   116668   57M  c W95 
jetson-nano-core-image-iot-jetson-nano-20190722124859.rootfs.wic2        116680 11021871 10905192  5.2G 83 Linu
jetson-nano-core-image-iot-jetson-nano-20190722124859.rootfs.wic3      11021872 21927063 10905192  5.2G 83 Linu
jetson-nano-core-image-iot-jetson-nano-20190722124859.rootfs.wic4      21927064 26121367  4194304    2G  c W95

fdisk -lu jetson-nano-core-image-iot-jetson-nano-20190722124859.sdcard
Disk jetson-nano-core-image-iot-jetson-nano-20190722124859.sdcard: 16 GiB, 17179869184 bytes, 33554432 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
Disk identifier: 5220E26C-0FFF-40C6-9906-DA49A13EB1BA

Device                                                         Start      End  Sectors  Size Type
jetson-nano-core-image-iot-jetson-nano-20190722124859.sdcard1  24576 33554398 33529823   16G Linux fi
jetson-nano-core-image-iot-jetson-nano-20190722124859.sdcard2   2048     2303      256  128K Linux fi
jetson-nano-core-image-iot-jetson-nano-20190722124859.sdcard3   4096     4991      896  448K Linux fi
jetson-nano-core-image-iot-jetson-nano-20190722124859.sdcard4   6144     7295     1152  576K Linux fi
jetson-nano-core-image-iot-jetson-nano-20190722124859.sdcard5   8192     8319      128   64K Linux fi
jetson-nano-core-image-iot-jetson-nano-20190722124859.sdcard6  10240    10623      384  192K Linux fi
jetson-nano-core-image-iot-jetson-nano-20190722124859.sdcard7  12288    13439     1152  576K Linux fi
jetson-nano-core-image-iot-jetson-nano-20190722124859.sdcard8  14336    14463      128   64K Linux fi
jetson-nano-core-image-iot-jetson-nano-20190722124859.sdcard9  16384    17663     1280  640K Linux fi
jetson-nano-core-image-iot-jetson-nano-20190722124859.sdcard10 18432    19327      896  448K Linux fi
jetson-nano-core-image-iot-jetson-nano-20190722124859.sdcard11 20480    20735      256  128K Linux fi
jetson-nano-core-image-iot-jetson-nano-20190722124859.sdcard12 22528    22687      160   80K Linux fi

Partition table entries are not in disk order.
```

The problem is that the board does not boot after I dd the *.wic file to the sd card. It does work with the old sdcard image. 

Under meta-tegra, I see an "sdcard-layout.in" file for jetson-nano:

`recipes-bsp/sdcard-layout/sdcard-layout/jetson-nano/sdcard-layout.in`

```
2,TBC,131072,nvtboot_cpu.bin.encrypt
3,RP1,458752,DTBFILE
4,EBT,589824,cboot.bin.encrypt
5,WB0,65536,warmboot.bin.encrypt
6,BPF,196608,sc7entry-firmware.bin.encrypt
7,TOS,589824,tos-mon-only.img.encrypt
8,EKS,65536,eks.img
9,LNX,655360,LNXFILE
10,DTB,458752,DTBFILE
11,RP4,131072,rp4.blob
12,BMP,81920,bmp.blob
1,APP,0,APPFILE
```

I don't understand what that is, but I can guess that my sd card image layout is broken, which is why it doesn't boot. What do I have to do to add the second partition and use the wic tool successfully? Thanks.

madisongh, 2019-07-23:
The `sdcard-layout.in` file is derived from the partition table in the `create-jetson-nano-sd-card-image.sh` script in the L4T kit.  That, in turn, is derived from the flash layout XML file that the tegraflash tools use.

I haven't played around with wic that much, so I'm not sure whether you can coax it into formatting and populating the SDcard in a way that will make the Jetson bootloader happy.  The partition table format should be GPT (with "protective" MBR is OK).  The non-APP partitions probably need to be in the exact order, with the exact sizes, listed in the layout file so that the signature checks that bootloader performs on the signed boot binaries will pass.  You could just change the last line to provide a size for APP, then add a `13,APP_B,0,APPFILE` to have a second rootfs partition.

phishstang65, 2019-07-23:
Thank you for the response. I tried this by modifying the sdcard-layout.in file as shown below:

```
2,TBC,131072,nvtboot_cpu.bin.encrypt
3,RP1,458752,DTBFILE
4,EBT,589824,cboot.bin.encrypt
5,WB0,65536,warmboot.bin.encrypt
6,BPF,196608,sc7entry-firmware.bin.encrypt
7,TOS,589824,tos-mon-only.img.encrypt
8,EKS,65536,eks.img
9,LNX,655360,LNXFILE
10,DTB,458752,DTBFILE
11,RP4,131072,rp4.blob
12,BMP,81920,bmp.blob
1,APP,4294967296,APPFILE
13,APP_B,4294967296,APPFILE
```
The image is created, but the jetson boots to the initramfs file system ( no home parition and df only shows 1 rfs ). It doesn't seem to make it to the rootfs.

madisongh, 2019-07-24:
What does the kernel command line look like when it boots?  The `init` script in the initramfs will mount the rootfs based on the `root=` argument there.

triblex, 2019-09-30:
After doing as listed in the issue, using all the same file contents, i am still unable to even boot at all. Did you manage to make it work @phishstang65? The created .wic image only has 4 parts when using the 'lsblk' command: PartA, PartB, boot and userdata.

phishstang65, 2019-09-30:
@triblex - No I never got wic to work with meta-tegra. I have Yocto build the rootfs and multiple other NVIDIA related partitions, and then I add my own afterward as a post build hook.

lumurillo, 2020-11-07:
Have anyone tried to create a WIC image to the jetson nano eMMC?

madisongh, 2020-11-08:
The topic has come up a couple of times. The big issue is that NVIDIA's early-stage bootloaders don't read the partition table, but instead get sector offsets programmed into the BCT (boot control table), so you really have to use their tools to ensure that the BCT and GPT are in sync.

It might be possible to write a wic plugin to handle this, maybe auto-generating the XML that the NVIDIA tools use from the layout you put into the wks files, or something along those lines.  It's not something I've really looked closely at, though.

madisongh, 2020-12-11:
Closing this now.  If anyone is interested in further discussing the possibility of making the Tegra flashing tools work with wic, feel free to start a [discussion](https://github.com/OE4T/meta-tegra/discussions).

closed 2020-12-11 by madisongh