OE4T/meta-tegra #158: GL app crashes over remote ssh

dagmcr, 2019-07-30:
Hi!

We are using part of `meta-tegra` (only the minimum necessary) as part of our bsp but we have some problems with GL applications while we execute them over ssh with X11 forwarding.

Applications tested: `qv4l2`, `glxinfo` and `glxgears`.
* glx apps: `X Error of failed request:  BadValue (integer parameter out of range for operation)`
* qv4l2: `QOpenGLFunctions created with non-current context`.

> **Note**: the applications are working fine over main display (`:0`).

Have you ever experienced this or you have any idea how we can trace it?

**Here the log description of the issue:**
- `qv4l2` works when you pick up any user rather than root.
- System boots okay and displays `qv4l2`, `glxinfo` and `glxgears`.
- `Xorg.0.log` fails to allocate memory:
```
# cat /var/log/Xorg.0.log | grep EE
[    11.597] Current Operating System: Linux qt5507 4.9.140 #1241 SMP PREEMPT Mon Jul 29 15:17:30 CEST 2019 aarch64
	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[    12.860] (EE) NVIDIA(0): Failed to allocate NVIDIA Error Handler
[    13.112] (II) Initializing extension MIT-SCREEN-SAVER
```
> Note: `libglxserver_nvidia.so` is renamed to `libglx.so`. Location: `/usr/lib/xorg/modules/extensions/`

* `qv4l2` log:
```
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
QOpenGLFunctions created with non-current context
Segmentation fault
```
* `qv4l2` gdb log:
```
warning: Unable to find libthread_db matching inferior's thread library, thread debugging will not be available.
[New LWP 4688]
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
[New LWP 4689]
[Detaching after fork from child process 4690]
QOpenGLFunctions created with non-current context

Thread 1 "qv4l2" received signal SIGSEGV, Segmentation fault.
0x0000007fb75444d0 in QOpenGLContext::shareGroup() const () from /usr/lib/libQt5Gui.so.5
```
* `glxinfo` and `glxgears` log:
```
root@qt5507:~# glxinfo 
name of display: localhost:10.0
X Error of failed request:  BadValue (integer parameter out of range for operation)
  Major opcode of failed request:  154 (GLX)
  Minor opcode of failed request:  24 (X_GLXCreateNewContext)
  Value in failed request:  0x0
  Serial number of failed request:  39
  Current serial number in output stream:  40
root@qt5507:~# glxgears
X Error of failed request:  BadValue (integer parameter out of range for operation)
  Major opcode of failed request:  154 (GLX)
  Minor opcode of failed request:  3 (X_GLXCreateContext)
  Value in failed request:  0x0
  Serial number of failed request:  27
  Current serial number in output stream:  28
```
**Any suggestion?**

**More tests with libglx sever:**
> Note: these are additional tests performed over the above scenario.

* If I keep `libglxserver_nvidia.so` installed in `/usr/lib/xorg/modules/extensions` from the nvidia_drivers.tbz2 like the recipe xserver recipe:

    - https://github.com/madisongh/meta-tegra/blob/master/recipes-bsp/tegra-binaries/xserver-xorg-video-nvidia.inc
    - https://github.com/madisongh/meta-tegra/blob/master/recipes-bsp/tegra-binaries/xserver-xorg-video-nvidia_32.1.0.bb

Then glx module loading fails:

```
[    10.418] (EE) Failed to load module "glx" (module does not exist, 0)
```

* If I copy/add `libglx.so` from L4T 32.1.0 release (ubuntu) then, Xorg fails:
> Note:
> ```
> # ls -l /usr/lib/xorg/modules/extensions/
> ls -l /usr/lib/xorg/modules/extensions/
> total 6972
> -rw-r--r-- 1 root root  240520 Jul 30 13:35 libglx.so
> -rwxr-xr-x 1 root root 6933704 Jul 20 15:33 libglxserver_nvidia.so
```

# cat /var/log/Xorg.0.log | grep EE
[    24.935] Current Operating System: Linux qt5507 4.9.140 #1241 SMP PREEMPT Mon Jul 29 15:17:30 CEST 2019 aarch64
	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[    26.012] (EE) NVIDIA(0): Failed to allocate NVIDIA Error Handler
[    26.255] (II) Initializing extension MIT-SCREEN-SAVER
[    26.255] (EE) AIGLX: reverting to software rendering
[    26.311] (EE) 
[    26.311] (EE) Backtrace:
[    26.311] (EE) 0: /usr/bin/Xorg (xorg_backtrace+0x5c) [0x55633eef14]
[    26.311] (EE) 1: /usr/bin/Xorg (0x556324f000+0x1a3cf0) [0x55633f2cf0]
[    26.311] (EE) 2: linux-vdso.so.1 (__kernel_rt_sigreturn+0x0) [0x7fa854e6a0]
[    26.311] (EE) 3: /usr/lib/xorg/modules/extensions/libglx.so (0x7fa7b9e000+0x2a18c) [0x7fa7bc818c]
[    26.311] (EE) 
[    26.311] (EE) Segmentation fault at address 0x7fa7175368
[    26.311] (EE) 
[    26.311] (EE) Caught signal 11 (Segmentation fault). Server aborting
[    26.311] (EE) 
[    26.311] (EE) 
[    26.311] (EE) Please also check the log file at "/var/log/Xorg.0.log" for additional information.
[    26.311] (EE) 
[    30.592] (EE) Server terminated with error (1). Closing log file.
```

madisongh, 2019-08-03:
If I'm understanding correctly, you want to ssh from the Jetson to another host to run an OpenGL app, is that right?

Did you add `+iglx` to the command line when starting the X server on the Jetson?  From what I've been able to dig up, the xorg server keeps "indirect GLX" turned off by default, but you can turn it on with that option.  I tried it (editing `/etc/default/xserver-nodm` to add `+iglx` to the `ARGS` setting and restarting xserver-nodm), and it appears to work - I could ssh to another host with X11 forwarding enabled, then run `glxinfo` with no problem.  Without `+iglx` enabled on the server, I got similar 'failed request' errors.  This was in a Sato image on my TX2.

dagmcr, 2019-08-05:
Yeap! That's correct what I'm trying to do.

I tried what you suggested and still not able to X11 forwarding `glxgears`/`glxinfo`/`qv4l2`.

Still, the same behavior: `qv4l2` application is working under any other user rather than root and `glx*` applications are not X11 forwarding at all.

Also, I tried to enable indirect GLX option in Xorg:

`/usr/share/X11/xorg.conf.d/00-tegra-xorg.conf`:
```
Section "ServerFlags"
    Option "AllowIndirectGLX" "on"
    Option "IndirectGLX" "on" 
EndSection
```

And the result is `/var/log/Xorg.0.log`:
```
(**) Option "IndirectGLX" "on"
```
> **Note**: before adding above setup, xorg log was `(II) Indirect GLX disabled.`

But with the same errors.

dagmcr, 2019-08-05:
In the meantime, I pull requested to fix glx module load error. Which might be solved by renaming the module as I did or by telling Xorg the right place to search. In ubuntu they had a similar issue that we can apply here:
https://bugs.launchpad.net/ubuntu/+source/nvidia-graphics-drivers-304/+bug/1276123

Pull requests:
https://github.com/madisongh/meta-tegra/pull/161
https://github.com/madisongh/meta-tegra/pull/161
https://github.com/madisongh/meta-tegra/pull/161
https://github.com/madisongh/meta-tegra/pull/161

madisongh, 2019-08-06:
That Ubuntu bug is due to having both the mesa glx extension and the NVIDIA one installed at the same time, with the X server trying to mix the mesa glx extension with the NVIDIA driver - a situation that shouldn't happen in an L4T setup.

Can you flash your Jetson using L4T/JetPack, or at least a stock Poky+meta-tegra core-image-sato build, and see if you have the same issue? As I mentioned, I could successfully get indirect GLX to work on my setup.

dagmcr, 2019-08-06:
My yocto setup only includes `poky`+`meta-tegra` poiting to `warrior` branches.
```
Build Configuration:
BB_VERSION           = "1.42.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-poky-linux"
MACHINE              = "jetson-tx2"
DISTRO               = "poky"
DISTRO_VERSION       = "2.7.1"
TUNE_FEATURES        = "aarch64 armv8a crc"
TARGET_FPU           = ""
meta                 
meta-poky            
meta-yocto-bsp       = "warrior:01b8a8b54bc569e5ef3f5e6fc6abcee365ab25d9"
meta-tegra           = "warrior:1bb7b89297cff9483ec1862ccaf726cf3d91bdcd"
```

After flashing the tx2 eval board and checking the X.org logs I can see how the glx module
fails.

```
root@jetson-tx2:~# cat /var/log/Xorg.0.log | grep EE
[    10.807] Current Operating System: Linux jetson-tx2 4.9.140-l4t-r32.1+g3c02a65d9170 #1 SMP PREEMPT Wed Jul 31 08:28:06 UTC 2019 aarch64
	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[    10.822] (EE) Failed to load module "glx" (module does not exist, 0)
[    15.252] (EE) NVIDIA(0): Failed to allocate NVIDIA Error Handler
[    15.470] (II) Initializing extension MIT-SCREEN-SAVER
```

To solve this issue I created above pull requests which just make the rename of the NVIDIA libglx module `libglxserver_nvidia.so` to `libglx.so`. But I've seen your comment about it:

> The X server should be loading nvidia_drv which should then load NVIDIA's proprietary glxserver library to provide GLX support.

So, I guess to solve it properly we should fix it through `nvidia_drv`.

Checking the difference with L4T/JetPack I can see how the default glx module (libglx.so) is loaded
but also glx nvidia submodule (libglxserver_nvidia.so).

Next logs are extracted from L4T 32.1 running Ubuntu:
```
[    15.494] (II) LoadModule: "glx"
[    15.496] (II) Loading /usr/lib/xorg/modules/extensions/libglx.so
[    15.512] (II) Module glx: vendor="X.Org Foundation"
...
[    15.543] (II) Loading sub module "glxserver_nvidia"
[    15.543] (II) LoadModule: "glxserver_nvidia"
[    15.632] (II) Module glxserver_nvidia: vendor="NVIDIA Corporation"
[    15.632]    compiled for 4.0.2, module version = 1.0.0
[    15.632]    Module class: X.Org Server Extension
```

In regards to the indirect GLX... 
If I enable `+iglx` I can see

Before:
```
root@jetson-tx2:~# cat /var/log/Xorg.0.log | grep GLX
[    15.145] (II) NVIDIA GLX Module  32.1.0  Release Build  (integ_stage_rel)  (buildbrain@mobile-u64-2988)  Wed Mar 13 00:25:11 PDT 2019
[    15.470] (II) Initializing extension GLX
[    15.471] (II) Initializing extension GLX
[    15.471] (II) Indirect GLX disabled.
[    15.471] (II) Initializing extension NV-GLX
```

After:
```
root@jetson-tx2:~# cat /var/log/Xorg.0.log | grep GLX
[    14.913] (II) NVIDIA GLX Module  32.1.0  Release Build  (integ_stage_rel)  (buildbrain@mobile-u64-2988)  Wed Mar 13 00:25:11 PDT 2019
[    15.163] (II) Initializing extension GLX
[    15.163] (II) Initializing extension GLX
[    15.163] (II) Initializing extension NV-GLX
```

`Xorg` process:
```
root@jetson-tx2:~# ps | grep Xsess
 2860 root      3680 S    xinit /etc/X11/Xsession -- /usr/bin/Xorg :0 -br -pn +iglx
```

But even though with indirect GLX enabled in sato or even using L4T/JetPack, I can't forward `glxinfo` application to the remote display `localhost:10.0`.

BTW, sorry for adding another topic here but the kernel is crashing in warrior. I will add an issue for that.

And thanks for the support :)

dagmcr, 2019-08-06:
Okay, I can see in the Xorg logs how to glxserver module is loaded properly so, I will close the PRs.

But still, I can't execute `glxinfo`/`glxgears` with remote display: 

```
$ ssh -X root@jetson-tx2.local
Are you sure you want to continue connecting (yes/no)? yes
root@jetson-tx2:~# glxinfo 
name of display: localhost:10.0
X Error of failed request:  BadValue (integer parameter out of range for operation)
  Major opcode of failed request:  154 (GLX)
  Minor opcode of failed request:  24 (X_GLXCreateNewContext)
  Value in failed request:  0x0
  Serial number of failed request:  39
  Current serial number in output stream:  40
root@jetson-tx2:~# ps | grep Xsession 
 2860 root      3680 S    xinit /etc/X11/Xsession -- /usr/bin/Xorg :0 -br -pn +iglx
root@jetson-tx2:~# cat /var/log/Xorg.0.log | grep -i glx
[    10.722] (II) "glx" will be loaded by default.
[    10.722] (II) LoadModule: "glx"
[    10.724] (WW) Warning, couldn't open module glx
[    10.724] (EE) Failed to load module "glx" (module does not exist, 0)
[    14.948] (II) Loading sub module "glxserver_nvidia"
[    14.948] (II) LoadModule: "glxserver_nvidia"
[    14.949] (II) Loading /usr/lib/xorg/modules/extensions/libglxserver_nvidia.so
[    15.051] (II) Module glxserver_nvidia: vendor="NVIDIA Corporation"
[    15.051] (II) NVIDIA GLX Module  32.1.0  Release Build  (integ_stage_rel)  (buildbrain@mobile-u64-2988)  Wed Mar 13 00:25:11 PDT 2019
[    15.355] (II) Initializing extension GLX
[    15.355] (II) Initializing extension GLX
[    15.355] (II) Initializing extension NV-GLX
```

dagmcr, 2019-08-06:
This is the kernel crash log in `sato` which I couldn't reproduce it in `weston` or `minimal`.

[dmesg.log](https://github.com/madisongh/meta-tegra/files/3472911/dmesg.log)

```
[   21.844945] ------------[ cut here ]------------
[   21.844946] ------------[ cut here ]------------
[   21.844964] WARNING: CPU: 5 PID: 3162 at /workdir/build/tmp/work-shared/jetson-tx2/kernel-source/drivers/tty/vt/vt_ioctl.c:223 vt_waitactive+0x148/0x1a8
[   21.844966] ------------[ cut here ]------------
[   21.844967] ------------[ cut here ]------------
[   21.844973] Modules linked in:
[   21.844974] WARNING: CPU: 4 PID: 3161 at /workdir/build/tmp/work-shared/jetson-tx2/kernel-source/drivers/tty/vt/vt_ioctl.c:223 vt_waitactive+0x148/0x1a8
[   21.844978] WARNING: CPU: 3 PID: 3159 at /workdir/build/tmp/work-shared/jetson-tx2/kernel-source/drivers/tty/vt/vt_ioctl.c:223 vt_waitactive+0x148/0x1a8
[   21.844979]  bnep<4>[   21.844980] Modules linked in:
[   21.844981] Modules linked in:
[   21.844982]  bcmdhd bnep bnep cfg80211 bcmdhd bcmdhd nvgpu cfg80211 cfg80211 nvgpu nvgpu
[   21.844990] 
[   21.844990] 
[   21.844993] CPU: 5 PID: 3162 Comm: console-kit-dae Not tainted 4.9.140-l4t-r32.1+g3c02a65d9170 #1
[   21.844995] CPU: 4 PID: 3161 Comm: console-kit-dae Not tainted 4.9.140-l4t-r32.1+g3c02a65d9170 #1
[   21.844997] CPU: 3 PID: 3159 Comm: console-kit-dae Not tainted 4.9.140-l4t-r32.1+g3c02a65d9170 #1
[   21.844998] Hardware name: quill (DT)
[   21.844999] Hardware name: quill (DT)
[   21.845000] Hardware name: quill (DT)
[   21.845001] task: ffffffc1eb503800 task.stack: ffffffc1d0010000
[   21.845002] task: ffffffc1eb507000 task.stack: ffffffc1d000c000
[   21.845003] task: ffffffc1eb0db800 task.stack: ffffffc1d0004000
[   21.845004] PC is at vt_waitactive+0x148/0x1a8
[   21.845005] PC is at vt_waitactive+0x148/0x1a8
[   21.845006] PC is at vt_waitactive+0x148/0x1a8
[   21.845006] LR is at vt_waitactive+0x144/0x1a8
[   21.845008] pc : [<ffffff80086c0ba8>] lr : [<ffffff80086c0ba4>] pstate: 40400045
[   21.845008] LR is at vt_waitactive+0x144/0x1a8
[   21.845009] LR is at vt_waitactive+0x144/0x1a8
[   21.845010] sp : ffffffc1d0013c40
[   21.845011] pc : [<ffffff80086c0ba8>] lr : [<ffffff80086c0ba4>] pstate: 40400045
[   21.845012] pc : [<ffffff80086c0ba8>] lr : [<ffffff80086c0ba4>] pstate: 40400045
[   21.845014] x29: ffffffc1d0013c40 
[   21.845014] sp : ffffffc1d000fc40
[   21.845015] sp : ffffffc1d0007c40
[   21.845017] x28: ffffffc1eb503800 x29: ffffffc1d000fc40 
[   21.845019] x29: ffffffc1d0007c40 
[   21.845020] x28: ffffffc1eb507000 x28: ffffffc1eb0db800 x27: ffffff8008f12000 
[   21.845024] x26: 000000000000001d x27: ffffff8008f12000 
[   21.845025] x27: ffffff8008f12000 
[   21.845027] x26: 000000000000001d x26: 000000000000001d x25: 0000000000000123 
[   21.845030] x24: 0000000000000001 x25: 0000000000000123 
[   21.845031] x25: 0000000000000123 
[   21.845032] x24: 0000000000000001 x24: 0000000000000001 x23: ffffff800a42596c 
[   21.845035] x22: 000000000000003d x23: ffffff800a42596c 
[   21.845036] x23: ffffff800a42596c 
[   21.845037] x22: 000000000000003c x22: 000000000000003a x21: ffffff800a211830 
[   21.845040] x20: 0000000000000000 x21: ffffff800a211830 
[   21.845041] x21: ffffff800a211830 
[   21.845043] x20: 0000000000000000 x20: 0000000000000000 x19: 0000000000000000 
[   21.845045] x18: 000000000000003f x19: 0000000000000000 
[   21.845046] x19: 0000000000000000 
[   21.845048] x18: 0000000000000008 x18: 000000000000003f x17: 0000000000000002 
[   21.845050] x16: 0000000000000000 x17: 0000000000000001 
[   21.845051] x17: 0000000000000002 
[   21.845052] x16: 0000000000000000 x16: 0000000000000000 x15: 0000000000005d28 
[   21.845055] x14: 000000000000010e x15: 000000000000bc00 
[   21.845056] x15: 0000000000023000 
[   21.845058] x14: 000db58580000000 x14: 000cdfe600000000 x13: 00000000000049c5 
[   21.845060] x12: 0000000000002f97 x13: 000000000002ccc7 
[   21.845061] x13: 0000000000019273 
[   21.845062] x12: 0000000000020f2c x12: 000000000001d54a x11: 000000000000020f 
[   21.845065] x10: 0000000000000a20 x11: 00000000000003f7 
[   21.845066] x11: 00000000000001f3 
[   21.845068] x10: 0000000000000a20 x10: 0000000000000a20 x9 : ffffffc1d0013ae0 
[   21.845070] x8 : ffffffc1eb504280 x9 : ffffffc1d000fae0 
[   21.845072] x9 : ffffffc1d0007ae0 
[   21.845073] x8 : ffffffc1eb507a80 x8 : ffffffc1eb0dc280 x7 : 0000000000000400 
[   21.845075] x6 : 0000000000008400 x7 : 0000000000000400 
[   21.845076] x7 : 0000000000000400 
[   21.845077] x6 : 000000000000ec00 x6 : 0000000000011c00 x5 : dead000000000100 
[   21.845080] x4 : dead000000000200 x5 : dead000000000100 
[   21.845081] x5 : dead000000000100 
[   21.845082] x4 : dead000000000200 x4 : dead000000000200 x3 : ffffffc1d0007cb0 
[   21.845085] x2 : ffffff800a211820 x3 : ffffffc1d0003cb0 
[   21.845086] x3 : ffffffc1d000bcb0 
[   21.845087] x2 : ffffffc1d000bcb0 x2 : ffffff800a211820 x1 : ffffffc1eb503800 
[   21.845090] x0 : 0000000000000008 x1 : ffffffc1eb507000 
[   21.845091] x1 : ffffffc1eb0db800 
[   21.845092] x0 : 0000000000000008 x0 : 0000000000000008 
[   21.845094] ---[ end trace 789c5c9be8052c5f ]---
[   21.845094] 
[   21.845095] 
[   21.845095] ---[ end trace 789c5c9be8052c60 ]---
```

madisongh, 2019-08-06:
I had to use `ssh -X -Y` to enable trusted X11 forwarding over the ssh connection.

dagmcr, 2019-08-06:
I also tried that option but still the same issue for `glxinfo`.

Here the logs with verbose option when I do ssh.

> Note: as you can see in the logs, `pcmanfm` works nice but not GLX applications.
> Note2: I'm doing this tests from 2 remote desktops: Debian testing and Ubuntu 16.04.5.

```
pd@pdws:~$ ssh -X -Y -v root@jetson-tx2.local
OpenSSH_7.2p2 Ubuntu-4ubuntu2.8, OpenSSL 1.0.2g  1 Mar 2016
debug1: Reading configuration data /etc/ssh/ssh_config
debug1: /etc/ssh/ssh_config line 19: Applying options for *
debug1: Connecting to jetson-tx2.local [192.168.2.86] port 22.
debug1: Connection established.
debug1: identity file /home/pd/.ssh/id_rsa type 1
debug1: key_load_public: No such file or directory
debug1: identity file /home/pd/.ssh/id_rsa-cert type -1
debug1: key_load_public: No such file or directory
debug1: identity file /home/pd/.ssh/id_dsa type -1
debug1: key_load_public: No such file or directory
debug1: identity file /home/pd/.ssh/id_dsa-cert type -1
debug1: key_load_public: No such file or directory
debug1: identity file /home/pd/.ssh/id_ecdsa type -1
debug1: key_load_public: No such file or directory
debug1: identity file /home/pd/.ssh/id_ecdsa-cert type -1
debug1: key_load_public: No such file or directory
debug1: identity file /home/pd/.ssh/id_ed25519 type -1
debug1: key_load_public: No such file or directory
debug1: identity file /home/pd/.ssh/id_ed25519-cert type -1
debug1: Enabling compatibility mode for protocol 2.0
debug1: Local version string SSH-2.0-OpenSSH_7.2p2 Ubuntu-4ubuntu2.8
debug1: Remote protocol version 2.0, remote software version dropbear_2019.78
debug1: no match: dropbear_2019.78
debug1: Authenticating to jetson-tx2.local:22 as 'root'
debug1: SSH2_MSG_KEXINIT sent
debug1: SSH2_MSG_KEXINIT received
debug1: kex: algorithm: curve25519-sha256@libssh.org
debug1: kex: host key algorithm: ssh-rsa
debug1: kex: server->client cipher: aes128-ctr MAC: hmac-sha2-256 compression: none
debug1: kex: client->server cipher: aes128-ctr MAC: hmac-sha2-256 compression: none
debug1: expecting SSH2_MSG_KEX_ECDH_REPLY
debug1: Server host key: ssh-rsa SHA256:mLQjdgxBV77ZDKnQ1gVkeXJWyETUpmef+yRWdoQt/oc
debug1: Host 'jetson-tx2.local' is known and matches the RSA host key.
debug1: Found key in /home/pd/.ssh/known_hosts:11
debug1: rekey after 4294967296 blocks
debug1: SSH2_MSG_NEWKEYS sent
debug1: expecting SSH2_MSG_NEWKEYS
debug1: SSH2_MSG_NEWKEYS received
debug1: rekey after 4294967296 blocks
debug1: SSH2_MSG_SERVICE_ACCEPT received
debug1: Authentication succeeded (none).
Authenticated to jetson-tx2.local ([192.168.2.86]:22).
debug1: channel 0: new [client-session]
debug1: Entering interactive session.
debug1: pledge: exec
debug1: Requesting X11 forwarding with authentication spoofing.
debug1: Sending environment.
debug1: Sending env LC_PAPER = es_ES.UTF-8
debug1: Sending env LC_ADDRESS = es_ES.UTF-8
debug1: Sending env LC_MONETARY = es_ES.UTF-8
debug1: Sending env LC_NUMERIC = es_ES.UTF-8
debug1: Sending env LC_TELEPHONE = es_ES.UTF-8
debug1: Sending env LC_IDENTIFICATION = es_ES.UTF-8
debug1: Sending env LANG = en_US.UTF-8
debug1: Sending env LC_MEASUREMENT = es_ES.UTF-8
debug1: Sending env LC_TIME = es_ES.UTF-8
debug1: Sending env LC_NAME = es_ES.UTF-8
root@jetson-tx2:~# pcmanfm 
debug1: client_input_channel_open: ctype x11 rchan 1 win 24576 max 32759
debug1: client_request_x11: request from 127.0.0.1 30360
debug1: channel 1: new [x11]
debug1: confirm x11
debug1: client_input_channel_open: ctype x11 rchan 2 win 24576 max 32759
debug1: client_request_x11: request from 127.0.0.1 31384
debug1: channel 2: new [x11]
debug1: confirm x11
debug1: channel 2: FORCE input drain
debug1: channel 2: free: x11, nchannels 3
Gtk-Message: 16:10:36.165: Failed to load module "canberra-gtk-module"
Gtk-Message: 16:10:36.376: Failed to load module "canberra-gtk-module"

** (pcmanfm:3161): WARNING **: 16:10:36.378: The directory '~/Templates' doesn't exist, ignoring it

(pcmanfm:3161): Gtk-WARNING **: 16:10:37.375: Theme parsing error: <data>:2:27: The style property GtkWidget:focus-padding is deprecated and shouldn't be used anymore. It will be removed in a future version

(pcmanfm:3161): Gtk-WARNING **: 16:10:37.375: Theme parsing error: <data>:3:30: The style property GtkWidget:focus-line-width is deprecated and shouldn't be used anymore. It will be removed in a future version

(pcmanfm:3161): Gtk-WARNING **: 16:10:37.387: Could not load a pixbuf from /org/gtk/libgtk/theme/Adwaita/assets/bullet-symbolic.svg.
This may indicate that pixbuf loaders or the mime database could not be found.

(pcmanfm:3161): Gtk-WARNING **: 16:10:42.002: gtk_menu_detach(): menu is not attached
debug1: channel 1: FORCE input drain
debug1: channel 1: free: x11, nchannels 2
root@jetson-tx2:~# glxinfo 
debug1: client_input_channel_open: ctype x11 rchan 1 win 24576 max 32759
debug1: client_request_x11: request from 127.0.0.1 32408
debug1: channel 1: new [x11]
debug1: confirm x11
name of display: localhost:10.0
X Error of failed request:  BadValue (integer parameter out of range for operation)
  Major opcode of failed request:  154 (GLX)
debug1: channel 1: FORCE input drain
debug1: channel 1: free: x11, nchannels 2
  Minor opcode of failed request:  24 (X_GLXCreateNewContext)
  Value in failed request:  0x0
  Serial number of failed request:  39
  Current serial number in output stream:  40
```
I will try to look more in detail what's going on.

dagmcr, 2019-08-06:
I also tried that option but still the same issue for `glxinfo`.

Here the logs with verbose option when I do ssh.

> Note: as you can see in the logs, `pcmanfm` works nice but not GLX applications.
> Note2: I'm doing this tests from 2 remote desktops: Debian testing and Ubuntu 16.04.5.

```
$ ssh -X -Y -v root@jetson-tx2.local
OpenSSH_7.2p2 Ubuntu-4ubuntu2.8, OpenSSL 1.0.2g  1 Mar 2016
debug1: Reading configuration data /etc/ssh/ssh_config
debug1: Authentication succeeded (none).
Authenticated to jetson-tx2.local ([192.168.2.86]:22).
root@jetson-tx2:~# pcmanfm 
debug1: client_input_channel_open: ctype x11 rchan 1 win 24576 max 32759
debug1: client_request_x11: request from 127.0.0.1 30360
debug1: channel 1: new [x11]
debug1: confirm x11
debug1: client_input_channel_open: ctype x11 rchan 2 win 24576 max 32759
debug1: client_request_x11: request from 127.0.0.1 31384
debug1: channel 2: new [x11]
debug1: confirm x11
debug1: channel 2: FORCE input drain
debug1: channel 2: free: x11, nchannels 3
Gtk-Message: 16:10:36.165: Failed to load module "canberra-gtk-module"
Gtk-Message: 16:10:36.376: Failed to load module "canberra-gtk-module"

** (pcmanfm:3161): WARNING **: 16:10:36.378: The directory '~/Templates' doesn't exist, ignoring it

(pcmanfm:3161): Gtk-WARNING **: 16:10:37.375: Theme parsing error: <data>:2:27: The style property GtkWidget:focus-padding is deprecated and shouldn't be used anymore. It will be removed in a future version

(pcmanfm:3161): Gtk-WARNING **: 16:10:37.375: Theme parsing error: <data>:3:30: The style property GtkWidget:focus-line-width is deprecated and shouldn't be used anymore. It will be removed in a future version

(pcmanfm:3161): Gtk-WARNING **: 16:10:37.387: Could not load a pixbuf from /org/gtk/libgtk/theme/Adwaita/assets/bullet-symbolic.svg.
This may indicate that pixbuf loaders or the mime database could not be found.

(pcmanfm:3161): Gtk-WARNING **: 16:10:42.002: gtk_menu_detach(): menu is not attached
debug1: channel 1: FORCE input drain
debug1: channel 1: free: x11, nchannels 2
root@jetson-tx2:~# glxinfo 
debug1: client_input_channel_open: ctype x11 rchan 1 win 24576 max 32759
debug1: client_request_x11: request from 127.0.0.1 32408
debug1: channel 1: new [x11]
debug1: confirm x11
name of display: localhost:10.0
X Error of failed request:  BadValue (integer parameter out of range for operation)
  Major opcode of failed request:  154 (GLX)
debug1: channel 1: FORCE input drain
debug1: channel 1: free: x11, nchannels 2
  Minor opcode of failed request:  24 (X_GLXCreateNewContext)
  Value in failed request:  0x0
  Serial number of failed request:  39
  Current serial number in output stream:  40
```
I will try to look more in detail what's going on.

Edited: I've deleted part of the logs.

madisongh, 2019-08-06:
Wait, so you're connecting from your Ubuntu system to the Jetson, and trying to use GLX to display stuff on the Ubuntu system?   Then you need to make sure your Ubuntu system's Xorg server has the +iglx argument, since that's where GLX services are going to be requested.  Nothing to do with GLX in the Xorg server on the Jetson.

I was testing the other way around, connecting from the Jetson to my Linux desktop and running glxinfo there.

dagmcr, 2019-08-06:
It worked!
Thanks a lot for the help and sorry for the confusion.

closed 2019-08-06 by dagmcr