OE4T/meta-tegra #159: Configure remote VNC on Jetson?

mwcoop, 2019-08-02:
Hello,

I'm trying to get a VNC configuration working on Jetson Nano/TX2/Xavier. So far I have tried adding tigervnc and x11vnc to my Yocto image. Everything builds fine. However, both programs complain about not finding "xauth."

My image is core-image-weston + other stuff. Any ideas/tips/success stories getting VNC added to a meta-tegra image?

madisongh, 2019-08-02:
I don't think VNC is an option if you're using Weston, from what I've read.  You could try using FreeRDP (recipe in meta-oe), and adding `--enable-rdp-compositor` in your weston build to enable it.

mwcoop, 2019-08-02:
Ahh. Excellent information. Thanks.

On a somewhat related note. I got Weston+FreeRDP working. However, on my Jetson Xavier, I only get a black screen via HDMI. Does that need something specific? Can I have them working simultaneously?

madisongh, 2019-08-03:
I suspect you need to have two weston services - one for RDP and one for the real display device. You may have to assign them different seat IDs as well, if weston can't figure that out for itself.

closed 2019-08-04 by mwcoop