OE4T/meta-tegra #166: cuda-binaries-native

MiBLE, 2019-08-11:
Hello,
During bitbake I got this error:

ERROR: cuda-binaries-native-10.0.326-1-r0 do_fetch: Fetcher failure for URL: 'file:///home/mixon/Downloads/nvidia/sdkm_downloads/cuda-repo-ubuntu1804-10-0-local-10.0.326-410.108_1.0-1_amd64.deb'. Unable to fetch URL from any source.

nVidia SDK_Manger download cuda-repo-ubuntu1604-10-0-local-10.0.326-410.108_1.0-1_amd64.deb as I use Ubuntu 16.04.3 LTS.

MiBLE, 2019-08-11:
I changed  cuda-binaries-native_10.0.326-1.bb with this:

`SRC_URI = "${NVIDIA_DEVNET_MIRROR}/cuda-repo-ubuntu1604-10-0-local-10.0.326-410.108_1.0-1_amd64.deb"
SRC_URI[md5sum] = "07ff157a7980aed07d0eafeebe86d64a"
SRC_URI[sha256sum] = `"b476fa6a5b702ac55977a9d9cad4d1761a850fc3fd76bd7caf04eb4f15b288a3"``

but I am not sure about LIC_FILES_CHKSUM.
What should it be?

madisongh, 2019-08-12:
If the checksum of the license text is different, you'll get an error during the build about the mismatch. I'd expect it to be the same, though.

I'll have to figure out some way to handle the 1604 vs 1804 naming difference in the recipes.

madisongh, 2019-08-20:
This should be fixed in master and the new warrior-l4t-r32.2 branch.  Just add
```
CUDA_BINARIES_NATIVE = "cuda-binaries-ubuntu1604-native"
```
to your build configuration (e.g., in `$BUILDDIR/conf/local.conf`) if you're using `sdkmanager` on Ubuntu 16.04.

MiBLE, 2019-08-24:
WARNING: cuda-binaries-ubuntu1604-native-10.0.326-1-r0 do_fetch: Failed to fetch URL file:///home/mixon/Downloads/nvidia/sdkm_downloads/cuda-repo-ubuntu1804-10-0-local-10.0.326-410.108_1.0-1_amd64.deb, attempting MIRRORS if available
ERROR: cuda-binaries-ubuntu1604-native-10.0.326-1-r0 do_fetch: Fetcher failure: Unable to find file file:///home/mixon/Downloads/nvidia/sdkm_downloads/cuda-repo-ubuntu1804-10-0-local-10.0.326-410.108_1.0-1_amd64.deb anywhere. The paths that were searched were:
    /home/mixon/projects/jetson-warrior/layers/meta-tegra/recipes-devtools/cuda/cuda-binaries-ubuntu1604-10.0.326-1/poky
    /home/mixon/projects/jetson-warrior/layers/meta-tegra/recipes-devtools/cuda/cuda-binaries-ubuntu1604/poky
    /home/mixon/projects/jetson-warrior/layers/meta-tegra/recipes-devtools/cuda/files/poky
    /home/mixon/projects/jetson-warrior/layers/meta-tegra/recipes-devtools/cuda/cuda-binaries-ubuntu1604-10.0.326-1/
    /home/mixon/projects/jetson-warrior/layers/meta-tegra/recipes-devtools/cuda/cuda-binaries-ubuntu1604/
    /home/mixon/projects/jetson-warrior/layers/meta-tegra/recipes-devtools/cuda/files/
    /home/mixon/projects/jetson-warrior/layers/meta-tegra/recipes-devtools/cuda/cuda-binaries-ubuntu1604-10.0.326-1/x86-64
    /home/mixon/projects/jetson-warrior/layers/meta-tegra/recipes-devtools/cuda/cuda-binaries-ubuntu1604/x86-64
    /home/mixon/projects/jetson-warrior/layers/meta-tegra/recipes-devtools/cuda/files/x86-64
    /home/mixon/projects/jetson-warrior/layers/meta-tegra/recipes-devtools/cuda/cuda-binaries-ubuntu1604-10.0.326-1/
    /home/mixon/projects/jetson-warrior/layers/meta-tegra/recipes-devtools/cuda/cuda-binaries-ubuntu1604/
    /home/mixon/projects/jetson-warrior/layers/meta-tegra/recipes-devtools/cuda/files/
    /home/mixon/projects/jetson-warrior/build/downloads
ERROR: cuda-binaries-ubuntu1604-native-10.0.326-1-r0 do_fetch: Fetcher failure for URL: 'file:///home/mixon/Downloads/nvidia/sdkm_downloads/cuda-repo-ubuntu1804-10-0-local-10.0.326-410.108_1.0-1_amd64.deb'. Unable to fetch URL from any source.
ERROR: cuda-binaries-ubuntu1604-native-10.0.326-1-r0 do_fetch: 
ERROR: cuda-binaries-ubuntu1604-native-10.0.326-1-r0 do_fetch: Function failed: base_do_fetch
ERROR: Logfile of failure stored in: /home/mixon/projects/jetson-warrior/build/tmp/work-shared/cuda-binaries-native-10.0.326-1-r0/temp/log.do_fetch.2902
ERROR: Task (/home/mixon/projects/jetson-warrior/layers/meta-tegra/recipes-devtools/cuda/cuda-binaries-ubuntu1604-native_10.0.326-1.bb:do_fetch) failed with exit code '1'

madisongh, 2019-08-24:
Sorry, I don't know how I missed that.  Will post a fix shortly.

madisongh, 2019-08-24:
OK, should be fixed now.  Update and give it another try.

closed 2019-08-24 by MiBLE

MiBLE, 2019-08-24:
Now It is Ok. Thank you.

beyondli, 2020-07-16:
@madisongh 
I cannot find any md5 files in my download, I installed with jectpack4.4,
any help is very appreciated!

[image]

madisongh, 2020-07-16:
@beyondli From the screenshot you initially posted, it looks like the host-side CUDA tools deb package (`cuda-repo-ubuntu1804-10-2-local-10.2.89-440.40_1.0-1_amd64.deb` or `cuda-repo-ubuntu1604-10-2-local-10.2.89-440.40_1.0-1_amd64.deb`) is missing from your downloads area.  You may need to fire up the SDK Manager and download it (again).

And note that with the L4T R32.4.3/JetPack 4.4 branches, the variable to set for selecting the ubuntu1604 package has change from `CUDA_BINARIES_NATIVE` to just `CUDA_BINARIES`.

beyondli, 2020-07-16:
@madisongh 
thanks for reply,   

env of mine:
     test on xavier， host：ubuntu16.04

you are right ,but from download JP4.4, file name is changed to cuda-repo-l4t-10-2-local-10.2.89_1.0-1_arm64.deb
so what all I modify are here
1:cuda-binaries-ubuntu1604_10.2.89-1.bb 
    SRC_URI = "${NVIDIA_DEVNET_MIRROR}/cuda-repo-l4t-10-2-local-10.2.89_1.0-1_arm64.deb"
    rewrite SRC_URI[sha256sum]
2:nvidia_devnet_downloads.bbclass
   set  NVIDIA_DEVNET_MIRROR ??= "file:///homed/jeff/Downloads/nvidia/sdkm_downloads" to correct path
3:conf/local.conf
 CUDA_BINARIES = "cuda-binaries-ubuntu1604"
 MACHINE ?= "jetson-xavier"

from these modified ,no error occurs ,but no img file can create ,seems no NV SDK can be readed

anyway, what surprise me is if I set NVIDIA_DEVNET_MIRROR to a wrong path ,it still no error when I run :bitbake core-image-minimal

what else can I do ? any help is very appreciated!

madisongh, 2020-07-16:
> from download JP4.4, file name is changed to cuda-repo-l4t-10-2-local-10.2.89_1.0-1_arm64.deb

No, that's the CUDA repo for the target CUDA binaries, not the host (x86-64) ones.  As I mentioned earlier, you should re-run the SDK Manager and download again.  Watch the download progress and the logs in SDK Manager to verify that it downloaded the ubuntu1604 deb file I mentioned above into the same directory you have set for NVIDIA_DEVNET_MIRROR.

1. On the STEP 01 page, make sure there is a green check mark next to "Host Machine" (it should be there by default; clicking on the "Host Machine" box will toggle it off/on).  You can de-select Target Hardware for this download (click the "Target Hardware" box to remove the green check mark).
2. Click "CONTINUE TO STEP 02"
3. You should see three lines under "HOST COMPONENTS" on the STEP 02 page - CUDA, Computer Vision, and Development Tools. Check the "I accept" box **and** the "Download now. Install later." box at the bottom of the page.
4. Click "CONTINUE TO STEP 03"
5. On the STEP 04 page, click on the TERMINAL tab to see the log output of the download process.  You should see a line like this:
        ```13:43:56 INFO: CUDA on Host: verifying checksum of /home/ubuntu/Downloads/nvidia/sdkm_downloads/cuda-repo-ubuntu1604-10-2-local-10.2.89-440.40_1.0-1_amd64.deb```
    if the download completed successfully.
6. Click "FINISH AND EXIT", then check that the deb package is in your sdkm_downloads directory.

Don't change the recipes or the bbclass.  Just set `NVIDIA_DEVNET_MIRROR = "file://..."` in your local.conf.

beyondli, 2020-07-16:
@madisongh 
thanks for quick reply
I have roll back code of meta-tegra, nothing modified,just modify here
1:local.conf
   MACHINE ?= "jetson-xavier"
   CUDA_BINARIES = "cuda-binaries-ubuntu1604"
   NVIDIA_DEVNET_MIRROR ??= "file://home/jeff/Downloads/nvidia/sdkm_downloads"
   path set for "file:///home/jeff/Downloads/nvidia/sdkm_downloads" ,file:/// here is same result with file://
2:add meta-tegra path in bblayers.conf
 
  after bitbake core-image-minimal
it showed, but no img file can be created

[image]

followed is SDK, host cuda has been downloaded
[image]

what can I do ? any help is very appreciated!

madisongh, 2020-07-16:
> it showed, but no img file can be created

So you the tegraflash package didn't created at `$BUILDDIR/tmp/deploy/images/jetson-xavier/core-image-minimal.tegraflash.tar.gz` ?  Did you get errors during the build?  If so, can you attach the logs?

beyondli, 2020-07-17:
[image]
nothing created now...........

beyondli, 2020-07-17:
I delete the deploy DIR, and try bitbake core-image-minimal, no deploy DIR can be created
[image]

madisongh, 2020-07-17:
Strange.  I'd try `rm -rf tmp` just to clear out anything in the build tree from your prior failed builds that could be getting in the way, then re-run the core-image-minimal build and see if that gets you back on track.

beyondli, 2020-07-17:
ok.....,I will try ,but actually I like keep it .build a totally new one is such a tedious work , waiting and waiting
anyway, thank you for your reply!

beyondli, 2020-07-17:
@madisongh , I rebuild ,and this is result, tar file 👍 core-image-minimal-jetson-xavier-20200717151218.tegraflash.tar.gz only 30M， is it still wrong?since cuda &cudnn will be in img file ,I think this file should be big >1G, no error occurs during building

[image]

madisongh, 2020-07-17:
core-image-minimal is, as the name implies, a minimal image that doesn't include much functionality.  It should at least boot, and is good for an initial test.

If that works, the next step would be to build a fuller image, such as core-image-base (for a basic command line interface) or core-image-sato (an X11-based environment) and set `CORE_IMAGE_EXTRA_INSTALL` in your local.conf file to add in other packages you want to have included in your build.

beyondli, 2020-07-17:
I have two questions:
1:how to test this tar.gz?just copy to Tegra bootloader DIR and flash?
2: I have set the wrong cuda path like this(define it in local.conf) : NVIDIA_DEVNET_MIRROR ??= "file://home/jeff/Downloads/nvidia/sdkm_downloads_error_path", but it is still can build without any errors and can create img file. so I think any Tegra lib is not build in it?

madisongh, 2020-07-17:
[This wiki page](https://github.com/madisongh/meta-tegra/wiki/Flashing-the-Jetson-Dev-Kit) talks about flashing.  Basically, you just need to unpack the tarball into an empty directory, hook up a USB cable (and a serial console, if you can), and run the doflash.sh script provided in the tarball (using sudo, if needed).

NVIDIA_DEVNET_MIRROR will only be used to locate packages when needed - and for the branches based off L4T R32.4.x, it's only needed for the host-side CUDA tools (`-native` and `nativesdk-` CUDA packages when you're building for an x86_64 host).  The rest of the packages needed for the Jetson target are available for network download during the build. So if you don't have it set, or have it set incorrectly, you won't necessarily notice that until you start building things that need `nvcc` or other host-side CUDA packages.

beyondli, 2020-07-17:
thanks ,I know how to flash it 
for NVIDIA_DEVNET_MIRROR for host is a little bit strange ,for my point I thinks it just use for locating deb file of jetson which confused me a long time.
and why jetson packages will download from network? since all of them have been downloaded to sdkm_downloads ,it will be faster to read directly from local

madisongh, 2020-07-17:
> NVIDIA_DEVNET_MIRROR for host is a little bit strange

It's only needed for the deb packages that require a developer network login to access; that's something that can't be set up directly within the bitbake builds. As of L4T R32.4.x/JetPack 4.4, it's only the CUDA host tools that are locked behind the developer network login.

> why jetson packages will download from network? 

Mainly because users find it more convenient, particularly if you're setting up a build environment to be shared with multiple developers on multiple machines.

beyondli, 2020-07-18:
thanks! got
I have tried to flash this minimal , it can boot, but seems need login name?
1)what the login name of this version, I waiting for a long time ,but it stuck here like below screenshot.
2)flash time is too short ,maybe only 30 seconds, seems NO nvidia SDK is flashed in it, is it wrong?
[image]

any help is very appreciated!

madisongh, 2020-07-18:
Right, it's just a bare-minimum OS - no JetPack components, just the basic kernel, bootloaders, and a small set of command line utilities.  You should be able to log in as root with no password, unless you configured otherwise in your local.conf file.

beyondli, 2020-07-18:
thanks! I log in 
[image]
but my xavier is 32G, is this partition right？max available is just 24.7G of mmcblkop1, and there is a strange "none" partition that 15.2G, seems this partiton is not correct

beyondli, 2020-07-18:
@madisongh 
I have tried sato image, but after run doflash.sh it showed:

ERR: could not retrieve chip ID
[image]

I have not change any code of meta-tegra code ,what is meaning?
thanks for reply

madisongh, 2020-07-18:
The flash size is 32G, but some of the flash is occupied by other partitions.  The rootfs partition size you see should be comparable to what you would get when using L4T/JetPack.

As for /dev - that's a kernel in-memory filesystem, which I believe defaults to a size of about half of available RAM.  So on a Xavier with 32GiB of DRAM, a bit less than 16GiB would be right.  On my older 16GiB Xavier, it's around 8GiB.

beyondli, 2020-07-18:
thanks! understand
I check SDK, only cuda can find , no nvcc and cudnn can be find, seems cudnn is  set in recipes, where are they?
[image]

madisongh, 2020-07-18:
There are recipes for them, you just have to include the ones you want in your build.  You could add, for example,
```
CORE_IMAGE_EXTRA_INSTALL = "cuda-toolkit cudnn"
```
to your local.conf file.

beyondli, 2020-07-19:
@madisongh ,I have installed cuda & cudnn, sys boot OK
thank you very much!
and appreciate your excellent work about meta-tegra which means you are very very familiar with nvidia jetson's hardware. 
how many years you played on Jetson? hope one day I could familiar with Jetson like you ,admire this great work!