OE4T/meta-tegra #168: malformedUrl

arrow53, 2019-08-13:
@madisongh I'm having trouble with this new pre-download thing. I'm getting these sorts of errors
```
Exception: bb.fetch2.MalformedUrl: The URL: '/home/arrow53/Downloads/nvidia/sdkm_downloads/libvisionworks-tracking-repo_0.88.2_arm64.deb' is invalid and cannot be interpreted
```
The file definitely exists. And it does seem to find it. Is there something wrong with the path setup in my local.conf?

```
NVIDIA_DEVNET_MIRROR = "/home/arrow53/Downloads/nvidia/sdkm_downloads"
```

kekiefer, 2019-08-13:
You should format the variable like this:

```
NVIDIA_DEVNET_MIRROR = "file:///home/arrow53/Downloads/nvidia/sdkm_downloads"
```

arrow53, 2019-08-14:
@kekiefer thanks. dumb mistake.

closed 2019-08-14 by arrow53