OE4T/meta-tegra #169: which version of gcc to use in warrior

arrow53, 2019-08-14:
It seems that from #93 that gcc 8 is supported?

Is that true or is gcc 7 still required?

If gcc 7 is required does anyone have layer that they could share that provides this? I naively just tried using poky/thud and got a bunch of errors.

madisongh, 2019-08-15:
To use CUDA, you'll still need gcc 7.  Otherwise, gcc 8 should work.  I don't test it regularly, though.

arrow53, 2019-08-15:
thanks. using gcc7 is rough. I tried using linaro as the toolchain and using poky's in thud and neither worked. If anyone is reading this and has a gcc7 recipe that works with warrior please let me know.

@madisongh you can close this if you'd like as you've answered the real question. If/when I figure it out I'll comment close too.

madisongh, 2019-08-16:
The older recipes did need some tweaking for warrior.  See https://github.com/madisongh/test-distro/tree/warrior/layers/meta-testdistro/recipes-devtools/gcc7

arrow53, 2019-08-16:
thanks @madisongh. I included your meta-testdistro and saw the same issues I was having before. So, I dug into the error and it appears to be the same as this

https://stackoverflow.com/questions/56481980/yocto-for-nvidia-jetson-fails-because-of-gcc-7-cannot-compute-suffix-of-object

The solution there was to set DEBUG_PREFIX_MAP to nothing so that gcc7 ignores the `-fmacro-debug-prefix`

setting this to "" in my local.conf lets gcc7 build. But, then busybox wouldn't build. I tried to track down that error and it's some sed thing I can't figure out.

Does any of this sound familiar?

dwalkes, 2019-08-16:
@arrow53 I went through a similar path getting linaro-gcc7 to work.  The DEBUG_PREFIX_MAP that worked for me was:
```
DEBUG_PREFIX_MAP = "-fdebug-prefix-map=${WORKDIR}=/usr/src/debug/${PN}/${EXTENDPE}${PV}-${PR} \
                    -fdebug-prefix-map=${STAGING_DIR_HOST}= \
                    -fdebug-prefix-map=${STAGING_DIR_NATIVE}= \
                    "
```
which is [the default from bitbake.conf ](https://git.yoctoproject.org/cgit/cgit.cgi/poky/tree/meta/conf/bitbake.conf?h=warrior#n613) minus the -fmacro-prefix-map.  There's probably a better way to do this.

arrow53, 2019-08-16:
thanks @dwalkes that worked great! I was struggling to find where that came from so thanks for the link.

closed 2019-08-16 by arrow53

dwalkes, 2019-10-06:
@arrow53 FYI I've added a [wiki page on linaro gcc7 support](https://github.com/madisongh/meta-tegra/wiki/Using-linaro-gcc7-for-CUDA-support) per discussion in #185.  If you have any additional suggestions or recommendations based on your experience please add there.

arrow53, 2019-10-07:
thanks @dwalkes !