OE4T/meta-tegra #172: Thud and GCC 6

lfdmn, 2019-08-30:
Hi,

I'm trying to build meta-tegra on thud branch

Would you have recommendation on how to add old GCC6 compiler to poky thud branch?

Thanks

madisongh, 2019-08-30:
You could add the [meta-linaro-toolchain](http://git.linaro.org/openembedded/meta-linaro.git/) layer to your builds and use their 6.x toolchain. Another option is to forward-port the recipes from OE-Core into your own layer; for example, see [here](https://github.com/madisongh/test-distro/tree/thud/layers/meta-testdistro/recipes-devtools/gcc6) - that's what I use in my test builds.

lfdmn, 2019-08-30:
I had tried to use linaro but when starting to build Qt, configure fails telling C compiler isn't working. 

I tried using your test-distro by adding ${BSPDIR}/sources/meta-test-distro/layers/meta-testdistro to bblayer.conf and GCCVERSION = "6.%" to local.conf

GCC6 seems to be detected for everything except gcc-sanitizers, although gcc-sanitizers_6.4.bb is in your layer.
```
NOTE: preferred version 6.% of gcc-sanitizers not available (for item gcc-sanitizers)
NOTE: versions of gcc-sanitizers available: 7.3.0 8.2.0
NOTE: preferred version 6.% of gcc-sanitizers not available (for item libtsan)
NOTE: versions of gcc-sanitizers available: 7.3.0 8.2.0
NOTE: preferred version 6.% of gcc-sanitizers not available (for item libasan)
NOTE: versions of gcc-sanitizers available: 7.3.0 8.2.0
NOTE: preferred version 6.% of gcc-sanitizers not available (for item libubsan)
NOTE: versions of gcc-sanitizers available: 7.3.0 8.2.0
NOTE: preferred version 6.% of gcc-sanitizers not available (for item liblsan)
NOTE: versions of gcc-sanitizers available: 7.3.0 8.2.0
```
Am I missing something?

lfdmn, 2019-08-30:
I suppose the issue comes from gcc-sanitizers.inc in gcc5 and gcc6 folders
```
# MIPS, aarch64, and SPARC are broken.
COMPATIBLE_HOST = '(x86_64|i.86|powerpc|arm).*-linux'
```
issue being Qt5.13 requires it.

madisongh, 2019-09-01:
You might be able to just enable it - see [this patch thread](http://lists.openembedded.org/pipermail/openembedded-core/2018-April/268067.html) that was posted last year.  Doesn't look like the patch made it into rocko (from which I got the gcc6 recipes, IIRC), but it implies that the sanitizers should work on aarch64 with gcc 6.

lfdmn, 2019-09-11:
@madisongh thanks for the patch.

I managed to build by making my own distro layer containing gcc-6.4 and glibc-2.27.

In sumo branch glibc-2.28 deprecated and removed some API. Due to this re-enabling the aarch64 support for gcc-sanitizer for gcc-6.4 was not enough.

Downgrading the glibc to 2.27 was necessary using GLIBCVERSION = “2.27%” in local.conf or distro config file.

Starting from sumo branch a new libnsl2 package was introduced and added back the removed functionality + some improvements. glibc and libnsl2 both installing the same headers, the build failed for this too.

So I had to remove the libnsl2 dependency for all packages depending on it with **DEPENDS_remove = "libnsl2"**. In my case:
* tcp-wrappers
* libnss-nis
* autofs
* postfix
* nis
* stunnel
* postgresql

closed 2019-09-11 by lfdmn

madisongh, 2019-09-11:
Wow, that sounds complicated.  Thanks for posting your solution.