OE4T/meta-tegra #173: Using docker on Jetson TX2

hesmar, 2019-09-02:
Hi,
I want to run docker on the Jetson TX2. So, I added the meta-virtualization layer and appended the following to my local.conf:
```
DISTRO_FEATURES_append = " virtualization"
CORE_IMAGE_EXTRA_INSTALL += "docker"
```
Unfortunately, starting docker fails with the following output:
```
Sep 02 08:40:44 jetson-tx2 systemd[1]: Starting Docker Application Container Engine...
Sep 02 08:40:44 jetson-tx2 systemd-timesyncd[3999]: Synchronized to time server for the first time 216.239.35.8:123 (time3.google.com).
Sep 02 08:40:44 jetson-tx2 dockerd[4337]: time="2019-09-02T08:40:44.600377024Z" level=info msg="Starting up"
Sep 02 08:40:44 jetson-tx2 dockerd[4337]: time="2019-09-02T08:40:44.641892768Z" level=info msg="libcontainerd: started new containerd process" pid=4373
Sep 02 08:40:44 jetson-tx2 dockerd[4337]: time="2019-09-02T08:40:44.642049632Z" level=info msg="parsed scheme: \"unix\"" module=grpc
Sep 02 08:40:44 jetson-tx2 dockerd[4337]: time="2019-09-02T08:40:44.642078848Z" level=info msg="scheme \"unix\" not registered, fallback to default scheme" module=grpc
Sep 02 08:40:44 jetson-tx2 dockerd[4337]: time="2019-09-02T08:40:44.642132768Z" level=info msg="ccResolverWrapper: sending update to cc: {[{unix:///var/run/docker/containerd/containerd.sock 0  <nil>}] }" module=grpc
Sep 02 08:40:44 jetson-tx2 dockerd[4337]: time="2019-09-02T08:40:44.642175264Z" level=info msg="ClientConn switching balancer to \"pick_first\"" module=grpc
Sep 02 08:40:44 jetson-tx2 dockerd[4337]: time="2019-09-02T08:40:44.645068576Z" level=info msg="pickfirstBalancer: HandleSubConnStateChange: 0x40007bde90, CONNECTING" module=grpc
Sep 02 08:40:44 jetson-tx2 dockerd[4337]: time="2019-09-02T08:40:44.856548736Z" level=info msg="starting containerd" revision=fd103cb716352c7e19768e4fed057f71d68902a0.m version=v1.2.7-11-gfd103cb7.m
Sep 02 08:40:44 jetson-tx2 dockerd[4337]: time="2019-09-02T08:40:44.857373152Z" level=info msg="loading plugin "io.containerd.content.v1.content"..." type=io.containerd.content.v1
Sep 02 08:40:44 jetson-tx2 dockerd[4337]: time="2019-09-02T08:40:44.857647296Z" level=info msg="loading plugin "io.containerd.snapshotter.v1.aufs"..." type=io.containerd.snapshotter.v1
Sep 02 08:40:44 jetson-tx2 dockerd[4337]: time="2019-09-02T08:40:44.881136832Z" level=warning msg="failed to load plugin io.containerd.snapshotter.v1.aufs" error="modprobe aufs failed: "modprobe: FATAL: Module aufs not found in directory /lib/modules/4.9.140-l4t-r32.2+g3dcbed5d27df\n": exit status 1"
Sep 02 08:40:44 jetson-tx2 dockerd[4337]: time="2019-09-02T08:40:44.881201472Z" level=info msg="loading plugin "io.containerd.snapshotter.v1.native"..." type=io.containerd.snapshotter.v1
Sep 02 08:40:44 jetson-tx2 dockerd[4337]: time="2019-09-02T08:40:44.881460064Z" level=info msg="loading plugin "io.containerd.snapshotter.v1.overlayfs"..." type=io.containerd.snapshotter.v1
Sep 02 08:40:44 jetson-tx2 dockerd[4337]: time="2019-09-02T08:40:44.881835104Z" level=info msg="loading plugin "io.containerd.snapshotter.v1.zfs"..." type=io.containerd.snapshotter.v1
Sep 02 08:40:44 jetson-tx2 dockerd[4337]: time="2019-09-02T08:40:44.882198304Z" level=warning msg="failed to load plugin io.containerd.snapshotter.v1.zfs" error="path /var/lib/docker/containerd/daemon/io.containerd.snapshotter.v1.zfs must be a zfs filesystem to be used with the zfs snapshotter"
Sep 02 08:40:44 jetson-tx2 dockerd[4337]: time="2019-09-02T08:40:44.882227936Z" level=info msg="loading plugin "io.containerd.metadata.v1.bolt"..." type=io.containerd.metadata.v1
Sep 02 08:40:44 jetson-tx2 dockerd[4337]: time="2019-09-02T08:40:44.882342304Z" level=warning msg="could not use snapshotter aufs in metadata plugin" error="modprobe aufs failed: "modprobe: FATAL: Module aufs not found in directory /lib/modules/4.9.140-l4t-r32.2+g3dcbed5d27df\n": exit status 1"
Sep 02 08:40:44 jetson-tx2 dockerd[4337]: time="2019-09-02T08:40:44.882366656Z" level=warning msg="could not use snapshotter zfs in metadata plugin" error="path /var/lib/docker/containerd/daemon/io.containerd.snapshotter.v1.zfs must be a zfs filesystem to be used with the zfs snapshotter"
Sep 02 08:40:44 jetson-tx2 dockerd[4337]: time="2019-09-02T08:40:44.892040480Z" level=info msg="loading plugin "io.containerd.differ.v1.walking"..." type=io.containerd.differ.v1
Sep 02 08:40:44 jetson-tx2 dockerd[4337]: time="2019-09-02T08:40:44.892183008Z" level=info msg="loading plugin "io.containerd.gc.v1.scheduler"..." type=io.containerd.gc.v1
Sep 02 08:40:44 jetson-tx2 dockerd[4337]: time="2019-09-02T08:40:44.892310528Z" level=info msg="loading plugin "io.containerd.service.v1.containers-service"..." type=io.containerd.service.v1
Sep 02 08:40:44 jetson-tx2 dockerd[4337]: time="2019-09-02T08:40:44.892400000Z" level=info msg="loading plugin "io.containerd.service.v1.content-service"..." type=io.containerd.service.v1
Sep 02 08:40:44 jetson-tx2 dockerd[4337]: time="2019-09-02T08:40:44.892456096Z" level=info msg="loading plugin "io.containerd.service.v1.diff-service"..." type=io.containerd.service.v1
Sep 02 08:40:44 jetson-tx2 dockerd[4337]: time="2019-09-02T08:40:44.892527808Z" level=info msg="loading plugin "io.containerd.service.v1.images-service"..." type=io.containerd.service.v1
Sep 02 08:40:44 jetson-tx2 dockerd[4337]: time="2019-09-02T08:40:44.892596544Z" level=info msg="loading plugin "io.containerd.service.v1.leases-service"..." type=io.containerd.service.v1
Sep 02 08:40:44 jetson-tx2 dockerd[4337]: time="2019-09-02T08:40:44.892652416Z" level=info msg="loading plugin "io.containerd.service.v1.namespaces-service"..." type=io.containerd.service.v1
Sep 02 08:40:44 jetson-tx2 dockerd[4337]: time="2019-09-02T08:40:44.892724448Z" level=info msg="loading plugin "io.containerd.service.v1.snapshots-service"..." type=io.containerd.service.v1
Sep 02 08:40:44 jetson-tx2 dockerd[4337]: time="2019-09-02T08:40:44.892779296Z" level=info msg="loading plugin "io.containerd.runtime.v1.linux"..." type=io.containerd.runtime.v1
Sep 02 08:40:44 jetson-tx2 dockerd[4337]: time="2019-09-02T08:40:44.893490528Z" level=info msg="loading plugin "io.containerd.runtime.v2.task"..." type=io.containerd.runtime.v2
Sep 02 08:40:44 jetson-tx2 dockerd[4337]: time="2019-09-02T08:40:44.894004160Z" level=info msg="loading plugin "io.containerd.monitor.v1.cgroups"..." type=io.containerd.monitor.v1
Sep 02 08:40:44 jetson-tx2 dockerd[4337]: time="2019-09-02T08:40:44.896441856Z" level=info msg="loading plugin "io.containerd.service.v1.tasks-service"..." type=io.containerd.service.v1
Sep 02 08:40:44 jetson-tx2 dockerd[4337]: time="2019-09-02T08:40:44.896566496Z" level=info msg="loading plugin "io.containerd.internal.v1.restart"..." type=io.containerd.internal.v1
Sep 02 08:40:44 jetson-tx2 dockerd[4337]: time="2019-09-02T08:40:44.896730368Z" level=info msg="loading plugin "io.containerd.grpc.v1.containers"..." type=io.containerd.grpc.v1
Sep 02 08:40:44 jetson-tx2 dockerd[4337]: time="2019-09-02T08:40:44.896800128Z" level=info msg="loading plugin "io.containerd.grpc.v1.content"..." type=io.containerd.grpc.v1
Sep 02 08:40:44 jetson-tx2 dockerd[4337]: time="2019-09-02T08:40:44.896888928Z" level=info msg="loading plugin "io.containerd.grpc.v1.diff"..." type=io.containerd.grpc.v1
Sep 02 08:40:44 jetson-tx2 dockerd[4337]: time="2019-09-02T08:40:44.896948832Z" level=info msg="loading plugin "io.containerd.grpc.v1.events"..." type=io.containerd.grpc.v1
Sep 02 08:40:44 jetson-tx2 dockerd[4337]: time="2019-09-02T08:40:44.897019200Z" level=info msg="loading plugin "io.containerd.grpc.v1.healthcheck"..." type=io.containerd.grpc.v1
Sep 02 08:40:44 jetson-tx2 dockerd[4337]: time="2019-09-02T08:40:44.897085056Z" level=info msg="loading plugin "io.containerd.grpc.v1.images"..." type=io.containerd.grpc.v1
Sep 02 08:40:44 jetson-tx2 dockerd[4337]: time="2019-09-02T08:40:44.897136128Z" level=info msg="loading plugin "io.containerd.grpc.v1.leases"..." type=io.containerd.grpc.v1
Sep 02 08:40:44 jetson-tx2 dockerd[4337]: time="2019-09-02T08:40:44.897203104Z" level=info msg="loading plugin "io.containerd.grpc.v1.namespaces"..." type=io.containerd.grpc.v1
Sep 02 08:40:44 jetson-tx2 dockerd[4337]: time="2019-09-02T08:40:44.897252896Z" level=info msg="loading plugin "io.containerd.internal.v1.opt"..." type=io.containerd.internal.v1
Sep 02 08:40:44 jetson-tx2 dockerd[4337]: time="2019-09-02T08:40:44.897924096Z" level=info msg="loading plugin "io.containerd.grpc.v1.snapshots"..." type=io.containerd.grpc.v1
[... cut ...]
Sep 02 08:40:47 jetson-tx2 systemd-journald[2327]: Forwarding to syslog missed 537 messages.
Sep 02 08:40:47 jetson-tx2 systemd[1]: docker.service: Main process exited, code=exited, status=1/FAILURE
Sep 02 08:40:47 jetson-tx2 systemd[1]: docker.service: Failed with result 'exit-code'.
Sep 02 08:40:47 jetson-tx2 systemd[1]: Failed to start Docker Application Container Engine.
```
It complains that some modules are missing. May I have to change the kernel config? Did anyone tried docker in combination with this layer?

madisongh, 2019-09-02:
Looks like there are some kernel module load failure messages there, so start by adding either `kernel-modules` or the specific `kernel-module-xxx` packages for the modules mentioned in the log to CORE_IMAGE_EXTRA_INSTALL.

hesmar, 2019-09-02:
Thanks, adding `kernel-modules` to `CORE_IMAGE_EXTRA_INSTALL` helped!

closed 2019-09-07 by madisongh