OE4T/meta-tegra #174: devnum mismatch UBoot vs. Kernel

TheNitek, 2019-09-03:
We are using meta-tegra to build a custom image for our jetson-tx2 and stumbled upon a problem when booting the rootfs from the SD card:
UBoot considers the external SD card to be device number 1 (interal eMMC is 0), but once the Kernel boots the SD card is mapped to /dev/mmcblk2

Since extlinux.conf uses the {devnum} placeholders for the root= image, the mounting fails and we have to modify the extlinux.conf. Is there a chance to have the Kernel address the external SD card as mmcblk1, so the numbering matches UBoots numbering?

TheNitek, 2019-09-04:
Boot logging looks like this:
```
[    4.624690] mmc0: SDHCI controller on 3460000.sdhci [3460000.sdhci] using ADMA 64-bit with 64 bit addr
[    4.632716] mmc1: SDHCI controller on 3440000.sdhci [3440000.sdhci] using ADMA 64-bit with 64 bit addr
[    4.644879] mmc2: SDHCI controller on 3400000.sdhci [3400000.sdhci] using ADMA 64-bit with 64 bit addr

```

madisongh, 2019-09-07:
Looks like NVIDIA started enabling mmc1 for the M.2 slot in recent L4T releases, which is what's causing the numbering to be out of sync with U-Boot.  I've patched U-Boot to match the kernel's numbering and added those changes to master.

madisongh, 2019-09-07:
You can also work around this by setting `KERNEL_ROOTSPEC` in your `local.conf` to replace the `devnum` U-Boot variable reference with a hard-coded `2` if you know you'll be booting the image you're building off the SDcard.

madisongh, 2019-09-08:
The Jetson-TX1 dev kit also suffered from sdhci device numbering compatibility issues, so I have also updated U-Boot for that platform.

closed 2019-11-23 by madisongh