OE4T/meta-tegra #178: Redundant Boot Update Support

dwalkes, 2019-09-20:
Thanks again for the great work on this project @madisongh.

I'm working on a project which is going to require [Bootloader Update and Redundancy](https://docs.nvidia.com/jetson/l4t/index.html#page/Tegra%2520Linux%2520Driver%2520Package%2520Development%2520Guide%2Fbootloader_update.html%23).  In the short term I need to at least be able to update the FDT through [mender update](https://github.com/Trellis-Logic/meta-mender-community/tree/warrior/meta-mender-tegra).  Ultimately I need to be able to update any of the redundant boot components.  I thought maybe I could use simpler approach in extlinux.conf outlined [here](https://devtalk.nvidia.com/default/topic/1036443/u-boot-error-when-loading-custom-device-tree-from-boot-with-extlinux-conf/) but I've noticed that the FDT doesn't load correctly in extlinux (specifically USB isn't working) presumably because something earlier in the chain is modifying it after loading.

I've noticed you've got a start on support for redundant boot with [tegra186-redundant-boot](https://github.com/madisongh/meta-tegra/blob/warrior/recipes-bsp/tegra-binaries/tegra186-redundant-boot_32.1.0.bb).  In order to work with mender image updates, I think I'll need to:
1. Figure out how to run [l4t_bup_gen.func](https://github.com/madisongh/meta-tegra/blob/f666c77a36e5679e7dc1cfdd35bb04d8cd981a32/recipes-bsp/tegra-binaries/files/tegra194-flash-helper.sh#L159) outside the tegra186-flash-helper.sh script and during the build process, then store the resulting bl_update_payload in the rootfs.
2. Add an [Artifact Install mender state script](https://docs.mender.io/2.1/artifacts/state-scripts) to my update which will mount the new rootfs, symlink the bl_update_payload to /opt/ota-package/, then run nv_update_engine --enable-ab; nv_update_image --install.

I'm curious if you think I'm on the right track with this or if I'm missing something about how it's used today.  Assuming I am on the right track I'm curious if you'd like any of item 1 listed above as a part of meta-tegra or if I should keep all of these changes in meta-mender-community.

Thanks again for sharing this project and for any suggestions you can provide!

madisongh, 2019-09-21:
Yeah, for the TX2 you can't (AFAIK) specify an FDT line in the `extlinux.conf` file - It's cboot, not U-Boot, that is responsible for loading the kernel device tree at boot time, and it has to be signed and stored in the `kernel-dtb` /`kernel-dtb_b` partitions so cboot will load it - cboot then applies fixups to the device tree based on other boot-time configuration and passes its address to U-Boot.

Note that in warrior and later branches, the u-boot build will automatically create a BUP package for you (see the `do_deploy_append_tegra186` function in `u-boot-tegra_2016.07.bb`), would that work for your item 1?

For item 2, you should only need to do `nv_update_engine --enable-ab` once - that setting is non-volatile.  You can use `nvbootctrl get-number-slots` and check if the result is 2; if so, A/B is already set.  You can also use `nvbootctrl get-current-slot` to see which slot you're currently running from, in case you need to know that to determine which rootfs partition to populate during the update.

The only other gotcha I've run into with using `nv_update_engine --install` is that it automatically does a reboot after a successful update, which wasn't convenient for the update script I was using.  To work around that, I ran it in a `chroot` environment where I provided my own `/usr/sbin/reboot` as a script that just exited without rebooting.  The only other files I had to populate in the chroot were:
```
/etc/nv_boot_control.conf
/usr/sbin/nv_update_engine
/usr/sbin/nv_bootloader_payload_udpater
/usr/sbin/nvbootctrl
/opt/ota_package/bl_update_payload
```
This was with L4T R28.2.1; there may be some differences with more recent versions.  But generally, I think what you describe sounds like it's the right way to get this to work.

dwalkes, 2019-09-23:
Thanks @madisongh,

> Note that in warrior and later branches, the u-boot build will automatically create a BUP package for you (see the do_deploy_append_tegra186 function in u-boot-tegra_2016.07.bb), would that work for your item 1?

Yes, and thanks for the pointer, I somehow missed that.  I'm testing the implementation now in [meta-mender-community](https://github.com/Trellis-Logic/meta-mender-community/commit/62a13d67d91753601c2b8237bc44bd5bdadce802) and it looks like it will work.  I also noticed the same issue you mentioned in #114 with u-boot environment storage and [posted a question about this](https://devtalk.nvidia.com/default/topic/1063652/jetson-tx2/mmcblk0boot1-usage-at-address-4177408-and-u-boot-parameter-storage-space-availability/) on the forum.

> The only other gotcha I've run into with using nv_update_engine --install is that it automatically does a reboot after a successful update, which wasn't convenient for the update script I was using

Thanks for the tip.  It looks like I can use a state script on [Artifact Reboot Enter](https://github.com/Trellis-Logic/meta-mender-community/commit/fe0c3e6f8d5cd2f7ec7ac2701e4610b40b2b5d66) and let the update engine start the reboot but I keep the chroot idea in mind if that doesn't work out.

closed 2019-09-23 by dwalkes