OE4T/meta-tegra #180: Jetson Nano Production images?

mdeneen, 2019-09-24:
The partitions on the emmc module on jetson nano production modules is different from that of the dev modules.  I happen to have one production module handy.  Is this something which is already in the pipeline or should I attempt to figure out how the flash.xml.in is generated?

madisongh, 2019-09-25:
It boots off the eMMC, too, rather than SPI flash, right?  I don't have one of those modules, but if you can take a stab at getting the machine configuration set up for it, I'll be happy to help with any questions.

mdeneen, 2019-09-25:
From what I can see it boots from the eMMC.  Unfortunately, the pinout is different with the production module and I can't just pop it in the devkit carrier and expect it to work.  It works enough to flash it, but ethernet, hdmi, etc don't work because their pins are going to the wrong place.

madisongh, 2019-09-28:
Looking at the config files in the L4T kit, I'm guessing that the device tree needs to be the `b00` flavor, rather than the `a02` flavor that is currently set in `conf/machine/jetson-nano.conf`.

mdeneen, 2019-09-28:
Right, I had gathered that from looking at what the sdkmanager does. Additionally, the partition numbers are different. I can have more details on Monday.

wremie, 2019-10-12:
Running into similar issues. According to following thread on [NVIDIA devtalk](https://devtalk.nvidia.com/default/topic/1059058/jetson-nano/confusing-in-board-name-for-the-quot-jetson-nano-developer-kit-quot-/post/5386090/#5386090) the b00 should work for all. Updating conf/machine/jetson-nano.conf should do the trick.

madisongh, 2019-10-13:
On closer inspection, the difference between the a02 and b00 device trees is minimal, just a few pinmux settings. The important difference is in the "SKU" part of the name - "0000" for the original SPI flash/SDcard Nano, and "0002" for the newer eMMC Nano.

wremie, 2019-10-17:
thanks for the analyses. The file naming is a bit ambiquous to me. By any chance, could you point me in the direction of some documentation where I can find the correct interpretation?

madisongh, 2019-10-17:
Where I looked was in the `process_board_version` function in the `p3448.conf.common` file in the L4T kit (R32.2.1 in my case).

wremie, 2019-10-18:
Thanks, makes sense now. Found more information about the naming in the [Bring-up Guide](https://developer.download.nvidia.com/embedded/L4T/r32-2_Release_v1.0/Tegra_Linux_Driver_Package_Nano_Adaptation_Guide.pdf?_H7ObEL464vNYFAy1EeMBZgPpVG5FKUfK2hI0CkvlyuJS9NPwKxbRcNYNuxrLurUAKgTlxnbf4g1Jw4n9I-qJOyb-35Bi9ixPkynmtZDC-PIrH7o4dTUN824kp1tLvnS0HgP-WyrGZ9bPiuujdJ7qRytZrjs4jwDE0UHSeYZ1iDv0ZzxnNaHwe7mh-ciDRu2ZdHD).

triblex, 2019-11-01:
I'm also attempting to build and flash an image to the Jetson Nano Production module. I've attempted to edit the configuration for the jetsn nano (jetson-nano.conf) to this:

```
#@TYPE: Machine
#@NAME: Nvidia Jetson Nano
#@DESCRIPTION: Nvidia Jetson Nano prod board

KERNEL_ARGS ?= "console=ttyS0,115200 console=tty0 fbcon=map:0 net.ifnames=0"
KERNEL_ROOTSPEC ?= "root=/dev/mmcblk0p${@uboot_var('distro_bootpart')} rw rootwait"
IMAGE_ROOTFS_ALIGNMENT ?= "1024"

require conf/machine/include/tegra210.inc

KERNEL_DEVICETREE ?= "_ddot_/_ddot_/_ddot_/_ddot_/nvidia/platform/t210/porg/kernel-dts/tegra210-p3448-0000-p3449-0000-b00.dtb"

MACHINE_FEATURES += "ext2 ext3 vfat"

UBOOT_MACHINE = "p3450-porg_defconfig"

EMMC_SIZE ?= "17179869184"
EMMC_DEVSECT_SIZE ?= "512"
BOOTPART_SIZE ?= "8388608"
BOOTPART_LIMIT ?= "10485760"
ROOTFSPART_SIZE ?= "3221225472"
ODMDATA ?= "0x94000"
EMMC_BCT ?= "P3448_A00_4GB_Micron_4GB_lpddr4_204Mhz_P987.cfg"
NVIDIA_BOARD ?= "t210ref"
NVIDIA_PRODUCT ?= "p3450-porg"
NVIDIA_BOARD_CFG ?= ""
TEGRA210_REDUNDANT_BOOT ?= "0"
PARTITION_LAYOUT_TEMPLATE ?= "flash_l4t_t210_emmc_p3448.xml"
TEGRA_SPIFLASH_BOOT ?= "0"
TEGRA_FAB ?= "300"
TEGRA_BOARDID ?= "3448"
``` 

But running this, gives me an error of "Error: Invalid size(262144) for BCT".  When looking for BCT files, I can only find:

```
E2220_LP3_DSC_931.2Mhz.cfg
P2180_A00_LP4_DSC_204Mhz.cfg
P2530_E01_3GB_Samsung_lpddr4_204Mhz_P968_v2.cfg
P2894_A00_Samsung_3GB_lpddr4_204Mhz_P984_v2.cfg
P3448_A00_4GB_Micron_4GB_lpddr4_204Mhz_P987.cfg
```

Whereas I assume the p3448*.cfg is the only "usable" in my instance on a Jetson Nano.

Do i need to do more to get the build going, than to change the machine conf? How do you get it done?

madisongh, 2019-11-03:
Took some time for me to track this down (it would really help when mentioning an error message if you include some more context about when/where that error was reported) but it looks like the problem is with this line:
```
BOOTPART_SIZE ?= "8388608"
```
It should instead be
```
BOOTPART_SIZE ?= ""
```
just as it is for the regular nano config.  That setting maps to the BOOTPARTSIZE setting in the L4T config file and it appears that it's left null there, too.  That skips the invocation of the `mkgpt` tool to create a GPT based on the flash configuration XML file.  The XML file for the nano-with-eMMC evidently has a size for the BCT partition that that tool doesn't like.

madisongh, 2020-01-27:
I got my hands on the rev B01 carrier and an eMMC Nano module last week, so I've been able to add support for `MACHINE=jetson-nano-emmc` on master and the zeus-l4t-r32.3.1 branch.

wremie, 2020-01-27:
That's great! What changes did you make to get it working? Only eMMC size?

madisongh, 2020-01-27:
I had to patch U-Boot so it would know where the environment block should be stored, but otherwise it was pretty straightforward.

The incorrect EMMC_SIZE setting really only caused issues with Mender (OTA updates) integration.

wremie, 2020-01-27:
Perfect, good to know Mender works with this layer. Currently we are running a fleet of Nano eMMC devices with BalenaOS (also uses meta-tegra) but we are looking into Mender as well. Thanks for the support!

madisongh, 2020-02-05:
Closing this now, since support for `jetson-nano-emmc` as a MACHINE is now present in both master and on the zeus-l4t-r32.3.1 branch.

closed 2020-02-05 by madisongh

vjdaemp, 2020-05-08:
@madisongh  The `jetson-nano-emmc` (Jetson Nano eMMC module with rev B01 carrier board) is the production module of Jetson Nano, correct?

madisongh, 2020-05-08:
I believe so.  It's [this part](https://developer.nvidia.com/embedded/jetson-nano), and for my testing I purchased one through Arrow.

asymptotic, 2022-12-21:
I know this is an old issue, but maybe it's related to what I'm trying to figure out now, and maybe someone here can point me in the right direction?

I have a Seeed Studio J101 carrier with production Nano module with 16GB eMMC. I'm able to flash it using sdkmanager and [this image](https://files.seeedstudio.com/wiki/reComputer-Jetson-Nano/6.jpg) suggests that it's fine as a jetson-nano-emmc (P3448-0002 module, P3449-0000 carrier, sdkmanager uses -b00 flavor).

I'm not able to flash it with Yocto/meta-tegra. I'm using tegra-demo-distro and just following all the instructions. It looks like it flashes ok? It gets through the flash process for every partition and then coldboots. But the system doesn't come up, no video output at all. Haven't been able to check the serial console yet, going to dig up a cable tomorrow.

The biggest discrepancy I've noticed is that when flashing, it gives me `Board ID(3448) version(401) SKU(0002) revision(G.0)`. The board ID and SKU look good, the version is not listed in `conf/machine/jetson-nano-devkit-emmc.conf`, and I have no idea what the revision is about.

Any thoughts? I'm learning as I go here so if I'm missing something obvious or there's a good doc source to read, I'd be happy to get pointed in that direction. Thanks for any help you can give!