OE4T/meta-tegra #182: Linux 4.4 takes way too long to load

sashko, 2019-09-30:
Linux 4.4.159 that is on `thud-l4t-r28.3` takes 7 seconds to load according to `systemd-analyze`, which is absolutely ridiculous.

I understand that the ticket is too generic, so let it be a call a start of a discussion on kernel optimization.

```
uname -a
Linux localhost 4.4.159-l4t-r28.3+gd6a4a9d3da11 #1 SMP PREEMPT Fri Sep 27 18:20:45 UTC 2019 aarch64 aarch64 aarch64 GNU/Linux
```

sashko, 2019-10-08:
[boot.log](https://github.com/madisongh/meta-tegra/files/3702405/boot.log)
[image]

madisongh, 2019-10-09:
You could probably speed things up by trimming unnecessary drivers out of the kernel for your setup.  The L4T configuration is pretty generic, since NVIDIA wanted to have drivers there for devices that you may or may not have connected to your dev kit board, like an audio codec, alternate cameras, etc. And with later L4T BSPs (32.x with the 4.9 kernel) they do HDMI EDID detection earlier in the boot process and pass that along to the kernel, which IIRC helps somewhat with boot time.

madisongh, 2019-10-13:
I tried a couple of tests with the 4.9 kernel, and was seeing systemd-analyze reporting about 8.5 seconds of kernel time during boot.  When I added `quiet` to the kernel command line, that went down to around 3.4 seconds.

madisongh, 2020-12-11:
Closing this now.  If anyone would like to discuss other boot time optimizations, feel free to start up a [discussion](https://github.com/OE4T/meta-tegra/discussions).

closed 2020-12-11 by madisongh