OE4T/meta-tegra #183: nvidia-docker support

UpsilonAudio, 2019-10-02:
Hello @madisongh 

Do you have any plan of integrating `nvidia-docker` into `l4t-32.2` ?

it would be interesting to use docker with nvidia backend,

If no, do you have any ideas / recommendations of how could this be done ?

FYI, I've seen on [this post](https://medium.com/jit-team/building-a-gpu-enabled-kubernets-cluster-for-machine-learning-with-nvidia-jetson-nano-7b67de74172a) how configure `docker` to use nvidia backend on `jetson-nano`.

Regards / Pierre

madisongh, 2019-10-05:
I don't have any plans to do so myself at the moment. I'm not keen on trying to integrate NVIDIA's pre-built binaries for docker support, and I haven't done the research to see if there's anything special they've done to docker itself. The [meta-virtualization](https://git.yoctoproject.org/cgit/cgit.cgi/meta-virtualization/) layer includes docker recipes, so unless there's some special magic in the NVIDIA pre-built copy, you could try using that layer to get docker working.

dwalkes, 2019-10-05:
@UpsilonAudio I used docker-ce with meta-virtualization during the sumo release.  It looks like the content in the article you linked is more current, the docker/jetpack versions I was using at the time didn't support the GPU passthrough described in the article.  I only used a single container running at a time, and I  used the --device argument to docker run to pass through the /dev/nvhost-XXX devices needed.

UpsilonAudio, 2019-10-07:
Thank you all for your answer, I will try to understand if installing nvidia pre-built binaries is enough, of if there is another nvidia magic here.

arnstein, 2019-10-24:
Could this be of help to you? I have no experience with Balena myself.
https://github.com/balena-os/balena-jetson-skx2

madisongh, 2020-03-21:
#266 and #267 bring in NVIDIA container support.

closed 2020-03-21 by madisongh