OE4T/meta-tegra #187: SDK build: nothing provides libnvidia-egl-wayland1

lfdmn, 2019-10-07:
Hi,

On warrior branch, populate SDK fails with error

```
do_populate_sdk: Unable to install packages. Command....
Collected errors:
 * Solver encountered 1 problem(s):
 * Problem 1/1:
 *   - nothing provides libnvidia-egl-wayland1 >= 1.1.2 needed by tegra-libraries-32.2.0-r0.armv8a_tegra186
```

tegra-libraries-32.2.0 do_install has:
`rm ${D}${libdir}/libnvidia-egl-wayland*`

Is this intentional?

madisongh, 2019-10-07:
Yes, it's removed because we build egl-wayland from source.  Looks like `libnvvulkan-producer.so` is the culprit; since we build Vulkan from source also, that library should be removed as part of the do_install task in the tegra-libraries recipe also.

lfdmn, 2019-10-09:
The suggested fix seems to work for my setup. I can start testing the SDK image now.

Thanks

closed 2019-10-09 by lfdmn