OE4T/meta-tegra #188: Jetson Nano fw-env.config

triblex, 2019-10-07:
I'm trying to get RAUC to work on the Jetson Nano, but only U-boot parameters for Jetson-TX1 dev kit is provided for the tegra210. How can i find the correct parameters for the Jetson Nano?

madisongh, 2019-10-07:
Doesn't look like the Nano supports saving the environment block with the stock U-Boot configuration. From the `include/configs/p3450-porg.h` file in the U-Boot source:
```
/* Environment s/b at end of SPI, fix it later */
#define CONFIG_ENV_IS_NOWHERE
```

triblex, 2019-10-08:
I found [this possible solution](https://devtalk.nvidia.com/default/topic/1057840/jetson-nano-u-boot-tool-fw_printenv-and-fw_setenv/?offset=2#) that i will try and recreate.

triblex, 2019-10-08:
That solution will have RAUC resetting the u-boot environment every time there is an update. How can i make RAUC work with my system then, any ideas?

madisongh, 2019-10-08:
I'm not sure how RAUC works, but there are a few options. If you opt for the file in `/boot` approach that was mentioned in the thread you linked to, just make sure RAUC saves a copy of that file before updating the filesystem, then restores it to the known location before rebooting.  If it has to update U-Boot environment variables as part of the update process, some additional tweaks might be needed to ensure the file is in the expected location when it does so - or patch the U-Boot env tools to take an extra argument (or use an environment variable) to locate the file.

Another option would be to carve out space either on the on-board SPI flash or on the SDcard to hold the environment block.  For the SPI flash, you'd have to do some digging into how NVIDIA's boot code uses the flash so U-Boot doesn't overwrite something important, and you'd have to make sure that the flash device is accessible from Linux -- I'm not sure if the device tree already has an entry for it or not, but if not it should be relatively easy to add. Reserving space on the SDcard might be easier - you could even add a partition for it to ensure that it doesn't conflict with other uses.  Since U-Boot itself is loaded from the SDcard at boot time, that might make the most sense, but it depends on your use case - if you need all or part of the U-Boot environment to be retained even when swapping in a different SDcard, the SPI flash would be the best place.

My understanding is that production Nano modules have an on-board eMMC, and it sounds like later dev kits will be based on that module - so they may not include the SPI flash, since it wouldn't be necessary.  Bear that in mind if you need to support this functionality on both the current dev kits and production modules.

triblex, 2019-10-09:
I will attempt to make a new partition for the env file like you mentioned. It seems like it is the most future friendly and easiest way to solve this issue.

As i can see from the other issue about TX partitioning, i made a bbappend file with
```
FILESEXTRAPATHS_prepend_tegra210 := "${THISDIR}/files:"
SRC_URI_prepend_tegra210 += " \
    file://flash_l4t_t210_custom.xml \
"

do_preconfigure_prepend_tegra210() {
    mkdir -p ${S}/bootloader/${NVIDIA_BOARD}/cfg/
    cp ${WORKDIR}/flash_l4t_t210_custom.xml ${S}/bootloader/${NVIDIA_BOARD}/cfg/
}
```
And put the additional partitions into the custom xml as well as edit the JetsonNano machine conf `SDCARD_LAYOUT_TEMPLATE` to point to the custom xml. As i understand with the Jetson Nano, I also made a bbappend to give a custom `sdcard-layout.in`:
```
FILESEXTRAPATHS_prepend := "${THISDIR}/files:"

SRC_URI = "file://sdcard-layout-custom.in"
```
Where i also add in the partitions i need that matches the custom xml file contents.

madisongh, 2019-10-09:
Right, if you go this route, you'll need to format the added partition with an ext4 filesystem and put an empty `uboot.env` file in it (although U-Boot may be able to create it if it's not present, I'm not sure).  For testing, you could do this manually.  Making it part of the build is a bit more complicated; it would mean creating/formatting a file with the ext4 filesystem in it and putting the name of that file in the XML and sdcard-layout files - use a real name.... then you'd add a `tegraflash_custom_pre()` function in your image recipe to copy that ext4 container file into the directory where the rest of the tegraflash content goes.

The config file with the `uboot.env 0x0000 0x2000` should still go in `${sysconfdir}/fw_env.config` in your rootfs.

The other approach would be to use CONFIG_ENV_IS_IN_MMC in U-Boot, so you don't have to format the partition.  The tricky part there is figuring out the correct offset to use, since you wouldn't be able to just reference the partition by number.

triblex, 2019-10-10:
I see. I'm not really sure how i go about 

> creating/formatting a file with the ext4 filesystem in it

and 

> copy that ext4 container file into the directory where the rest of the tegraflash content goes.

but it's doable.

If I do go with the  CONFIG_ENV_IS_IN_MMC approach, do you know how i could figure out the offset? I found this:

```
#define CONFIG_ENV_SECT_SIZE    (8 * 1024)
#define CONFIG_ENV_SIZE         CONFIG_ENV_SECT_SIZE

#define CONFIG_ENV_IS_IN_MMC	1
#define CONFIG_ENV_OFFSET	(768 * 1024)
```

In a meta-freescale patch, you think this offset could be the case for me too? And if so, what would i put into the fw_env.config file?

madisongh, 2019-10-12:
I looked into this, and it was pretty simple to add the environment block to the SPI flash, so I went ahead and updated u-boot-tegra and the u-boot recipes in master to do that.

If you'd prefer to use CONFIG_ENV_IS_IN_MMC, you would compute CONFIG_ENV_OFFSET by multiplying the start sector for the partition you have added by the sector size (usually 512).  `sgdisk --print /dev/XXX` should provide you with both of those numbers once you have formatted the card.  Then in your `fw_env.config` file, use `/dev/mmcblk0pN` as the device (where `N` is the partition number), `0` as the offset, and `0x2000` as the size.

triblex, 2019-10-17:
The SPI flash is working like a charm. But now that I am trying to make an mmc partition with the environment, i can't seem to specify the file for the partition correctly. No matter what i do, bitbaking my image fails with `dd: failed to open 'uboot.env': No such file or directory` when doing `do_image_tegraflash`.

Where and how do i specify the file i need in the partition via the meta-tegra layer?

madisongh, 2019-10-17:
You should just leave the partition empty - U-Boot will initialize it when you do your first `saveenv`.  So just use `/dev/zero` in place of the `uboot.env` file name you have in your modified `partition-layout.in`.

triblex, 2019-10-22:
It appears when i write `14,ENV,196608,/dev/zero` in the sdcard-layout.in and add

```
        <partition name="ENV" id="25" type="data">
            <allocation_policy> sequential </allocation_policy>
            <filesystem_type> basic </filesystem_type>
            <size> 196608 </size>
            <file_system_attribute> 0 </file_system_attribute>
            <allocation_attribute> 0x8 </allocation_attribute>
            <percent_reserved> 0 </percent_reserved>
            <filename> uboot.env </filename>
        </partition>
```
to my custom flash_l4t_t210_custom.xml file that the machine conf is pointing to, the last part of the bitbake `do_tegraflash` will take every single byte of space left on my disk (which is 800gb). It simply just goes on forever untill there is no more room left on the disk. The file `/build/tmp/work/jetson_nano-poky-linux/image/1.0-r0/deploy-image-complete/image_*.sdcard` were 327gb when i noticed and stopped the build. What did i do wrong here?

madisongh, 2019-10-22:
Nothing... that's my fault.  I thought I had tested that approach to having a null-filled partition, but I must have got it wrong - the `dd` command that copies the partition contents is not size-limited. I'll fix it so it works if you specify `/dev/zero`, or just use `14,ENV,196608,` (with no file name at all), it does the right thing.

triblex, 2019-10-22:
But if I were to actually specify a file for a new partition, how/where would i specify it in the meta-tegra layer?

madisongh, 2019-10-22:
In your image recipe, you would add a function called `tegraflash_custom_pre` to create or copy the file into the current working directory.  For example,
```
tegraflash_custom_pre() {
    dd if=/dev/zero of=uboot.env bs=196608 count=1
}
```
To create a zero-filled file of the size for your ENV partition.  Then you could specify `uboot.env` in your `sdcard-layout.in` file.

triblex, 2019-10-23:
I've made a patch for the u-boot-tegra environment location:

```
diff --git a/include/configs/p3450-porg.h b/include/configs/p3450-porg.h
index 7be05f3..427fb60 100644
--- a/include/configs/p3450-porg.h
+++ b/include/configs/p3450-porg.h
@@ -34,12 +34,10 @@
 	func(PXE, pxe, na) \
 	func(DHCP, dhcp, na)
 
-/* Environment s/b at end of SPI, fix it later */
-#define CONFIG_ENV_IS_IN_SPI_FLASH
-/* 64KB at end of SPI flash is used, so position at 128K from end
-   in the padding area */
-#define CONFIG_ENV_OFFSET		(CONFIG_SPI_FLASH_SIZE-131072)
-#define CONFIG_ENV_SECT_SIZE		4096
+/* Env is located in it's own partition */
+#define CONFIG_SYS_MMC_ENV_DEV		1
+#define CONFIG_ENV_IS_IN_MMC
+#define CONFIG_ENV_OFFSET		12897484800
 
 /* SPI */
 #define CONFIG_SF_DEFAULT_MODE		SPI_MODE_0
```

Where the `CONFIG_ENV_OFFSET` is calculated from the `gdisk -l /dev/mccblk0` output :
```
Number  Start (sector)    End (sector)  Size       Code  Name
   1           24576        12607487   6.0 GiB     8300  APP
   2            2048            2303   128.0 KiB   8300  TBC
   3            4096            4991   448.0 KiB   8300  RP1
   4            6144            7295   576.0 KiB   8300  EBT
   5            8192            8319   64.0 KiB    8300  WB0
   6           10240           10623   192.0 KiB   8300  BPF
   7           12288           13439   576.0 KiB   8300  TOS
   8           14336           14463   64.0 KiB    8300  EKS
   9           16384           17663   640.0 KiB   8300  LNX
  10           18432           19327   448.0 KiB   8300  DTB
  11           20480           20735   128.0 KiB   8300  RP4
  12           22528           22687   80.0 KiB    8300  BMP
  13        12607488        25190399   6.0 GiB     8300  APP_B
  14        25190400        25190783   192.0 KiB   8300  ENV

```

Where `CONFIG_ENV_OFFSET` = (25190400 * 512).

When I `saveenv` and print it I see the correct environment variables. But when i boot it states that the mmcblk0p1 (the first of the two rootfs):
```
Failed to mount ext2 filesystem...
** Unrecognized filesystem type **
```
And when i print the env with `fw_print_env` I don't see the same environment as in u-boot.

So i was wondering if these issues are because of incorrect `CONFIG_ENV_OFFSET` or if it could be something else?

triblex, 2019-10-23:
If I decide i would want the env partition to be of ext4 format, how would I do this?

madisongh, 2019-10-24:
Looking at the U-Boot source (`common/env_mmc.c`), the offset is a 32-bit unsigned integer, so any offset over 4GiB isn't going to work very well.  You might need to rearrange the positions of the partitions so you can squeeze in the ENV partition at a smaller offset.

closed 2019-11-07 by triblex

Ecordonnier, 2019-12-11:
@madisongh  I am trying to make mender work on the jetson-nano and found your patch 990d978a86195597ac2 "enable env block storage in SPI flash". How do you update the SPI flash? Is it done using the usb connector of the jetson nano dev. kit? Up to now I only flashed SD cards, and didn't find online how to flash the SPI flash.

madisongh, 2019-12-11:
Yes, via USB.  See the "Preparing a Jetson Development Kit for Use" in the [L4T Quick Start Guide](https://docs.nvidia.com/jetson/l4t/Tegra%20Linux%20Driver%20Package%20Development%20Guide/quick_start.html#) for NVIDIA's docs on this, and some basic instructions here [on the wiki](https://github.com/madisongh/meta-tegra/wiki/Flashing-the-Jetson-Dev-Kit).

Ecordonnier, 2019-12-13:
Thanks @madisongh . Is there an advantage of storing the environment in the SPI flash? In the branch https://github.com/Trellis-Logic/meta-mender-community/pull/7 adding support for Mender to the jetson-nano, for which you'll probably get a pull request soon, the u-boot environment gets moved to a partition in the SD-card. Would you merge this, or do you want to keep it in the SPI flash?

Having it in the SD-card makes it easier to replace it with the rest of the software, and also I think when swapping modules you don't want to keep the u-boot environment state anyway, so you would have to replace the content of the SPI flash each time, which is an extra step / error-prone.

madisongh, 2019-12-13:
Keeping the U-Boot environment in the SPI flash means that it is retained across reboots regardless of what you've got on the SDcard you insert. That might be an advantage in some cases.  For Mender, it would depend on whether you want to retain the Mender settings regardless of the SDcard contents.  Some settings, like the machine ID, might be worth retaining; others, like the boot partition, maybe not.

Production Nano modules have on-board eMMC rather than the SPI flash/SDcard combination, so this choice only affects the development boards.