OE4T/meta-tegra #189: appending custom kernel device tree

klutchell, 2019-10-08:
Hey Matt, you seem to be swamped lately so I'll keep this short.

What were the main reason(s) behind using kernel files from https://github.com/madisongh/linux-tegra-4.9 rather than the kernel provided with the L4T BSP? I don't need the whole history just a quick reason or two.

I only noticed when I was overriding customized files in tegra-binaries and I found my pmc config and pinmux config were being used in the image (verified via hashes) but my kernel device tree (.dtb) was not. I had to include the files as part of the linux-tegra recipe so they would end up in the git source tree.

```bash
# installs in the expected location but not used by linux-tegra
recipetool appendsrcfile -wm jetson-tx2-4gb ../meta-local tegra-binaries $HOME/tegra186-quill-p3489-0888-a00-00-base.dtb 'Linux_for_Tegra/kernel/dtb/tegra186-quill-p3489-0888-a00-00-base.dtb'

# installs and used by linux-tegra and other downstream recipes
recipetool appendsrcfile -wm jetson-tx2-4gb ../meta-local linux-tegra $HOME/tegra186-quill-p3489-0888-a00-00-base.dtb '../build/arch/${ARCH}/boot/dts/tegra186-quill-p3489-0888-a00-00-base.dtb'
```

What we have is a modified version of the kernel device tree provided by Nvidia, so I'm wondering if we should be using linux-tegra-4.9 as a base instead.

madisongh, 2019-10-08:
Well, I wanted to be able to build the kernel from source, so it can be customized/fixed/etc..  NVIDIA makes the kernel source available via git, but it's split across 16 or so separate repositories, which wasn't very convenient, so with each L4T release I fetch their sources and recombine them into one source tree.  Device tree sources are in there... if you have a customized device tree, you should be able to fork linux-tegra-4.9 and make the necessary modifications.

klutchell, 2019-10-08:
Good answer, thank you!

closed 2019-10-08 by klutchell