OE4T/meta-tegra #193: Secureboot fuse programming on Jetson TX2 32.2.1

dwalkes, 2019-10-18:
Hi Everyone,
I'm trying to go through secureboot setup steps on 32.2.1 (using [nvidia setup instructions](https://docs.nvidia.com/jetson/l4t/index.html#page/Tegra%2520Linux%2520Driver%2520Package%2520Development%2520Guide%2Fbootloader_secure_boot.html%23))  and I'm blocked on the [odmfuse.sh step](https://docs.nvidia.com/jetson/l4t/index.html#page/Tegra%2520Linux%2520Driver%2520Package%2520Development%2520Guide%2Fbootloader_secure_boot.html%23wwpID0E0LG0HA).  I haven't been able to get this command to succeed on any part I've tried, and [my forum post](https://devtalk.nvidia.com/default/topic/1064080/jetson-tx2/secure-boot-odmfuse-programming-step-fail-on-32-2-1/post/5393200) has so far not resulted in any feedback which helps understand or determine a workaround or fix for the problem.  I also have not yet been able to get a ```bash -x``` or ```strace``` output of a successful ``odmfuse.sh``` command from nvidia.  This might be useful to compare to what I'm seeing and come up with a workaround.

My question is whether anyone has successfully performed secureboot fuse programming on 32.2.1 using ```odmfuse.sh``` on any part but especially Jetson TX2 and, if so, if they would be willing to share a ```bash -x``` and/or ```strace``` of the successful execution which I could use to compare to [what I see when attempting to run](https://github.com/Trellis-Logic/secureboot-tegra/blob/master/burn_pkc_first_attempt.log). 

I've got a project at https://github.com/Trellis-Logic/secureboot-tegra which automates the steps related to downloading and setting up the files needed for secureboot.  If I can figure out how to get past this currently blocking issue my long term goal would be to put this into a set of patches which could be contributed to meta-tegra.

Thanks for any suggestions or feedback.

madisongh, 2019-10-19:
Looks like the problem is with the chip UID checks - the device is very finicky about when the UID can be read, so `--skipuid` has to be applied to the `tegraflash.py` invocations at exactly the right time... and that can be different for each SoC type.

With the attached patch to `odmfuse.sh` I was able to get a successful fuse burn: [fix-uid-check-in-odmfuse.patch.txt](https://github.com/madisongh/meta-tegra/files/3746867/fix-uid-check-in-odmfuse.patch.txt).

Of course, now that I've done this, I've got to get meta-tegra updated so I can generate signed images for flashing...

dwalkes, 2019-10-21:
> With the attached patch to odmfuse.sh I was able to get a successful fuse burn: fix-uid-check-in-odmfuse.patch.txt.

Thanks @madisongh!  I can confirm that works for me too.

> Of course, now that I've done this, I've got to get meta-tegra updated so I can generate signed images for flashing...

I'd be happy to help with this if I can be useful.  I'll see how far I can get this week and open a new issue if I can get anything ready to share.  Or if you are able to complete it before I can I'd be happy to help with testing/documentation.

closed 2019-10-21 by dwalkes

madisongh, 2019-10-22:
@dwalkes I have some work in progress on the [wip-secure-boot-tegra186](https://github.com/madisongh/meta-tegra/tree/wip-secure-boot-tegra186) branch.