OE4T/meta-tegra #196: Secure boot documentation

akavoukis, 2019-10-21:
Hi, Could we please have some more information regarding the use of the flashing scripts to secure a device (secure boot and encryption)? A wiki page would be ideal.

Specifically i am more interested in an Xavier device.

I have already been though the Instructions [here](https://docs.nvidia.com/jetson/archives/l4t-archived/l4t-322/index.html#page/Tegra%2520Linux%2520Driver%2520Package%2520Development%2520Guide%2Fbootloader_secure_boot.html%23) but when it comes to usage of the generated flash scripts it gets more complicated and cannot follow the instructions supplied by the documentation. 

I noticed there is a WIP branch for secure boot on TX2 (T186), Is the work common for the Xavier(T194)?

Also I have noticed that on the "tegra194-flash-helper.sh" there is no handle for the "--burnfuses" parameter. 

Am I miss-understanding the flash scripts or is the work incomplete for Xavier device ?

madisongh, 2019-10-22:
The work is not even begun for Xavier, and unfortunately the secure boot setup is sufficiently different among the different processor models that a simple copy-paste approach isn't going to work. It also doesn't help that the documentation and/or scripts that NVIDIA provides don't appear to be well-tested. I've run into several issues in the past on both the TX1 and TX2 where the documented procedure for fusing hasn't worked at all, or broke between revisions of the L4T BSP.

That said, I'll work on extending the WIP branch to TX1/Nano and Xavier when I can, and write up some documentation on it.

akavoukis, 2019-10-23:
Thank you very much @madisongh, I am currently working on a project using an xavier, let me know if there is anything i can help with.

closed 2019-10-23 by akavoukis

madisongh, 2019-10-27:
@akavoukis I've merged the image signing support into master.  Signing appears to work properly for Xavier builds, although I don't have a secured Xavier device to test with.  Some documentation is [on the wiki](https://github.com/madisongh/meta-tegra/wiki/Secure-Boot-Support).

akavoukis, 2019-10-28:
Thank you very much @madisongh, I really appreciate it. I will try this as soon as possible and let you know if it works.