OE4T/meta-tegra #197: Onboard camera of Jetson TX2 dev board

johannesgensheimer, 2019-10-22:
Hello,

I successfully build the yocto image for the Jetson TX2 dev board including OpenCV, Cuda, Python3, gcc, g++, make with GCCVERSION="6.%"
Branches: 
meta-tegra: rocko-l4t-r28.2
meta-openembedded: rocko
poky: rocko

Do you know how I can use the camera Image in Python or C/C++?

I tried the following in Python 3:
- cap=cv2.VideoCapture('/dev/video0')
Error:
    VIDEOIO ERROR: libv4l unable convert to requested pixfmt
    VIDEOIO ERROR: libv4l unable to ioctl VIDIOCSPICT
    (python3:468): GLib-GObject-CRITICAL **: g_object_set: assertion 'G_IS_OBJECT (object)' failed

- cap = cv2.VideoCapture("udpsrc port=5000 ! application/x-rtp, encoding-name=H264, payload=96 ! rtph264depay ! h264parse ! avdec_h264 ! videoconvert ! video/x-raw, format=(string)BGR ! appsink", cv2.CAP_GSTREAMER)
Error:
GStreamer: Error opening bin: no element "udpsrc"

-cap = cv2.VideoCapture("nvcamerasrc ! video/x-raw(memory:NVMM), width=(int)1280, height=(int)720,format=(string)I420, framerate=(fraction)30/1 ! nvvidconv flip-method=0 ! video/x-raw, format=(string)BGRx ! videoconvert ! video/x-raw, format=(string)BGR ! appsink")
Error: GStreamer: Error opening bin: no element "nvcamerasrc"

Thank you for your help and please let me know if you need any further information.

Best,

Johannes

madisongh, 2019-10-22:
Looks like some gstreamer plugins are either missing or blacklisted. IIRC, the `nvcamerasrc` element would be in one of the plugins in the `gstreamer1.0-plugins-tegra` package, so make sure that got installed.  I don't remember off the top of my head which plugin package includes the `udpsrc` element.

Double-check the presence of the desired elements using the `gst-inspect-1.0` command. You can check for blacklisted plugins with `gst-inspect-1.0 -b`.

johannesgensheimer, 2019-10-22:
I can check this on monday. Thank you for the respond. I’ll let you know if it is solved!

johannesgensheimer, 2019-10-28:
@madisongh your answer absolutley improved the situation.
Although I still get errors when I try to open the camera.

**My CORA_IMAGE_EXTRA_INSTALL looks like this:**
"python-core python3 opencv opencv-samples libopencv-core libopencv-imgproc cuda-core cuda-samples gcc g++ make cmake gstreamer1.0-plugins-tegra"

**With the command gst-inspect-1.0 I get:**
root@(none):~# gst-inspect-1.0 
nvjpeg:  nvjpegenc: JPEG image encoder
nvjpeg:  nvjpegdec: JPEG image decoder
nvvidconv:  nvvidconv: NvVidConv Plugin
nvcamera:  nvcamerasrc: Nvidia Camera Source
nvcamera:  nvtee: NvTee
nveglstreamsrc:  nveglstreamsrc: nVidia EGL Stream
nvivafilter:  nvivafilter: NvIVAFilter Plugin
nvvideosink:  nvvideosink: nVidia Video Sink
videocuda:  videocuda: CUDA Post processor
coretracers:  latency (GstTracerFactory)
coretracers:  log (GstTracerFactory)
coretracers:  rusage (GstTracerFactory)
coretracers:  stats (GstTracerFactory)
coretracers:  leaks (GstTracerFactory)
nvegltransform:  nvegltransform: NvEGLTransform
coreelements:  capsfilter: CapsFilter
coreelements:  concat: Concat
coreelements:  dataurisrc: data: URI source element
coreelements:  downloadbuffer: DownloadBuffer
coreelements:  fakesrc: Fake Source
coreelements:  fakesink: Fake Sink
coreelements:  fdsrc: Filedescriptor Source
coreelements:  fdsink: Filedescriptor Sink
coreelements:  filesrc: File Source
coreelements:  funnel: Funnel pipe fitting
coreelements:  identity: Identity
coreelements:  input-selector: Input selector
coreelements:  output-selector: Output selector
coreelements:  queue: Queue
coreelements:  queue2: Queue 2
coreelements:  filesink: File Sink
coreelements:  tee: Tee pipe fitting
coreelements:  typefind: TypeFind
coreelements:  multiqueue: MultiQueue
coreelements:  valve: Valve element
coreelements:  streamiddemux: Streamid Demux
nvarguscamerasrc:  nvarguscamerasrc: NvArgusCameraSrc
staticelements:  bin: Generic bin
staticelements:  pipeline: Pipeline object

Total count: 12 plugins, 39 features

**And with gst-inspect-1.0 -b:**
root@(none):~# gst-inspect-1.0 -b
Blacklisted files:

Total count: 0 blacklisted files

**The error I get with the line "cap = cv2.VideoCapture("udpsrc port=5000 ! application/x-rtp, encoding-name=H264, payload=96 ! rtph264depay ! h264parse ! avdec_h264 ! videoconvert ! video/x-raw, format=(string)BGR ! appsink", cv2.CAP_GSTREAMER)" is:**
OpenCV Error: Unspecified error (GStreamer: cannot find appsink in manual pipeline
) in cvCaptureFromCAM_GStreamer, file /usr/src/debug/opencv/3.3+gitAUTOINC+87c27a074d_2a9d1b22ed_a62e20676a_34e4206aef_fccf7cd6a4-r0/git/modules/videoio/src/cap_gstreamer.cpp, line 796
VIDEOIO(cvCreateCapture_GStreamer (CV_CAP_GSTREAMER_FILE, filename)): raised OpenCV exception:

/usr/src/debug/opencv/3.3+gitAUTOINC+87c27a074d_2a9d1b22ed_a62e20676a_34e4206aef_fccf7cd6a4-r0/git/modules/videoio/src/cap_gstreamer.cpp:796: error: (-2) GStreamer: cannot find appsink in manual pipeline
 in function cvCaptureFromCAM_GStreamer

**Is there an additional plugin for appsink?**

Best and many thanks, 
Johannes

madisongh, 2019-10-28:
Yes.  You need to include `gstreamer1.0-plugins-base-app` in your image, to get that one. You might want to just include `gstreamer1.0-plugins-base`, which will pull in all of the base plugins.

johannesgensheimer, 2019-10-29:
Ok, I included this too.
This is what I included from gstreamer:
gstreamer1.0, gstreamer1.0-plugins-tegra, gstreamer1.0-plugins-base, gstreamer1.0-plugins-bad, gstreamer1.0-omx-tegra, gstreamer1.0-plugins-nveglgles.

**I get the following errors in Python3:**

- import cv2
   cap = cv2.VideoCapture("udpsrc port=5000 ! application/x-rtp, encoding-name=H264, payload=96 ! rtph264depay ! h264parse ! avdec_h264 ! videoconvert ! video/x-raw, format=(string)BGR ! appsink", cv2.CAP_GSTREAMER)

**Output:** 
OpenCV Error: Unspecified error (GStreamer: unable to start pipeline
) in cvCaptureFromCAM_GStreamer, file /usr/src/debug/opencv/3.3+gitAUTOINC+87c27a074d_2a9d1b22ed_a62e20676a_34e4206aef_fccf7cd6a4-r0/git/modules/videoio/src/cap_gstreamer.cpp, line 881
VIDEOIO(cvCreateCapture_GStreamer (CV_CAP_GSTREAMER_FILE, filename)): raised OpenCV exception:
/usr/src/debug/opencv/3.3+gitAUTOINC+87c27a074d_2a9d1b22ed_a62e20676a_34e4206aef_fccf7cd6a4-r0/git/modules/videoio/src/cap_gstreamer.cpp:881: error: (-2) GStreamer: unable to start pipeline
in function cvCaptureFromCAM_GStreamer

**When I use the shell to try my video connection I get:**

- $ gst-launch-1.0 videotestsrc num-buffers=150 ! 'video/x-raw, format=(string)I420,width=(int)640, height=(int)480' ! omxh264enc ! 'video/x-h264, stream-format=(string)byte-stream' ! filesink location=test.h264 -e

**Output:** 
Setting pipeline to PAUSED ...
Pipeline is PREROLLING ...
Framerate set to : 30 at NvxVideoEncoderSetParameterNvMMLiteOpen : Block : BlockType = 4
===== MSENC =====
NvMMLiteBlockCreate : Block : BlockType = 4
===== MSENC blits (mode: 1) into tiled surfaces =====
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
Got EOS from element "pipeline0".
Execution ended after 0:00:00.377923808
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...

- $ gst-launch-1.0 nvcamerasrc fpsRange="30.0 30.0" ! 'video/x-raw(memory:NVMM), width=(int)1920, height=(int)1080, format=(string)I420, framerate=(fraction)30/1' ! \
nvvidconv flip-method=2 ! 'video/x-raw(memory:NVMM), format=(string)I420' ! fakesink silent=false -v

**Outout:**
Setting pipeline to PAUSED ...
ERROR: Pipeline doesn't want to pause.
Setting pipeline to NULL ...
Freeing pipeline ...

I think I have included everything that's there, or did I forget anything?
Do have have to set up anything else?

johannesgensheimer, 2019-11-18:
Any other ideas how I can solve this issue?

madisongh, 2019-11-18:
Have you tried any of the examples in NVIDIA's [Accelerated GStreamer User Guide](https://developer.nvidia.com/embedded/dlc/l4t-accelerated-gstreamer-guide-28-2-ga)? That's what I started with to come up with the test pipeline I've been using. You can also increase GStreamer's log levels to get more information about what is going on, as mentioned, for example, [here](https://gstreamer.freedesktop.org/documentation/tutorials/basic/debugging-tools.html?gi-language=c).

madisongh, 2020-12-11:
No activity on this for over a year, so closing.

closed 2020-12-11 by madisongh