OE4T/meta-tegra #199: Partitions and unallocated space (Jetson Nano)

triblex, 2019-10-29:
I have my partition scheme:
```
4,TBC,131072,nvtboot_cpu.bin.encrypt
5,RP1,458752,DTBFILE
6,EBT,589824,cboot.bin.encrypt
7,WB0,65536,warmboot.bin.encrypt
8,BPF,196608,sc7entry-firmware.bin.encrypt
9,TOS,589824,tos-mon-only.img.encrypt
10,EKS,65536,eks.img
11,LNX,655360,LNXFILE
12,DTB,458752,DTBFILE
13,RP4,131072,rp4.blob
14,BMP,81920,bmp.blob
1,UBOOTENV,81920,uboot.env
2,APP,7862222848,APPFILE
3,APP_B,7862222848,APPFILE
```
The same sizes for the partitions are specified in a custom partition xml file.
When I build and flash the image to the sdcard, the partitions are there as defined in correct sizes but looking at gparted, I can see that only a few megabytes are available in my rootfs partitions, while several gigabytes are unallocated space in both partitions.

Is this something to do with the 
```
<allocation_policy> sequential </allocation_policy>
<filesystem_type> basic </filesystem_type>
<size> 7862222848 </size>
<file_system_attribute> 0 </file_system_attribute>
<allocation_attribute> 0x8 </allocation_attribute>
<percent_reserved> 0 </percent_reserved>
<align_boundary> 4096 </align_boundary>
```
in the xml file that i need to set differently?

I can fix it with running `e2fsck -f -y -v -C 0 /dev/mmcblk0pX` through gparted after flashing to the sdcard, but I would like it to not be necessary to manually check and repair the partitions for every image i flash.

triblex, 2019-10-30:
I ended up doing something similar as with #154 (last comment) where I have yocto building the rootfs and creating an image with all the necessary 11 partitions + a uboot.env partition and then I add my two rootfs partitions afterward as a post build hook.

madisongh, 2019-10-30:
You could try setting `IMAGE_ROOTFS_SIZE` to a value that matches the size of the partition (but expressed in KiB rather than bytes).  That's what's used to size the ext4 filesystem for the rootfs. You could also try using `resize2fs` as part of a post-build hook to get the filesystem to match the partition size.

triblex, 2019-11-05:
When building image for Jetson Nano production module emmc, it's the same problem. The partition size is as wanted, but the image that is created from rootfs and flashed to the partition is only 1gb in size, leaving the rest of the partition unallocated. I've tried giving the variable `IMAGE_ROOTFS_SIZE` the size i want but with no change.

I made a workaround that will have me creating the rootfs partition image myself via the `tegraflash_custom_pre()` function:
```
    dd if=/dev/zero of=rootfs.img bs=512 count=15355904
    mkfs.ext4 -j rootfs.img
    loop_dev="$(sudo losetup --show -f -P rootfs.img)"
    tmpdir="$(sudo mktemp -d)"
    sudo mount "${loop_dev}" "${tmpdir}"
    sudo cp -a "${WORKDIR}/rootfs"/* "${tmpdir}"
    sudo umount "${tmpdir}"
    sudo rmdir "${tmpdir}"
    sudo losetup -d "${loop_dev}"
```

But I don't like this way as the original rootfs image is still created even though it isn't used and because I need to use `sudo` for it.

triblex, 2019-11-06:
Now that i try to do the same with the Jetson Nano production module, the same issue arises. If i let `tegraflash` build the `rootfs.img`, the image will only be 1GiB while i have set the partition to be 3GiB and thus creating 2GiB of unallocated space in my two rootfs partitions.

When assigning the variable IMAGE_ROOTFS_SIZE to the desired size in KiB, there is no difference and the rootfs image is still only 1GiB. 

Any other ideas on how I can fix it? without the post-build hooks preferably.

madisongh, 2019-11-06:
I just tested this myself with a jetson-tx1, using these settings in my `local.conf` to get the rootfs filesystem size to match the partition size:
```
IMAGE_ROOTFS_SIZE = "${@int(d.getVar('ROOTFSPART_SIZE')) // 1024 - 4096}"
IMAGE_OVERHEAD_FACTOR = "1.0"
```
Booting `core-image-minimal` flashed to the TX1, I see:
```
root@jetson-tx1:~# df /
Filesystem           1K-blocks      Used Available Use% Mounted on
/dev/mmcblk0p1        11730056    115480  10969048   1% /
```
which indicates that the filesystem was sized pretty close to the actual size of the partition.  I did have to go a couple of iterations to get that size, although it looks like there's still 1G or so of additional adjustment I might be able to make. Look at your `log.do_image_ext4` log file to see the sizes it is reporting. Besides the overhead factor, there are other additions that can get made in calculating the rootfs size that you might need to account for.

triblex, 2019-11-08:
If i put in:
```
IMAGE_ROOTFS_SIZE = "${@int(d.getVar('ROOTFSPART_SIZE')) // 1024 - 4096}"
IMAGE_OVERHEAD_FACTOR = "1.0"
```
Into my `local.conf`, i get the the following error:
```
bb.data_smart.ExpansionError: Failure expanding variable IMAGE_ROOTFS_SIZE, expression was ${@int(d.getVar('ROOTFSPART_SIZE')) // 1024 - 4096} which triggered exception TypeError: int() argument must be a string, a bytes-like object or a number, not 'NoneType'
```
Looking into `log.do_image_ext4` i can see that the rootfs partition image is created with:
```
dd if=/dev/zero of=..../image.rootfs.ext4 seek=3248128 count=0 bs=1024
mkfs.ext4 -F -i 4096 -b 4096 ..../image.rootfs.ext4 -d ..../rootfs
```
Where I've put in the partition size in KiB directly in `local.conf`

This actually creates a 3GiB file (filename ending with rootfs.ext4), but the rootfs partition image put in the tegraflash folder is apparently created independent of this, as it is only a fraction of the size.

madisongh, 2019-11-08:
Interesting that `ROOTFSPART_SIZE` is not defined, since all of the machine `.conf` files in `meta-tegra` have that variable set.  It's used in tegraflash packaging to tell the NVIDIA flashing tools what the size of the rootfs partition should be.  Those tools use an Android "sparse image" format, which is why the `.img` file is much smaller.

madisongh, 2019-11-11:
> `partition size not sufficient for storing unsparsed data`
That indicates that the final size of the rootfs filesystem is too large - it won't fit into the 3GiB you've allocated for the partition.  Adjust IMAGE_ROOTFS_SIZE downward some more.

triblex, 2019-11-12:
I've managed to get the `IMAGE_ROOTFS_SIZE` adjusted low enough to flash without errors. The problem now is, when i boot up, the system starts en *maintenance* and presents me with: 

```
Press Enter for maintenance
(or press Control-D to continue):
```

Where `Control-D` does absolutely nothing. The same thing happens if I try to make the rootfs image myself with:

```
dd if=/dev/zero of=rootfs.img bs=512 count=6291456
mkfs.ext4 -j rootfs.img
loop_dev="$(sudo losetup --show -f -P rootfs.img)"
tmpdir="$(sudo mktemp -d)"
sudo mount "${loop_dev}" "${tmpdir}"
sudo cp -a "${WORKDIR}/rootfs"/* "${tmpdir}"
```

and flash it. I tried making the image smaller with a lower `count` value but the same thing happens. Any idea what this could be?

triblex, 2019-11-13:
How is the actual rootfs image created? The file in the end result tegraflash.zip? If i try to create the rootfs image myself with the exact same procedure as shown in `log.do_image_ext4`, the jetson boots into maintenance mode. I even tried flashing the resulting image of `do_image_ext4` which produced the same problem, Maintenance mode. But if i flash the default rootfs image created by tegra layer, the jetson boots normal. 

I can also see if i try to run `sudo gparted image.im` on the default image, i get 100% unallocated space and unknown filesystem. But if i run gparted on the image i created myself, there is no unallocated space (filesystem still unknown though). So how/where is it created, I can't seem to find it anywhere in the logs.

triblex, 2019-11-13:
It seems to boot fine if I create the image with
```
dd if=/dev/zero of=rootfs.ext4 seek=3145728 count=0 bs=1024
sudo mkfs.ext4 -F -i 4096 -b 4096 rootfs.ext4 -d rootfs
```

closed 2019-12-18 by triblex