OE4T/meta-tegra #202: Unable to get checksum for cuda-binaries SRC_URI entry cuda-repo-l4t-10-0-local-10.0.326_1.0-1_arm64.deb: file could not be found

Austriker, 2019-11-03:
Hi,

I am having trouble with warrior l4t r32.2 and the cuda libraries. I am running on a host ubuntu 18.04. We had it running before on an older release of the warrior branch. I have tried with a clean repo but it didn't solve it.

I get this error : 
```sh
WARNING: /home/ubuntu/tegra/build/../sources/meta-tegra/recipes-devtools/visionworks/libvisionworks_1.6.0.500n.bb: Unable to get checksum for libvisionworks SRC_URI entry libvisionworks-repo_1.6.0.500n_arm64.deb: file could not be found
WARNING: /home/ubuntu/tegra/build/../sources/meta-tegra/recipes-devtools/visionworks/libvisionworks-tracking_0.88.2.bb: Unable to get checksum for libvisionworks-tracking SRC_URI entry libvisionworks-tracking-repo_0.88.2_arm64.deb: file could not be found
WARNING: /home/ubuntu/tegra/build/../sources/meta-tegra/recipes-devtools/cuda/cuda-binaries-ubuntu1604-native_10.0.326-1.bb: Unable to get checksum for cuda-binaries-ubuntu1604-native SRC_URI entry cuda-repo-ubuntu1604-10-0-local-10.0.326-410.108_1.0-1_amd64.deb: file could not be found
WARNING: /home/ubuntu/tegra/build/../sources/meta-tegra/recipes-devtools/cuda/cuda-binaries_10.0.326-1.bb: Unable to get checksum for cuda-binaries SRC_URI entry cuda-repo-l4t-10-0-local-10.0.326_1.0-1_arm64.deb: file could not be found
WARNING: /home/ubuntu/tegra/build/../sources/meta-tegra/recipes-devtools/visionworks/libvisionworks-sfm_0.90.4.bb: Unable to get checksum for libvisionworks-sfm SRC_URI entry libvisionworks-sfm-repo_0.90.4_arm64.deb: file could not be found
WARNING: /home/ubuntu/tegra/build/../sources/meta-tegra/recipes-devtools/cuda/cuda-binaries-ubuntu1804-native_10.0.326-1.bb: Unable to get checksum for cuda-binaries-ubuntu1804-native SRC_URI entry cuda-repo-ubuntu1804-10-0-local-10.0.326-410.108_1.0-1_amd64.deb: file could not be found
WARNING: /home/ubuntu/tegra/build/../sources/meta-tegra/recipes-bsp/cboot/cboot_28.3.0.bb: Unable to get checksum for cboot SRC_URI entry cboot_src.tbz2: file could not be found
WARNING: /home/ubuntu/tegra/build/../sources/meta-tegra/recipes-devtools/gie/tensorrt_5.1.6-1.bb: Unable to get checksum for tensorrt SRC_URI entry libnvinfer5_5.1.6-1+cuda10.0_arm64.deb: file could not be found
WARNING: /home/ubuntu/tegra/build/../sources/meta-tegra/recipes-devtools/gie/tensorrt_5.1.6-1.bb: Unable to get checksum for tensorrt SRC_URI entry libnvinfer-dev_5.1.6-1+cuda10.0_arm64.deb: file could not be found
WARNING: /home/ubuntu/tegra/build/../sources/meta-tegra/recipes-devtools/gie/tensorrt_5.1.6-1.bb: Unable to get checksum for tensorrt SRC_URI entry libnvinfer-samples_5.1.6-1+cuda10.0_all.deb: file could not be found
WARNING: /home/ubuntu/tegra/build/../sources/meta-tegra/recipes-devtools/cudnn/cudnn_7.5.0.56-1.bb: Unable to get checksum for cudnn SRC_URI entry libcudnn7_7.5.0.56-1+cuda10.0_arm64.deb: file could not be found
WARNING: /home/ubuntu/tegra/build/../sources/meta-tegra/recipes-devtools/cudnn/cudnn_7.5.0.56-1.bb: Unable to get checksum for cudnn SRC_URI entry libcudnn7-dev_7.5.0.56-1+cuda10.0_arm64.deb: file could not be found
WARNING: /home/ubuntu/tegra/build/../sources/meta-tegra/recipes-devtools/cudnn/cudnn_7.5.0.56-1.bb: Unable to get checksum for cudnn SRC_URI entry libcudnn7-doc_7.5.0.56-1+cuda10.0_arm64.deb: file could not be found
Parsing recipes: 100% |###############################################################| Time: 0:00:29
Parsing of 1709 .bb files complete (0 cached, 1709 parsed). 2458 targets, 116 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies
NOTE: preferred version 10.0% of cuda-binaries-native not available (for item cuda-binaries-native)
NOTE: versions of cuda-binaries-native available: 6.5-53

Build Configuration:
BB_VERSION           = "1.42.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-poky-linux"
MACHINE              = "jetson-tx2"
DISTRO               = "poky"
DISTRO_VERSION       = "2.7.2"
TUNE_FEATURES        = "aarch64 armv8a crc"
TARGET_FPU           = ""
meta
meta-poky
meta-yocto-bsp       = "warrior:d0f73121551dc98f6924cd77952bf9ebf5ef3dd7"
meta-tegra           = "warrior-l4t-r32.2:bd8023e171b67a437443b8ef03f76aa6315be3ad"
meta-mender-core     = "warrior:0df2ee7b9108f22d249bae644cd1fee06dc19b5c"
meta-mender-tegra    = "warrior:e51eaf11461755339a42431216339184b03fd894"

Initialising tasks: 100% |############################################################| Time: 0:00:05
Sstate summary: Wanted 1864 Found 8 Missed 1856 Current 330 (0% match, 15% complete)
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
WARNING: Unable to get checksum for cuda-binaries SRC_URI entry cuda-repo-l4t-10-0-local-10.0.326_1.0-1_arm64.deb: file could not be found
WARNING: cuda-binaries-10.0.326-1-r0 do_fetch: Failed to fetch URL file://home/ubuntu/nvidia_downloads2/cuda-repo-l4t-10-0-local-10.0.326_1.0-1_arm64.deb, attempting MIRRORS if available
ERROR: cuda-binaries-10.0.326-1-r0 do_fetch: Fetcher failure: Unable to find file file://home/ubuntu/nvidia_downloads2/cuda-repo-l4t-10-0-local-10.0.326_1.0-1_arm64.deb anywhere. The paths that were searched were:
    /home/ubuntu/tegra/build/../sources/meta-tegra/recipes-devtools/cuda/cuda-binaries-10.0.326-1/poky
    /home/ubuntu/tegra/build/../sources/meta-tegra/recipes-devtools/cuda/cuda-binaries/poky
    /home/ubuntu/tegra/build/../sources/meta-tegra/recipes-devtools/cuda/files/poky
    /home/ubuntu/tegra/build/../sources/meta-tegra/recipes-devtools/cuda/cuda-binaries-10.0.326-1/jetson-tx2
    /home/ubuntu/tegra/build/../sources/meta-tegra/recipes-devtools/cuda/cuda-binaries/jetson-tx2
    /home/ubuntu/tegra/build/../sources/meta-tegra/recipes-devtools/cuda/files/jetson-tx2
    /home/ubuntu/tegra/build/../sources/meta-tegra/recipes-devtools/cuda/cuda-binaries-10.0.326-1/tegra186
    /home/ubuntu/tegra/build/../sources/meta-tegra/recipes-devtools/cuda/cuda-binaries/tegra186
    /home/ubuntu/tegra/build/../sources/meta-tegra/recipes-devtools/cuda/files/tegra186
    /home/ubuntu/tegra/build/../sources/meta-tegra/recipes-devtools/cuda/cuda-binaries-10.0.326-1/armv8a
    /home/ubuntu/tegra/build/../sources/meta-tegra/recipes-devtools/cuda/cuda-binaries/armv8a
    /home/ubuntu/tegra/build/../sources/meta-tegra/recipes-devtools/cuda/files/armv8a
    /home/ubuntu/tegra/build/../sources/meta-tegra/recipes-devtools/cuda/cuda-binaries-10.0.326-1/aarch64
    /home/ubuntu/tegra/build/../sources/meta-tegra/recipes-devtools/cuda/cuda-binaries/aarch64
    /home/ubuntu/tegra/build/../sources/meta-tegra/recipes-devtools/cuda/files/aarch64
    /home/ubuntu/tegra/build/../sources/meta-tegra/recipes-devtools/cuda/cuda-binaries-10.0.326-1/tegra
    /home/ubuntu/tegra/build/../sources/meta-tegra/recipes-devtools/cuda/cuda-binaries/tegra
    /home/ubuntu/tegra/build/../sources/meta-tegra/recipes-devtools/cuda/files/tegra
    /home/ubuntu/tegra/build/../sources/meta-tegra/recipes-devtools/cuda/cuda-binaries-10.0.326-1/cuda
    /home/ubuntu/tegra/build/../sources/meta-tegra/recipes-devtools/cuda/cuda-binaries/cuda
    /home/ubuntu/tegra/build/../sources/meta-tegra/recipes-devtools/cuda/files/cuda
[... 51 lines cut ...]
libvisionworks-sfm-repo_0.90.4_amd64.deb
libvisionworks-sfm-repo_0.90.4_arm64.deb
libvisionworks-tracking-repo_0.88.2_amd64.deb
libvisionworks-tracking-repo_0.88.2_arm64.deb
NsightSystems-linux-public-2019.4.1.10-a76094a.deb
nvidia-container-runtime_0.9.0_beta.1+docker18.09.2-1_arm64.deb
nvidia-container-runtime-hook_0.9.0_beta.1-1_arm64.deb
nvidia-docker2_0.9.0_beta.1+docker18.09.2-1_arm64.deb
NVIDIA_VisionWorks_References.zip
python3-libnvinfer_5.1.6-1+cuda10.0_arm64.deb
python3-libnvinfer-dev_5.1.6-1+cuda10.0_arm64.deb
python-libnvinfer_5.1.6-1+cuda10.0_arm64.deb
python-libnvinfer-dev_5.1.6-1+cuda10.0_arm64.deb
sdkml3_jetpack_l4t_421_b107_rev2.json
Tegra_Linux_Sample-Root-Filesystem_R32.2.0_aarch64.tbz2
Tegra_Multimedia_API_R32.2.0_aarch64.tbz2
tensorrt_5.1.6.1-1+cuda10.0_arm64.deb
tensorrt.csv
uff-converter-tf_5.1.6-1+cuda10.0_arm64.deb
visionworks.csv
```

Also tried forcing the variable `CUDA_BINARIES_NATIVE = "cuda-binaries-ubuntu1804-native"` but it doesn't solve it also.

Best regards,

madisongh, 2019-11-04:
Check your setting of `NVIDIA_DEVNET_MIRROR`... looks like you need another `/` in the prefix, to make it an absolute, rather than a relative, path name.

Austriker, 2019-11-04:
Yes that was it. Thank you.

closed 2019-11-04 by Austriker