OE4T/meta-tegra #203: Warrior kernel error multiple definition of tegra_bpmp_debugfs_add_file

Austriker, 2019-11-04:
Hi,

I am running into an error while compiling the linux kernel. I am trying to build the branch warrior-L4T-32.2.
In my config I am using GCC 7 from the contrib folder. Is this a race condition when the kernel ils built ?

Best regards

```sh
Parsing of 1676 .bb files complete (0 cached, 1676 parsed). 2402 targets, 110 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies
NOTE: preferred version 10.0% of cuda-binaries-native not available (for item cuda-binaries-native)
NOTE: versions of cuda-binaries-native available: 6.5-53
Build Configuration:
BB_VERSION           = "1.42.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-18.04"
TARGET_SYS           = "aarch64-poky-linux"
MACHINE              = "jetson-tx2"
DISTRO               = "poky"
DISTRO_VERSION       = "2.7.2"
TUNE_FEATURES        = "aarch64 armv8a crc"
TARGET_FPU           = ""
meta
meta-poky
meta-yocto-bsp       = "warrior:d0f73121551dc98f6924cd77952bf9ebf5ef3dd7"
meta-tegra
contrib              = "warrior-l4t-r32.2:bd8023e171b67a437443b8ef03f76aa6315be3ad"
meta-oe              = "warrior:a24acf94d48d635eca668ea34598c6e5c857e3f8"
NOTE: Fetching uninative binary shim from http://downloads.yoctoproject.org/releases/uninative/2.7/x86_64-nativesdk-libc.tar.xz;sha256sum=9498d8bba047499999a7310ac2576d0796461184965351a56f6d32c888a1f216
Initialising tasks: 100% |##########################################################################################################################################################################| Time: 0:00:05
Sstate summary: Wanted 2164 Found 0 Missed 2164 Current 0 (0% match, 0% complete)
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
#
# ...
#
  LD      drivers/cpufreq/built-in.o
aarch64-poky-linux-ld.bfd: drivers/cpufreq/tegra194_cpufreq.o: in function `tegra_bpmp_debugfs_add_file':
/usr/src/kernel/nvidia/include/soc/tegra/tegra_bpmp.h:72: multiple definition of `tegra_bpmp_debugfs_add_file'; drivers/cpufreq/tegra_cpufreq.o:/usr/src/kernel/nvidia/include/soc/tegra/tegra_bpmp.h:72: first defined here
/home/ubuntu/tegra/build/tmp/work-shared/jetson-tx2/kernel-source/scripts/Makefile.build:509: recipe for target 'drivers/cpufreq/built-in.o' failed
make[4]: *** [drivers/cpufreq/built-in.o] Error 1
/home/ubuntu/tegra/build/tmp/work-shared/jetson-tx2/kernel-source/scripts/Makefile.build:649: recipe for target 'drivers/cpufreq' failed
make[3]: *** [drivers/cpufreq] Error 2
/home/ubuntu/tegra/build/tmp/work-shared/jetson-tx2/kernel-source/Makefile:1105: recipe for target 'drivers' failed
make[2]: *** [drivers] Error 2
make[2]: *** Waiting for unfinished jobs....
  CC      net/ipv6/addrlabel.o
  CC      fs/compat.o
  CC      kernel/module.o
#
# ...
#
  LD      net/wireless/cfg80211.o
  LD      net/wireless/built-in.o
  LD      net/built-in.o
ERROR: oe_runmake failed
Makefile:169: recipe for target 'sub-make' failed
make[1]: *** [sub-make] Error 2
Makefile:24: recipe for target '__sub-make' failed
make: *** [__sub-make] Error 2
WARNING: exit code 1 from a shell command.
ERROR: Function failed: do_compile (log file is located at /home/ubuntu/tegra/build/tmp/work/jetson_tx2-poky-linux/linux-tegra/4.9+gitAUTOINC+3dcbed5d27-r0/temp/log.do_compile.9011)
```

Austriker, 2019-11-05:
Tried also with the linaro toolchain. Same issue.

Austriker, 2019-11-05:
Here is the compile debug line
```sh
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: make -j 8 HOSTCC=gcc  -isystem/home/ubuntu/tegra/build/tmp/work/jetson_tx2-poky-linux/linux-tegra/4.9+gitAUTOINC+3dcbed5d27-r0/recipe-sysroot-native/usr/include -O2 -pipe -L/home/ubuntu/tegra/build/tmp/work/jetson_tx2-poky-linux/linux-tegra/4.9+gitAUTOINC+3dcbed5d27-r0/recipe-sysroot-native/usr/lib                         -L/home/ubuntu/tegra/build/tmp/work/jetson_tx2-poky-linux/linux-tegra/4.9+gitAUTOINC+3dcbed5d27-r0/recipe-sysroot-native/lib                         -Wl,--enable-new-dtags                         -Wl,-rpath-link,/home/ubuntu/tegra/build/tmp/work/jetson_tx2-poky-linux/linux-tegra/4.9+gitAUTOINC+3dcbed5d27-r0/recipe-sysroot-native/usr/lib                         -Wl,-rpath-link,/home/ubuntu/tegra/build/tmp/work/jetson_tx2-poky-linux/linux-tegra/4.9+gitAUTOINC+3dcbed5d27-r0/recipe-sysroot-native/lib                         -Wl,-rpath,/home/ubuntu/tegra/build/tmp/work/jetson_tx2-poky-linux/linux-tegra/4.9+gitAUTOINC+3dcbed5d27-r0/recipe-sysroot-native/usr/lib                         -Wl,-rpath,/home/ubuntu/tegra/build/tmp/work/jetson_tx2-poky-linux/linux-tegra/4.9+gitAUTOINC+3dcbed5d27-r0/recipe-sysroot-native/lib                         -Wl,-O1 -Wl,--allow-shlib-undefined -Wl,--dynamic-linker=/home/ubuntu/tegra/build/tmp/sysroots-uninative/x86_64-linux/lib/ld-linux-x86-64.so.2 HOSTCPP=gcc  -E LIBGCC= Image CC=aarch64-poky-linux-gcc   -fuse-ld=bfd -fdebug-prefix-map=/home/ubuntu/tegra/build/tmp/work/jetson_tx2-poky-linux/linux-tegra/4.9+gitAUTOINC+3dcbed5d27-r0=/usr/src/debug/linux-tegra/4.9+gitAUTOINC+3dcbed5d27-r0                     -fdebug-prefix-map=/home/ubuntu/tegra/build/tmp/work/jetson_tx2-poky-linux/linux-tegra/4.9+gitAUTOINC+3dcbed5d27-r0/recipe-sysroot=                     -fdebug-prefix-map=/home/ubuntu/tegra/build/tmp/work/jetson_tx2-poky-linux/linux-tegra/4.9+gitAUTOINC+3dcbed5d27-r0/recipe-sysroot-native=                     -fdebug-prefix-map=/home/ubuntu/tegra/build/tmp/work-shared/jetson-tx2/kernel-source=/usr/src/kernel -ffile-prefix-map=/home/ubuntu/tegra/build/tmp/work/jetson_tx2-poky-linux/linux-tegra/4.9+gitAUTOINC+3dcbed5d27-r0/git=/kernel-source/  LD=aarch64-poky-linux-ld.bfd
|   GEN     ./Makefile
| scripts/kconfig/conf  --silentoldconfig Kconfig
|   CHK     include/config/kernel.release
|   UPD     include/config/kernel.release
|   GEN     ./Makefile
|   WRAP    arch/arm64/include/generated/asm/clkdev.h
|   WRAP    arch/arm64/include/generated/asm/dma.h
|   WRAP    arch/arm64/include/generated/asm/div64.h
|   WRAP    arch/arm64/include/generated/asm/emergency-restart.h
|   WRAP    arch/arm64/include/generated/asm/cputime.h
|   WRAP    arch/arm64/include/generated/asm/errno.h
|   WRAP    arch/arm64/include/generated/asm/early_ioremap.h
|   WRAP    arch/arm64/include/generated/asm/delay.h
|   WRAP    arch/arm64/include/generated/asm/hw_irq.h
|   WRAP    arch/arm64/include/generated/asm/dma-contiguous.h
|   WRAP    arch/arm64/include/generated/asm/bugs.h
|   WRAP    arch/arm64/include/generated/asm/ioctl.h
|   WRAP    arch/arm64/include/generated/asm/ioctls.h
|   WRAP    arch/arm64/include/generated/asm/local.h
|   WRAP    arch/arm64/include/generated/asm/ipcbuf.h
|   WRAP    arch/arm64/include/generated/asm/mcs_spinlock.h
|   WRAP    arch/arm64/include/generated/asm/kvm_para.h
|   WRAP    arch/arm64/include/generated/asm/irq_regs.h
|   WRAP    arch/arm64/include/generated/asm/kmap_types.h
|   WRAP    arch/arm64/include/generated/asm/mm-arch-hooks.h
|   WRAP    arch/arm64/include/generated/asm/kdebug.h
|   WRAP    arch/arm64/include/generated/asm/mman.h
|   WRAP    arch/arm64/include/generated/asm/mutex.h
|   WRAP    arch/arm64/include/generated/asm/local64.h
|   WRAP    arch/arm64/include/generated/asm/msi.h
|   WRAP    arch/arm64/include/generated/asm/poll.h
|   WRAP    arch/arm64/include/generated/asm/rwsem.h
|   WRAP    arch/arm64/include/generated/asm/msgbuf.h
|   WRAP    arch/arm64/include/generated/asm/preempt.h
|   WRAP    arch/arm64/include/generated/asm/resource.h
|   WRAP    arch/arm64/include/generated/asm/segment.h
|   WRAP    arch/arm64/include/generated/asm/sembuf.h
|   WRAP    arch/arm64/include/generated/asm/shmbuf.h
|   WRAP    arch/arm64/include/generated/asm/serial.h
|   WRAP    arch/arm64/include/generated/asm/simd.h
|   WRAP    arch/arm64/include/generated/asm/sizes.h
|   WRAP    arch/arm64/include/generated/asm/socket.h
|   WRAP    arch/arm64/include/generated/asm/sockios.h
|   WRAP    arch/arm64/include/generated/asm/swab.h
|   WRAP    arch/arm64/include/generated/asm/termios.h
|   WRAP    arch/arm64/include/generated/asm/trace_clock.h
|   WRAP    arch/arm64/include/generated/asm/termbits.h
|   WRAP    arch/arm64/include/generated/asm/types.h
|   WRAP    arch/arm64/include/generated/asm/user.h

```

madisongh, 2019-11-05:
Are you using a modified kernel configuration?  Looks like that error can happen if the kernel config does not have CONFIG_NV_TEGRA_BPMP set but does have CONFIG_DEBUG_FS set.  The stock L4T R32.2 kernel config for TX2 has both of those set, so you should not see that error at compile time unless you're using a modified configuration.

Austriker, 2019-11-05:
The only thing I have modified is by appending the kernel recipe that modifies the defconfig
I will  try building without it.
```sh
$ cat recipes-kernel/linux-tegra_4.%.bbappend
FILESEXTRAPATHS_prepend := "${THISDIR}/files:"

SRC_URI_append = "file://board_defconfig"

do_configure_prepend(){
    mv ${WORKDIR}/board_defconfig ${WORKDIR}/defconfig
}
```

Austriker, 2019-11-06:
Hi,

In fact I messed up my defconfig. That's why my kernel was not compiling.

Best regards.

closed 2019-11-06 by Austriker