OE4T/meta-tegra #204: Yocto Warrior dependency loop when enable secure boot configuration

danielwangksu, 2019-11-05:
When enable the following Yocto Warrior with meta-tegra report dependency loop
```
KERNEL_CLASSES ?= “ kernel-fitimage ”
KERNEL_IMAGETYPES_append = “ fitImage”
UBOOT_SIGN_KEYDIR = “${TOPDIR}/conf”
UBOOT_SIGN_KEYNAME = “dev”
UBOOT_SIGN_ENABLE  = “1”
```

dwalkes, 2019-11-05:
Might be related to https://github.com/madisongh/meta-tegra/commit/8c201acdb77daed866a4ad7b9f26445f80985529 I don't think this is on the warrior branch yet.

madisongh, 2019-11-05:
Yeah, I asked @danielwangksu to open this to remind me to cherry-pick the change here.  Done now.

closed 2019-11-05 by madisongh

danielwangksu, 2019-11-06:
Thanks @madisongh I will test it tomorrow!

compenguy, 2019-11-07:
@danielwangksu , would you be willing to adding a meta-tegra wiki page about enabling secure boot?  I know I would really like to see what you're doing.

madisongh, 2019-11-08:
I had already started a [wiki page](https://github.com/madisongh/meta-tegra/wiki/Secure-Boot-Support) on secure boot - please feel free to update that if needed.

danielwangksu, 2019-11-09:
@madisongh @compenguy sure I would love to share. I just finished the actual work will start asap. thanks guys.