OE4T/meta-tegra #206: Failure to load blob.bin for Jetson TX2i

Atomicchipmonk, 2019-11-11:
Hi @madisongh and @lfdmn,

First - thank you so much for the insane amount of work both of you have done. Its really impressive, and easy enough to follow!

Here is my setup: CTI Orbitty Carrier + Jetson TX2i
Poky: warrior
meta-tegra: warrior-l4t-r32.2
meta-cti: l4t-r32.2-cti-v126
--(using linux-tegra-4.9) github.com/lfdmn/linux-tegra-4.9.git:patches-32.2-cti-v126

So far I have re-worked through many things you have worked through. Some notes (and hopefully eventually PRs/wiki page updates)

meta-cti: l4t-r32.2-cti-v126 is missing a " at the end of line 11, recipes-kernel/linux/linux-tegra_4.9.bbappend

cboot.bin is missing from the final zip unless you add 'PREFERRED_PROVIDER_virtual/bootloader = "cboot-prebuilt" ' to local.conf (and appears to be required for TX2i, not only Xavier.

Line 87 of the generated tegra186-flash-helper.sh (if [ "${boardrev}" < "300" ]; then) creates the following output during "./doflash.sh"

> ./tegra186-flash/tegra186-flash-helper.sh: line 87: 300: No such file or directory

And finally this issue I have been unable to crack - when running ./doflash.sh things look 99% equivalent to when I flash with L4T, but gets to the following and dies:

```
[   9.4141 ] tegrahost_v2 --chip 0x18 --generateblob blob.xml blob.bin
[   9.4149 ] number of images in blob are 9
[   9.4153 ] blobsize is 3586648
[   9.4155 ] Added binary blob_nvtboot_recovery_cpu_sigheader.bin.encrypt of size 203312
[   9.4173 ] Added binary blob_nvtboot_recovery_sigheader.bin.encrypt of size 89360
[   9.4181 ] Added binary blob_preboot_d15_prod_cr_sigheader.bin.encrypt of size 63104
[   9.4190 ] Added binary blob_mce_mts_d15_prod_cr_sigheader.bin.encrypt of size 2082144
[   9.4196 ] Added binary blob_bpmp_sigheader.bin.encrypt of size 533904
[   9.4202 ] Added binary blob_tegra186-a02-bpmp-storm-p3489-a00-00-ta795sa-ucm1_sigheader.dtb.encrypt of size 76080
[   9.4211 ] Added binary blob_tos-trusty_sigheader.img.encrypt of size 313152
[   9.4216 ] Added binary blob_eks_sigheader.img.encrypt of size 1440
[   9.4221 ] Added binary blob_tegra186-tx2i-cti-ASG003_sigheader.dtb.encrypt of size 224000
[   9.4229 ] 
[   9.4230 ] Sending bootloader and pre-requisite binaries
[   9.4239 ] tegrarcm_v2 --download blob blob.bin
[   9.4247 ] Applet version 01.00.0000
[   9.4448 ] Sending blob
[   9.5665 ] 
Error: Return value 1
Command tegrarcm_v2 --download blob blob.bin
```

While i cannot find blob.xml and blob.bin in my folder (they appear to be ephemeral, since rerunning the commands without the python helper comes up empty), I do believe they are being created. The only _.img.encrypt part that is more than ~5% different in size is the blob_tos-trusty_sigheader.img.encrypt vs the blob_tos_sigheader.img.encrypt from the L4T (313k vs 63k).

My question - does anyone know what tegrarcm_v2 does and what the error codes mean? And any pointers to figuring out why my blob would be rejected? Has anyone besides the belena crowd gotten Yocto to run on the TX2i?

I will grab the default dev board this week and see if I am able to build for that.

Thanks again!

Chris

madisongh, 2019-11-11:
@compenguy contributed the TX2i support, he might be able to help.

That `< "300"` comparison in the flash helper definitely doesn't look right.

`tegrarcm_v2` implements the communication protocol over the USB connection.  It's binary-only, and I don't recall seeing any of the error codes documented, but it looks like it's either not seeing the TX2i on the USB, or something else early on in getting the communication established.

compenguy, 2019-11-11:
In nvidia_sdk/JetPack_4.2_Linux_P3489/Linux_for_Tegra/jetson-tx2i.conf, it includes p2771-0000.conf.common, which contains the following logic:
```if [ "${bdv}" \< "300" ]; then
    ODMDATA=0x2090000;
    uboot_build=000;
    tofab="-a00";
    ...
    kerndtbfab="-a00";
elif ...```

It turns out that the '\' character in front of the '<' is important.  With the '\', it does a numeric comparison, without it it attempts to connect the named file as stdin.

compenguy, 2019-11-11:
> cboot.bin is missing from the final zip unless you add 'PREFERRED_PROVIDER_virtual/bootloader = "cboot-prebuilt" ' to local.conf (and appears to be required for TX2i, not only Xavier.

I'm using the TX2i on the standard nvidia dev board (P2597), and I've not had to do anything special for cboot.bin to be included.  This may be an artifact of the meta-cti layer, but I don't know - I'm not at all familiar with it.

> Line 87 of the generated tegra186-flash-helper.sh (if [ "${boardrev}" < "300" ]; then) creates the following output during "./doflash.sh"
> 
> > ./tegra186-flash/tegra186-flash-helper.sh: line 87: 300: No such file or directory

That's my bad - the original nvidia scripts do the comparison with '\\<', but I mixed up my bash and only did '<', which gives the error output you see, and prevents it from correctly handling boardrevs earlier than 300.

> And finally this issue I have been unable to crack - when running ./doflash.sh things look 99% equivalent to when I flash with L4T, but gets to the following and dies:

What is blob.bin for?  I'm not familiar with it.

> My question - does anyone know what tegrarcm_v2 does and what the error codes mean? And any pointers to figuring out why my blob would be rejected? Has anyone besides the belena crowd gotten Yocto to run on the TX2i?

I find tegrarcm_v2 to be an annoying set of tools with some really quirky demands (the recovery mode communication seems to be selective about USB3 chipset/kernel version combinations, with any remotely recent kernel version just plain not working with it in my experience).  There seems to be [source code for v1 on github](https://github.com/NVIDIA/tegrarcm) (from nvidia, no less), but I've found almost no information about v2.

I've been doing development on a TX2i for a few months now using the meta-tegra layer on a P2597 dev board.  I don't know what blob.bin is, so if there are any problems there, I wouldn't have found them, but it's mostly all worked for me, including ECC enablement.  But we'll see - our custom carrier board is in development right now, so I'll be trying something much more demanding soon.

Atomicchipmonk, 2019-11-13:
Hi All,

I reverted back to using meta-tegra and the tx2i machine instead of the meta-cti layer with the updated kernel, and it boots! I still had to pull over cboot.bin manually (which i believe can be fixed with PREFERRED_PROVIDER_virtual/bootloader = "cboot-prebuilt" ), but I am at least booting - and on the orbitty to make things even sweeter. I am sure there is some hardware mismatch going on, so im going to try and figure that out soon.

Thanks again for the help!

closed 2019-11-13 by Atomicchipmonk