OE4T/meta-tegra #207: How to reduce the size of the SD-card for jetson-nano images?

Ecordonnier, 2019-11-12:
Has anyone successfully reduced the size of the SD card used for jetson-nano images? For development I am only replacing the rootfs, but in order to prepare full images I am copying 16GB of data, which is a big waste of time (it takes 20 minutes with the SD cards I have), especially since my image contains less than 1GB of real data and the rest is just empty space which will never be used for my application.

I tried modifying jetson-nano.conf (ROOTFSPART_SIZE to 2032385536 instead of 15032385536 and TEGRAFLASH_SDCARD_SIZE to 4GB instead of 16GB) but I probably did something wrong, because the rootfs is not booting and I see some error "bad geometry block count exceeds size".

triblex, 2019-11-14:
I was able to reduce the size by only changing `TEGRAFLASH_SDCARD_SIZE` in the `jetson-nano.conf` and you can also use a flashtool to skip the zeros in the resulting image when flashing the nano.

Ecordonnier, 2019-11-18:
triblex, I tried to change only TEGRAFLASH_SDCARD_SIZE to 4GB as you mentioned, This generated a .sdcard file of 5.4GB in tmp/deploy/images/jetson-nano/, which I copied using: 

> $ sudo dd bs=4M status=progress if=ogrip-robot-jetson-nano.sdcard of=/dev/sdc conv=fsync

but then I get this error  in the boot process and my rootfilesystem is not starting:

> [    5.252304] EXT4-fs (mmcblk0p1): bad geometry: block count 1407232 exceeds size of device (1045499 blocks)

Is it really working for you?

triblex, 2019-11-18:
Sounds like a partitions table problem.

I've only edited the machine config with
```
SDCARD_LAYOUT_TEMPLATE ?= "flash_l4t_t210_custom.xml"
TEGRAFLASH_SDCARD_SIZE ?= "14782M"
```
In the custom layout, I've removed the APP partition so only the 11 or so needed partitions are created. (same with the `sdcard-layout` file) Then afterwards when Yocto is done building, I add my own partitions (including the rootfs partition) to the `.sdcard` image before flashing. 

I haven't tried making the image smaller than the `14782M` as that is the size of my sdcard storage I was using.

Ecordonnier, 2019-11-18:
Thanks for your help. Which flashtool are you using in order to skip the zeroes when writing the SD card?

triblex, 2019-11-18:
I think i actually ended up using balenaEtcher but I don't think it skips zeros.

Perhaps you can use the mksparse tool from tegra like it is said [here](https://devtalk.nvidia.com/default/topic/1063535/jetson-nano/yocto-sdcard-rootfs-flash-file-size/post/5386087/#5386087) 

```
sudo mksparse -v --fillpattern=0 myimage-jetson-nano.sdcard myimage-jetson-nano-clean.sdcard
```

I have not tried it myself however.

Ecordonnier, 2019-11-27:
Thank you for your help.

closed 2019-11-27 by Ecordonnier

Ecordonnier, 2019-12-11:
Actually, I checked the file image_types_tegra.bbclass and there is some logic which causes mksparse not to be called for the jetson-nano. I don't know why:

```
  if [ "${TEGRA_SPIFLASH_BOOT}" != "1" ]; then
        mksparse -b ${TEGRA_BLBLOCKSIZE} -v --fillpattern=0 "${IMAGE_TEGRAFLASH_ROOTFS}" ${IMAGE_BASENAME}.img
  fi
```

madisongh, 2019-12-11:
The `mksparse` tool is used for flashing the image via USB; the NVIDIA flashing tools understand that format and expect it.  We don't use it for Nano builds because we're formatting and programming the SDcard separately from the USB flashing process.

Ecordonnier, 2019-12-16:
One tool I found is bmaptool (see https://www.yoctoproject.org/docs/latest/mega-manual/mega-manual.html#flashing-images-using-bmaptool ). This allows to skip the padding when flashing SD-cards and is significantly faster than using dd:

```
tmp/deploy/images/jetson-nano$ bmaptool create myimage-jetson-nano--20191216164834.sdcard > myimage.sdcard.bmap
build/tmp/deploy/images/jetson-nano$ sudo bmaptool copy --bmap myimage.sdcard.bmap myimage-jetson-nano--20191216164834.sdcard /dev/sdc
```