OE4T/meta-tegra #208: Unable to find file : cuda-repo-ubuntu1804-10-0-local-10.0.326-410.108_1.0-1_amd64.deb

FXRer, 2019-11-14:
Hi,

I'm trying to build a Yocto image for my Jetson Nano using the warrior-l4t-r32.2 branch on my Ubuntu-16.04 host machine. I set the NVIDIA_DEVNET_MIRROR = "file:///opt/nvidia/sdkm_downloads", where there's different  cuda-repo-ubuntu1804 file downloaded by the sdkmanager : 
`$ ls /opt/nvidia/sdkm_downloads/ -l
total 10575032

-rw-rw-r-- 1 pascal pascal  394475774 Th11 14 08:22 cuda-repo-cross-aarch64-10-0-local-10.0.326_1.0-1_all.deb  
-rw-rw-r-- 1 pascal pascal  887219344 Th11 13 17:47 cuda-repo-l4t-10-0-local-10.0.166_1.0-1_arm64.deb  
-rw-rw-r-- 1 pascal pascal 1000333292 Th11 14 08:25 cuda-repo-l4t-10-0-local-10.0.326_1.0-1_arm64.deb  
-rw-rw-r-- 1 pascal pascal 1859785444 Th08 10 04:17 cuda-repo-ubuntu1804-10-1-local-10.1.243-418.87.00_1.0-1_amd64.deb
` 
Cuda installation under my host machine is below : 
`lrwxrwxrwx  1 root  root      9 Th10  2 10:45 cuda -> cuda-10.0
drwxr-xr-x 16 root  root   4096 Th10  2 10:45 cuda-10.0
drwxr-xr-x 15 root  root   4096 Th09 24 11:42 cuda-10.1
`
Can anyone tell me how to correctly resolve this issue, please ?

Thanks in advance,
Pascal.

Error log : 

`WARNING: cuda-binaries-ubuntu1804-native-10.0.326-1-r0 do_fetch: Failed to fetch URL file:///opt/nvidia/sdkm_downloads/cuda-repo-ubuntu1804-10-0-local-10.0.326-410.108_1.0-1_amd64.deb, attempting MIRRORS if available
ERROR: cuda-binaries-ubuntu1804-native-10.0.326-1-r0 do_fetch: Fetcher failure: Unable to find file file:///opt/nvidia/sdkm_downloads/cuda-repo-ubuntu1804-10-0-local-10.0.326-410.108_1.0-1_amd64.deb anywhere. The paths that were searched were:
    /home/pascal/Workspace/Ref/Platform/Jetson/poky-nano/meta-tegra/recipes-devtools/cuda/cuda-binaries-ubuntu1804-10.0.326-1/poky
    /home/pascal/Workspace/Ref/Platform/Jetson/poky-nano/meta-tegra/recipes-devtools/cuda/cuda-binaries-ubuntu1804/poky
    /home/pascal/Workspace/Ref/Platform/Jetson/poky-nano/meta-tegra/recipes-devtools/cuda/files/poky
    /home/pascal/Workspace/Ref/Platform/Jetson/poky-nano/meta-tegra/recipes-devtools/cuda/cuda-binaries-ubuntu1804-10.0.326-1/
    /home/pascal/Workspace/Ref/Platform/Jetson/poky-nano/meta-tegra/recipes-devtools/cuda/cuda-binaries-ubuntu1804/
    /home/pascal/Workspace/Ref/Platform/Jetson/poky-nano/meta-tegra/recipes-devtools/cuda/files/
    /home/pascal/Workspace/Ref/Platform/Jetson/poky-nano/meta-tegra/recipes-devtools/cuda/cuda-binaries-ubuntu1804-10.0.326-1/x86-64
    /home/pascal/Workspace/Ref/Platform/Jetson/poky-nano/meta-tegra/recipes-devtools/cuda/cuda-binaries-ubuntu1804/x86-64
    /home/pascal/Workspace/Ref/Platform/Jetson/poky-nano/meta-tegra/recipes-devtools/cuda/files/x86-64
    /home/pascal/Workspace/Ref/Platform/Jetson/poky-nano/meta-tegra/recipes-devtools/cuda/cuda-binaries-ubuntu1804-10.0.326-1/
    /home/pascal/Workspace/Ref/Platform/Jetson/poky-nano/meta-tegra/recipes-devtools/cuda/cuda-binaries-ubuntu1804/
    /home/pascal/Workspace/Ref/Platform/Jetson/poky-nano/meta-tegra/recipes-devtools/cuda/files/
    /home/pascal/Workspace/Ref/Platform/Jetson/poky-nano/build/downloads
ERROR: cuda-binaries-ubuntu1804-native-10.0.326-1-r0 do_fetch: Fetcher failure for URL: 'file:///opt/nvidia/sdkm_downloads/cuda-repo-ubuntu1804-10-0-local-10.0.326-410.108_1.0-1_amd64.deb'. Unable to fetch URL from any source.
ERROR: cuda-binaries-ubuntu1804-native-10.0.326-1-r0 do_fetch: 
ERROR: cuda-binaries-ubuntu1804-native-10.0.326-1-r0 do_fetch: Function failed: base_do_fetch
`

triblex, 2019-11-14:
Did you try this (from meta-tegra wiki page):

> If you ran the SDK Manager on Ubuntu 16.04 to download the JetPack packages, you should add the following setting to your build configuration:
>
>`CUDA_BINARIES_NATIVE = "cuda-binaries-ubuntu1604-native"`

FXRer, 2019-11-14:
Hi again,

I found a work-around for myself by downloading the cuda-repo-ubuntu1604-10-0-local-10.0.326-410.108_1.0-1_amd64.deb from this link  : https://devtalk.nvidia.com/default/topic/1061244/sdkmanager-error/

closed 2019-11-14 by FXRer