OE4T/meta-tegra #209: problem building kernel in zeus with gcc 7

arrow53, 2019-11-15:
@madisongh I'm not really sure how to report this as the error chain is really long. But, when building the kernel a lot of stuff such as 
```
tmp/work-shared/jetson-tx2/kernel-source/drivers/base/regmap/regcache-rbtree.c:244:12: error: taking address of packed member of 'struct regcache_rbtree_node' may result in an unaligned pointer value [-Werror=address-of-packed-member]
|   244 |   rb_erase(&rbtree_node->node, &rbtree_ctx->root);
|       |            ^~~~~~~~~~~~~~~~~~

```
this is build core-image-minimal.

any immediate thoughts on this?

kekiefer, 2019-11-15:
I had this problem too from a while back, and forgot about it. At the time I attributed it to a mish-mash of versions of things I was using at the time, and then promptly forgot about it. But I've still got the following in a bbappend to work around this:

```
EXTRA_OEMAKE += "KCFLAGS='-Wno-packed-not-aligned -Wno-attribute-alias -Wno-stringop-truncation -Wno-stringop-overflow -Wno-tautological-compare -Wno-sizeof-pointer-memaccess -Wno-address-of-packed-member -Wno-missing-attributes'"
```

madisongh, 2019-11-15:
Which version gcc 7 are you using? I'm not seeing those warnings in either the Linaro gcc 7.2 or ported-from-thud gcc 7.3 compilers.

kekiefer, 2019-11-15:
Yeah, that's the difference. Right now I'm using whatever is in openembedded-core. So that's 9.2.0 in zeus.

arrow53, 2019-11-16:
I had this originall
```
GCCVERSION = "7.%"

DEBUG_PREFIX_MAP = "-fdebug-prefix-map=${WORKDIR}=/usr/src/debug/${PN}/${EXTENDPE}${PV}-${PR} \
                    -fdebug-prefix-map=${STAGING_DIR_HOST}= \
                    -fdebug-prefix-map=${STAGING_DIR_NATIVE}= \
                    "
```
but then removed it and still saw problems. I'm using your test-distro to grab gcc 7 @madisongh

madisongh, 2019-11-18:
I'm not sure it's really taking, since it looks like those warnings are new for gcc 9.  BTW,  gcc 7 recipes are available in the `meta-tegra/contrib` layer now, too.

I've cherry-picked some patches from linux-stable to make the kernel buildable with gcc 9.

arrow53, 2019-12-03:
thanks @madisongh I've updated to use `meta-tegra/contrib`. I see new problems
```
| /yocto/zeus-tx2/tmp/work-shared/jetson-tx2/kernel-source/include/linux/spinlock.h:367:30: error: 'lock' undeclared (first use in this function); did you mean 'lOck'?
|   return raw_spin_trylock_bh(&lock->rlock);
```
I have this at bottom of local.conf
```
GCCVERSION = "7.%"
require contrib/conf/include/gcc-compat.conf
```

and this in my bblayers.conf
```
  /opt/oe/zeus/meta-tegra/contrib \
```

madisongh, 2019-12-03:
The settings look fine... not sure why the kernel compilation is failing.  Is that the default kernel recipe from the layer with the stock kernel config?  If so, can you provide a bit more log file for context?

arrow53, 2019-12-06:
performing a clean build I don't see this problem so closing. Sorry for the hassle.

closed 2019-12-06 by arrow53