OE4T/meta-tegra #216: xavier: mount error

lfdmn, 2019-11-29:
Hi,

I just received a xavier dev kit and trying to port our tx2 image for xavier. 

After adding few jetson-xavier machine specific configurations, the image builds and I can flash. Quite smooth.

During the boot procedure I get:
```
mount: mounting   on /mnt failed: Invalid argument
```
Looks like the partition to mount on is missing.

Which file should I be looking into for this?

Thanks

madisongh, 2019-11-29:
At what point in the boot is it reporting that error?  The scripts `tegra-minimal-init` that run while booting from the initramfs use `/mnt` to temporarily mount the rootfs before calling using `switch_root` to hand off to it, looking it up by its partition label, which should be "APP" if you're using the stock L4T flash XML configuration.

If that's where it's failing, you could add `rdinit=/bin/sh` to your kernel args and then manually `set -x` and execute `/init` to see what's going on there.

lfdmn, 2019-11-29:
It's bit lengthy, but here's the console output:

```
Scanning mmc 0:1...
Found /boot/extlinux/extlinux.conf
Retrieving file: /boot/extlinux/extlinux.conf
304 bytes read in 99 ms (2.9 KiB/s)
Boot Options
1:	primary Image-4.4.38-l4t-r28.2+g0b02e8b
Enter choice: 1:	primary Image-4.4.38-l4t-r28.2+g0b02e8b
Retrieving file: /boot/initrd
2027086 bytes read in 138 ms (14 MiB/s)
Retrieving file: /boot/Image-4.4.38-l4t-r28.2+g0b02e8b
20712784 bytes read in 603 ms (32.8 MiB/s)
append: console=ttyS0,115200 memtype=0 video=tegrafb no_console_suspend=1 earlycon=uart8250,mmio32,0x03100000 nvdumper_reserved=0x2772e0000 gpt tegraid=18.1.2.0.0 tegra_keep_boot_clocks maxcpus=6 boot.slot_suffix= boot.ratchetvalues=0.2.1 androidboot.serialno=0421118034785 bl_prof_dataptr=0x10000@0x277040000 sdhci_tegra.en_boot_part_access=1 root=/dev/mmcblk0p1 rw rootwait vmalloc=256M cma=128M coherent-pool=96M
## Flattened Device Tree blob at 92000000
   Booting using the fdt blob at 0x92000000
   reserving fdt memory region: addr=80000000 size=10000
   Using Device Tree in place at 0000000092000000, end 0000000092038ed2

Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x100
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.4.38-l4t-r28.2+g0b02e8b (oe-user@oe-host) (gcc version 6.4.0 (GCC) ) #3 SMP PREEMPT Tue Jun 26 15:55:27 EEST 2018
[    0.000000] Boot CPU: AArch64 Processor [411fd073]
[    0.000000] earlycon: Early serial console at MMIO32 0x3100000 (options '')
[    0.000000] bootconsole [uart0] enabled
[    0.000000] Reserved memory: initialized node ramoops_carveout, compatible id nvidia,ramoops
[    0.000000] cma: Reserved 128 MiB at 0x00000000f8000000
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.0 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] PERCPU: Embedded 17 pages/cpu @ffffffc1f6643000 s31256 r8192 d30184 u69632
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 2019936
[    0.000000] Kernel command line: console=ttyS0,115200 memtype=0 video=tegrafb no_console_suspend=1 earlycon=uart8250,mmio32,0x03100000 nvdumper_reserved=0x2772e0000 gpt tegraid=18.1.2.0.0 tegra_keep_boot_clocks maxcpus=6 boot.slot_suffix= boot.ratchetvalues=0.2.1 androidboot.serialno=0421118034785 bl_prof_dataptr=0x10000@0x277040000 sdhci_tegra.en_boot_part_access=1 root=/dev/mmcblk0p1 rw rootwait vmalloc=256M cma=128M coherent-pool=96M
[    0.000000] PID hash table entries: 4096 (order: 3, 32768 bytes)
[    0.000000] Dentry cache hash table entries: 1048576 (order: 11, 8388608 bytes)
[    0.000000] Inode-cache hash table entries: 524288 (order: 10, 4194304 bytes)
[    0.000000] Memory: 7908912K/8208384K available (11384K kernel code, 1795K rwdata, 5760K rodata, 1152K init, 688K bss, 168400K reserved, 131072K cma-reserved)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vmalloc : 0xffffff8000000000 - 0xffffffbdbfff0000   (   246 GB)
[    0.000000]     vmemmap : 0xffffffbdc0000000 - 0xffffffbfc0000000   (     8 GB maximum)
[    0.000000]               0xffffffbdc2000000 - 0xffffffbdc9da0000   (   125 MB actual)
[    0.000000]     fixed   : 0xffffffbffa7fd000 - 0xffffffbffac00000   (  4108 KB)
[    0.000000]     PCI I/O : 0xffffffbffae00000 - 0xffffffbffbe00000   (    16 MB)
[    0.000000]     modules : 0xffffffbffc000000 - 0xffffffc000000000   (    64 MB)
[    0.000000]     memory  : 0xffffffc000000000 - 0xffffffc1f6800000   (  8040 MB)
[    0.000000]       .init : 0xffffffc00113f000 - 0xffffffc00125f000   (  1152 KB)
[    0.000000]       .text : 0xffffffc000080000 - 0xffffffc00113f000   ( 17148 KB)
[    0.000000]       .data : 0xffffffc001280000 - 0xffffffc001440d50   (  1796 KB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=6, Nodes=1
[    0.000000] Preemptible hierarchical RCU implementation.
[    0.000000] 	Build-time adjustment of leaf fanout to 64.
[    0.000000] 	RCU restricting CPUs from NR_CPUS=64 to nr_cpu_ids=6.
[    0.000000] RCU: Adjusting geometry for rcu_fanout_leaf=64, nr_cpu_ids=6
[    0.000000] NR_IRQS:64 nr_irqs:64 0
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] Architected cp15 timer(s) running at 31.25MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xe6a171046, max_idle_ns: 881590405314 ns
[    0.000003] sched_clock: 56 bits at 31MHz, resolution 32ns, wraps every 4398046511088ns
[    0.009425] Console: colour dummy device 80x25
[    0.014087] kmemleak: Kernel memory leak detector disabled
[    0.019826] Calibrating delay loop (skipped), value calculated using timer frequency.. 62.50 BogoMIPS (lpj=125000)
[    0.030650] pid_max: default: 32768 minimum: 301
[    0.035516] Security Framework initialized
[    0.039825] Mount-cache hash table entries: 16384 (order: 5, 131072 bytes)
[... 3030 lines cut ...]
        </partition>
        <partition name="UDA" type="data">
            <allocation_policy> sequential </allocation_policy>
            <filesystem_type> basic </filesystem_type>
            <file_system_attribute> 0 </file_system_attribute>
            <allocation_attribute> 0x808 </allocation_attribute>
            <percent_reserved> 0 </percent_reserved>
            <align_boundary> 4096 </align_boundary>
[... cut ...]
            <allocation_policy> sequential </allocation_policy>
            <filesystem_type> basic </filesystem_type>
            <size> 0xFFFFFFFFFFFFFFFF </size>
            <file_system_attribute> 0 </file_system_attribute>
            <allocation_attribute> 8 </allocation_attribute>
            <percent_reserved> 0 </percent_reserved>
        </partition>
    </device>
</partition_layout>
```
I guess `rdinit=/bin/sh` involves a full rebuild. 

I'll put that to run and check again on Monday.

Thanks

madisongh, 2019-11-29:
> I guess `rdinit=/bin/sh` involves a full rebuild.

You should just be able to set `KERNEL_ARGS_append = " rdinit=/bin/sh"` in your local.conf... shouldn't require a full rebuild, but you will need to re-flash.

BTW, which branch are you working off of?

lfdmn, 2019-11-29:
Alright, thanks for the tip.

I'm on warrior at the moment.

lfdmn, 2019-12-02:
What `rdinit=/bin/sh ` should do?

I've added it but I don't get any shell and the system boots as without.

My kernel arguments are 
```Linux Cmdline: console=ttyTCU0,115200 console=ttyTCU0,115200 console=tty0 OS=l4t fbcon=map:0 rdinit=/bin/sh video=tegrafb no_console_suspend=1 earlycon=tegra_comb_uart,mmio32,0x0c168000 gpt tegra_fbmem=0x140000@0xa0697000 lut_mem=0x20```

And after checking, my kernel version is
```Linux version 4.9.140-l4t-r32.2+g3dcbed5d27df (oe-user@oe-host) (gcc version 7.3.0 (GCC) ) #1 SMP PREEMPT Fri Nov 29 15:38:06 UTC 2019```

madisongh, 2019-12-03:
It should bring up a shell when the kernel first starts - in the initial ramdisk.  Do you have a keyboard and monitor attached?  Since there's a `console=tty0` in the kernel command line, the shell prompt will come up on the graphical console.  If you only have the serial console attached, try adding `KERNEL_ARGS_remove = "console=tty0"` in your local.conf to get rid of the reference to the graphical console.

Hmm.  I could swear I saw '4.4' when I looked at that log before.

lfdmn, 2019-12-03:
Yes you did see 4.4 =) minicom just kept writing to an old TX2 file... and wasn't careful enough. I updated the lengthy post.

I basically only have Xavier dev kit with USB C port connected to my PC and minicom on ttyUSB3. I have a monitor connected too.

I'll try what you suggested, I'd like to learn to make that bit work.

In the meanwhile I think I've just discovered the culpitt.

[boot2qt ditro layer](https://code.qt.io/cgit/yocto/meta-boot2qt.git/) is overwriting `init-boot.sh` to add nfs boot option. See
```sh
opt="rw"
wait=""
nfsroot=""
nfsopts=""
for bootarg in `cat /proc/cmdline`; do
    case "$bootarg" in
    root=*) rootdev="${bootarg##root=}" ;;
    nfsroot=*)
            nfsroot=$(echo ${bootarg##nfsroot=} | cut -d ',' -f 1)
            nfsopts=$(echo ${bootarg##nfsroot=} | cut -d ',' -f 2-)
            nfsopts=${nfsopts##${nfsroot}}
            ;;
    ro) opt="ro" ;;
    rootwait) wait="yes" ;;
    esac
done
if [ -n "$wait" -a ! -b "${rootdev}" ]; then
    echo "Waiting for ${rootdev}..."
    count=0
    while [ $count -lt 25 ]; do
    test -b "${rootdev}" && break
    sleep 0.1
    count=`expr $count + 1`
    done
fi
echo "Mounting ${rootdev}..."
if [ "$rootdev" = "/dev/nfs" ]; then
    echo "Using NFS to boot..."
    mount -t nfs -o "${opt},${nfsopts}" "${nfsroot}" /mnt || exec sh
else
    mount -t ext4 -o "$opt" "${rootdev}" /mnt || exec sh
fi
echo "Switching to rootfs on ${rootdev}..."
mount --move /sys  /mnt/sys
mount --move /proc /mnt/proc
mount --move /dev  /mnt/dev
exec switch_root /mnt /sbin/init
```
and that script doesn't use the `platform-preboot` script added for xavier (not officially a supported board).

I'll try to merge these 2. 

The core-image-initial image with only meta-tegra and poky builds and boots fine.

lfdmn, 2019-12-05:
Fixed by adding `platform-preboot` script execution

closed 2019-12-05 by lfdmn