OE4T/meta-tegra #218: enable eglfs for qt5

orhaneee, 2019-12-04:
Hi all,

It may be a silly question but I couldn't figure out how to enable opengl/egl platform backend for meta-qt5 on core-image-base.

Thank you for your help.

kekiefer, 2019-12-04:
Try setting
```
export QT_QPA_EGLFS_INTEGRATION=eglfs_kms_egldevice
export QT_QPA_PLATFORM=eglfs
```
In your environement

orhaneee, 2019-12-05:
Thank you for your answer, but eglfs is not recognized as a platform thus no eglfs integration.

kekiefer, 2019-12-05:
Ah, gotcha. You probably need to add `eglfs` and `kms` options to `PACKAGECONFIG` for qtbase, either in your local.conf or in a bbappend.

orhaneee, 2019-12-05:
I have created a new layer with `qtbase_%.bbappend` as required for Qt documentation, but if I log do_configure part it says, 
```
QPA backends:
  DirectFB ............................... no
  EGLFS .................................. no
  LinuxFB ................................ no
  VNC .................................... yes
  Mir client ............................. no
  XCB:
    Using system-provided XCB libraries .. yes
    XCB XKB .............................. yes
    XCB XInput ........................... yes
    Native painting (experimental) ....... no
    GL integrations:
      GLX Plugin ......................... yes
        XCB GLX .......................... yes
      EGL-X11 Plugin ..................... yes
```

I have also defined `DISTRO_FEATURES_append = " opengl gl"` and `DISTRO_FEATURES_remove = "X11 wayland"` in my local.conf

kekiefer, 2019-12-05:
So you probably need a `PACKAGECONFIG_append_tegra = " eglfs kms"` in that bbappend too.

lfdmn, 2019-12-05:
@orhaneee I had a similar issue. See [https://bugreports.qt.io/browse/QTBUG-78789](https://bugreports.qt.io/browse/QTBUG-78789)

If you use qt5 prior to the fix mentioned, you need to add a folder with your machine name and reuse the oe-device-extra.pri file from jetson-tx2 folder for example.
[https://code.qt.io/cgit/yocto/meta-boot2qt.git/tree/meta-boot2qt-distro/recipes-qt/qt5/qtbase?h=thud](https://code.qt.io/cgit/yocto/meta-boot2qt.git/tree/meta-boot2qt-distro/recipes-qt/qt5/qtbase?h=thud)

That is is you use boot2qt distro.

orhaneee, 2019-12-05:
@lfdmn wow thank you. I fixed it..

closed 2019-12-05 by orhaneee