OE4T/meta-tegra #219: l4tbr0 for nano

orhaneee, 2019-12-09:
Hi everyone,

How can I enable USB ethernet bridge i.e l4tbr0 on `core-image-base` for `jetson-nano`?

Thank you.

madisongh, 2020-01-04:
We don't currently have any recipes for installing the `/opt/nvidia/l4t-usb-device-mode` scripts and related files from the BSP package. I think that's all it would take, but I haven't used that feature myself.

madisongh, 2020-02-05:
I ran some tests this week with getting the device mode support integrated, and got it to work, more or less. The scripts NVIDIA provides weren't completely reliable and make several assumptions about the OS environment having everything that their Ubuntu-based distro has.  I'm not crazy about the hacks I had to do to the scripts, so when I get a chance I'll see if there's a better way.

madisongh, 2020-02-08:
I have added an `l4t-usb-device-mode` recipe that implements most of what the L4T scripts provide but in a way that is hopefully a bit more suited to an OE-based system.

closed 2020-02-08 by madisongh