OE4T/meta-tegra #225: Static IP in /etc/network/interfaces not taking effect

mwcoop, 2019-12-14:
Using a distro based on madisongh/test-distro, I'm finding that /etc/network/interfaces no longer takes effect. My eth0 interface is still using DHCP. I am using systemd as my startup manager.

Have things moved to a new mechanism for configuring interfaces? I see things like dhcpcd and udhcpc aroudn Google. Any pointers on how to get this configured would be greatly appreciated!

madisongh, 2019-12-14:
OE-Core started defaulting to using systemd-networkd, so I did the same in my test distro.  Take a look at the [systemd.network](https://www.freedesktop.org/software/systemd/man/systemd.network.html#) man page for info on the settings.

mwcoop, 2019-12-14:
Any particular recipe that I should look at using to overwrite the original configuration?

madisongh, 2019-12-14:
`systemd-conf` - there's already a [bbappend for it here](https://github.com/madisongh/meta-tegra/blob/master/recipes-core/systemd/systemd-conf_%25.bbappend) where we drop in a Tegra-specific copy of `wired.network`.

mwcoop, 2019-12-30:
Oddly, I don't see the wired.network file being put into the rootfs. The systemd-conf recipe is being built. Is there a distro setting I might be missing to get this?

madisongh, 2020-01-06:
It does get renamed as part of the install, and should be at `/lib/systemd/network/80-wired.network`.

madisongh, 2020-03-04:
@mwcoop did you get this working?  #259 looks like it might have been the same issue.

closed 2020-05-07 by madisongh