OE4T/meta-tegra #226: Can we encrypt and sign BUP payload?

danielwangksu, 2019-12-16:
https://github.com/madisongh/meta-tegra/blob/c7968882087fbddef5e10ca32ad964fa421bfe42/recipes-bsp/tegra-binaries/files/tegra186-flash-helper.sh#L207-L217

It seems to me that the following two line should be removed, because this line ` l4t_bup_gen "$flashcmd" "$spec" "$fuselevel" t186ref "$keyfile" 0x18 || exit 1` gonna add `$keyfile` to $flashcmd in `l4t_bup_gen.func`.

Also, when I run `doflash.sh --bup -u [KEY_PATH_RSA] -v [KEY_PATH_SBK]` with both `-u and -v`, `tegraflash.py` only generates `encrypted_signed` folder without `signed` folder. Which cause `l4t_bup_gen.func` fail to find `signed` folder.

`[ -z "$keyfile" ] || flashcmd="${flashcmd} --key \"$keyfile\""
 [ -z "$sbk_keyfile" ] || flashcmd="${flashcmd} --encrypt_key \"$sbk_keyfile\""`

Question: for generating BUP payload can we use encryption and signing? or only signing?

madisongh, 2019-12-24:
The BUP payload *can* be encrypted and signed, but the full setup for making that work only happens within the build - see the BUP-related functions in `image_types_tegra.bbclass`.  The tegraflash package generated by the build only works for flashing, not for generating BUP payloads outside of the bitbake build.  Probably could be added, maybe by shifting some of the logic/hacks in the `oe_make_bup_payload_xxx` functions into a `makebup.sh` script or into the flash helper script itself.

danielwangksu, 2019-12-30:
Thank you for your response. This is related to https://github.com/madisongh/meta-tegra/pull/222. I'm trying to make encrypted/signed BUP-payload outside of bitbake build. After making the changes in the patch (moving `rollback_parser.py` to `rollback` folder) and using the follow script (inspired by you doflash.sh), I can generated signed BUP-payload for tegra186. But cannot encrypt it. If I remember it correct, it is due to `l4t_bup_gen.func` looking for `signed` folder (line 129), but instead of `encrypted_signed` folder. I will see if I can figure it out. Thank you for your response!

```
#!/bin/sh
export BOARDID="3489"
export FAB="300"
export fuselevel=fuselevel_production
export MACHINE=jetson-tx2i
export localbootfile="boot.img"
# BUP generator really wants to use 'boot.img' for the LNX
ln -sf ./u-boot-jetson-tx2i.bin ./boot.img
sed -i -e's,u-boot-jetson-tx2i.bin,boot.img,' flash.xml.in

./tegra186-flash-helper.sh flash.xml.in --bup tegra186-quill-p3489-1000-a00-00-ucm1.dtb jetson-tx2i.cfg 0x1090000 "$@"
```

madisongh, 2019-12-30:
Looks like you didn't back-port my patch to `l4t_bup_gen.func` that enables the encrypt+signed functionality. The stock version that NVIDIA provides is, in fact, missing that support.

danielwangksu, 2019-12-30:
Oh I see, thank you very much for the feedback. I will rework on the backport patch

closed 2020-01-05 by danielwangksu