OE4T/meta-tegra #229: Request Update to L4T 32.3.1

JW94, 2019-12-26:
Merry Christmas!

L4T 32.3.1 was released recently and I would like to know, when it's included in your meta-layer. 
I would highly appreciate it!

Thank you in advance!

madisongh, 2019-12-27:
I started looking at it, and am hoping to make some time over the holidays to get through the updates both for the BSP and the updated JetPack components.  I can't promise a specific target date, though.

madisongh, 2020-01-02:
I've got a [work-in-progress branch going](https://github.com/madisongh/meta-tegra/tree/wip-zeus-l4t-r32.3.1).  What's there appears to work OK so far, but I have some further updates to make to account for some of the bootloader and flashing changes that NVIDIA made.  Once I get past those, I'll merge the changes in to master and create a zeus-based branch.  Will keep you posted.

JW94, 2020-01-08:
> I've got a [work-in-progress branch going](https://github.com/madisongh/meta-tegra/tree/wip-zeus-l4t-r32.3.1). What's there appears to work OK so far, but I have some further updates to make to [...]

Thank you very much for your effort!

madisongh, 2020-01-09:
I have merged the WIP into master and a new [zeus-l4t-r32.3.1 branch.](https://github.com/madisongh/meta-tegra/tree/zeus-l4t-r32.3.1)

I've also put a [notes page up on the wiki](https://github.com/madisongh/meta-tegra/wiki/L4T-R32.3.1-Notes).

paroque28, 2020-01-17:
Hi @madisongh ,
I see that the master branch is using the kernel source from madisongh/linux-tegra-4.9 and branch 'patches-l4t-r32.3.1'.

Unfortunately, this branch has many many bugs and warnings. And yocto won't let me compile the kernel with these warnings. Mostly uninitialized variables and tautologies. And all of them are inside the folder Nvidia.

I was trying to fix these, one by one but I found that the branch 'patches-l4t-r32.2.1' is 118 commits ahead and has these fixed.

Is there any reason why you are doing this?

paroque28, 2020-01-17:
Well, actually its 'patches-l4t-r32.3.1' the branch that is ahead. 

And now I understand,  the newest branch introduced more drivers to the source code.

See for example:
https://github.com/madisongh/linux-tegra-4.9/blob/patches-l4t-r32.3.1/nvidia/drivers/net/wireless/realtek/rtl8812au/hal/rtl8812a/usb/rtl8812au_xmit.c#L433

There is a bug there:

((rtw_read32(adapter, REG_PKTBUF_DBG_CTRL) & BIT23) == 1) will never be equal to 1.
Masking a value and comparing to one will only work if the mask itself is equal to one (which is not the case here). Comparing to zero works for any mask.

This error is fixed on other repositories: 
https://github.com/aircrack-ng/rtl8812au/issues/308

madisongh, 2020-01-17:
I'm guessing you're using gcc 9 out of OE-Core rather than gcc 7.  Thanks for reminding me that I need to do a build pass with the newer compiler and resolve any new compatibility issues that have cropped up with the new BSP.

My test builds so far have all been done with gcc 7.  If you plan on using CUDA, you'll need to use gcc 7 as well, as mentioned in the [README](https://github.com/madisongh/meta-tegra/blob/master/README.md). If not, give me a day or two to fix up any gcc 9 build issues.

paroque28, 2020-01-17:
Oh, you are right @madisongh  I am using GCC 9. I don't plan to use CUDA only TensorFlow but I will switch to GCC 7 just in case.

Thank you very much

madisongh, 2020-01-19:
I have run through some test builds with GCC 9 on master and fixed the compilation errors in the kernel. I didn't find any other compilation errors in my sato image builds.

Will close this now as R32.3.1 is available on both master and the zeus-l4t-r32.3.1 branches.

closed 2020-01-19 by madisongh

paroque28, 2020-01-28:
Thanks, it works well now