OE4T/meta-tegra #230: Jetson and docker

triblex, 2020-01-03:
I've been successful in using docker on my Jetson Nano production module, but it appears that the `Nvidia Container Runtime` allows some GPU access that are otherwise unavailable. I can see the Nvidia Container Runtime is available in the binaries from the Nvidia SDK manager, but how to i get the binary installed into the image? Is there a recipe i can bitbake or is this not officially available from the meta-tegra layer? 

Thanks!

madisongh, 2020-01-04:
There are no recipes currently.  I see the various container-related deb packages in the SDK Manager, but I don't know if it's just a matter of unpacking them and installing the relevant files, or whether some tweaks might be needed to deal with environment differences (e.g., Debian/Ubuntu rootfs layout vs. the plainer layout we use with OE/Yocto builds).  If someone were willing to take a stab at putting some recipes together for this, that would be great.

triblex, 2020-01-09:
I have (successfully) installed Nvidia Container Runtime manually buy extracting and copying the necessary files from the following .deb packages from the Nvidia SDK-manager:
- libnvidia-container0_0.9.0_beta.1_arm64.deb
- libnvidia-container-tools_0.9.0_beta.1_arm64.deb
- nvidia-container-runtime_3.1.0-1_arm64.deb
- nvidia-container-runtime-hook_0.9.0_beta.1-1_arm64.deb

Then I've added a json file to `/etc/daemon.json` with the following content:
```
{
    "runtimes": {
        "nvidia": {
            "path": "/usr/bin/nvidia-container-runtime",
            "runtimeArgs": []
        }
    }
}
```

After which the `docker info` command shows 
```
...
Runtimes: nvidia runc
...
```

Which indicates the nvidia container runtime is available. Running a docker with `--runtime=nvidia` runs without problems. However, when i try to run CUDA samples like the `nbody` example (based of [this](https://ngc.nvidia.com/catalog/containers/nvidia:l4t-base/tags) nvidia base image) , it returns with the error `Error: only 0 Devices available, 1 requested.  Exiting.` which indicates that the container doesn't have access to GPU acceleration via the Nvidia Container Runtime.

Running a deviceQuery returns with the error:

```
cudaGetDeviceCount returned 35
-> CUDA driver version is insufficient for CUDA runtime version
Result = FAIL
```

Which further agree with the suspicion that the Nvidia Container Runtime isn't working as intended.

I've tried a lot of ways to solve this, but I seem to be pretty stuck at this point.

triblex, 2020-01-23:
I've found `.cvs` files in `/etc/nvidia-container-runtime/host-files-for-container.d/` that contains files that are merged with the container when running a docker image with nvidia as runtime. I've tried getting all of the files stated in the list but eventually no matter what I've tried with this, an error occours stating that there is a `nvidia-container-cli: mount error: (null)`.

More on this from [this post](https://devtalk.nvidia.com/default/topic/1070320/cuda-setup-and-installation/nvidia-container-runtime-on-jetson-nano-using-yocto/post/5422425/#5422425) from nvidia devtalk.

triblex, 2020-01-23:
I've managed to get CUDA ready GPU acceleration access in the container.. FINALLY! Some amount of lib/dir/sym are needed to be merged into the container. This can be done by creating `.csv` files in the directory `/etc/nvidia-container-runtime/host-files-for-container.d/` (though I am not at this point sure which are needed) Once the needed files has been listed, the docker image can be run with nvidia runtime and the access should be available in the container.

madisongh, 2020-01-24:
Great!  Any chance you could turn this into a recipe?

paroque28, 2020-01-27:
Hi @triblex,
I need to get containers running with CUDA support and turn it into a recipe. How did you manage to make it work?

triblex, 2020-01-28:
> 
> 
> Great! Any chance you could turn this into a recipe?

Sure, I'll take a look at it. 

> 
> 
> I need to get containers running with CUDA support and turn it into a recipe. How did you manage to make it work?

You need some `.csv` files that merges listed files into your container running Nvidia Container Runtime. You can find them in the default Nvidia SDK-manager Ubuntu rootfs under `/etc/nvidia-container-runtime/host-files-for-container/` You can look at them and then make your own `.csv` files or you can copy paste the default ones, but you need to change a lot of the paths, as they are not the same as in the standard Yocto build rootfs.

>
>
> Is it needed to compile with GCC7?

You can read about this in the [wiki section](https://github.com/madisongh/meta-tegra/wiki/Compatibility-notes).

paroque28, 2020-01-29:
@madisongh which license should one use for these recipes?

madisongh, 2020-01-29:
Looks like multiple licenses are potentially involved.  See what it says in the README.md and NOTICE files - the GPL license in COPYING applies if you link against libelf.

paroque28, 2020-01-31:
So, I got libnvidia-container recipe working. I will now do the recipe for nvidia-container-runtime

mclayton7, 2020-02-01:
@paroque28 thanks for working on this, I'm interested in it as well. Have you made this recipe available on github, and will the libnvidia-container/nvidia-container-runtime replace a recipe like meta-virtualization from open-embedded? Thanks!

triblex, 2020-02-03:
@paroque28 Will you make these recipes available? If yes, then I won't start spending time on making them from scratch myself ;)

paroque28, 2020-02-03:
@mclayton7 @triblex I will upload them today afternoon

paroque28, 2020-02-04:
I got nvidia-container-toolkit recipe building 

```bash
rpm -ql tmp-glibc/work/aarch64-wrs-linux/nvidia-container-toolkit/1.0.5-r0/deploy-rpms/aarch64/nvidia-container-toolkit-1.0.5-r0.aarch64.rpm
error: cannot open Packages database in /var/lib/rpm
error: cannot open Packages database in /var/lib/rpm
/etc
/etc/nvidia-container-runtime
/etc/nvidia-container-runtime/config.toml
/usr
/usr/bin
/usr/bin/nvidia-container-toolkit
/usr/libexec
/usr/libexec/oci
/usr/libexec/oci/hooks.d
/usr/libexec/oci/hooks.d/oci-nvidia-hook
/usr/share
/usr/share/licenses
/usr/share/licenses/nvidia-container-toolkit-1.0.5
/usr/share/licenses/nvidia-container-toolkit-1.0.5/LICENSE
/usr/share/oci
/usr/share/oci/hooks.d
/usr/share/oci/hooks.d/oci-nvidia-hook.json
```

Will try tomorrow and upload if works

triblex, 2020-02-04:
@paroque28 I can see you are making the recipes from source, which I am 100% for. But using the pre-configured and pre-build nvidia-container debian packages for the Jetson Nano that is downloaded alongside all the CUDA stuff is also an option. That is what the CUDA recipes are doing (Correct me if I'm wrong). These packages and their versions are pretty much guaranteed to work with one another. 

I made some quick recipes to use these pre-build debian packages from the NVIDIA SDK manager and they work just fine. But if you can get it all to work together through source, then by all means, I would actually prefer it. It seems like you almost got it done, I just wanted to mention this.

paroque28, 2020-02-04:
Thanks, @triblex I actually based my work on the original .deb files provided by Nvidia so I didn't miss any file.

What docker image did you use for testing? The nvidia ones https://hub.docker.com/r/nvidia/cuda are only for amd64 and I am using aarch64

paroque28, 2020-02-04:
Hi all,  I am getting error:
```bash
docker: Error response from daemon: could not select device driver "" with capabilities: [[gpu]].
```

I am not sure if I installed correctly the driver for CUDA.

Can you confirm that the device driver is `cuda-driver` ? I also added `cuda-toolkit` just in case but still I cannot see the command `nvidia-smi`

triblex, 2020-02-05:
@paroque28 

> What docker image did you use for testing?

I am using [this](https://ngc.nvidia.com/catalog/containers/nvidia:l4t-base) docker image to test with. It's an official base L4T docker image from NVIDIA.

> ... but still I cannot see the command `nvidia-smi`

Unfortunately, `nvidia-smi` command [isn't supported](https://devtalk.nvidia.com/default/topic/1055327/jetson-nano/gpu-usage-info-nvidia-smi-is-not-there-/post/5349846/#5349846) on Tegra-based platforms at this time.

paroque28, 2020-02-06:
Hi @triblex ,

Thanks for your reply it was very helpful!
I got almost everything working the only thing that I am needing now is the host-files-for-container.d files. I have this folder empty at the moment. Can you please do an `ls` so that I know which files to include?

Thanks,
Pablo

paroque28, 2020-02-06:
I cannot seem to find l4t.csv, where did you get this csv?

I am generating these files from scratch since the patch might change depending on yocto installation.

The thing is that I only have four of them: 
- cuda.csv
- cudnn.csv
- tensorrt.csv
- visionworks.csv

madisongh, 2020-02-06:
@paroque28 I If you look in the L4T BSP, in `nv_tegra/config.tbz2`, you'll see an `etc/nvidia-container-runtime/host-files-for-container.d/l4t.csv`.  Is that the file you're looking for?

triblex, 2020-02-07:
@paroque28 
It IS the file that @madisongh said yes.
I took this file and modified it.

This is the `l4t.csv` file i got to work with nvidia-container-runtime:

```
dev, /dev/fb0
dev, /dev/fb1
dev, /dev/nvhost-as-gpu
dev, /dev/nvhost-ctrl
dev, /dev/nvhost-ctrl-gpu
dev, /dev/nvhost-dbg-gpu
dev, /dev/nvhost-gpu
dev, /dev/nvhost-nvdec
dev, /dev/nvhost-nvdec1
dev, /dev/nvhost-prof-gpu
dev, /dev/nvhost-vic
dev, /dev/nvhost-ctrl-nvdla0
dev, /dev/nvhost-ctrl-nvdla1
dev, /dev/nvhost-nvdla0
dev, /dev/nvhost-nvdla1
dev, /dev/nvidiactl
dev, /dev/nvmap
dev, /dev_dc_0
dev, /dev_dc_1
dev, /dev_dc_ctrl
dev, /dev/nvhost-msenc
dev, /dev/nvhost-nvenc1
dev, /dev/nvhost-nvjpg
dir, /lib/firmware21x
lib, /usr/lib/libv4l2.so.0
lib, /usr/lib/weston/desktop-shell.so
lib, /usr/lib/weston/drm-backend.so
lib, /usr/lib/weston/EGLWLInputEventExample
lib, /usr/lib/weston/EGLWLMockNavigation
lib, /usr/lib/weston/gl-renderer.so
lib, /usr/lib/weston/hmi-controller.so
lib, /usr/lib/weston/ivi-controller.so
lib, /usr/lib/weston/ivi-shell.so
lib, /usr/lib/weston/LayerManagerControl
lib, /usr/lib/weston/libilmClient.so.2.2.0
lib, /usr/lib/weston/libilmCommon.so.2.2.0
lib, /usr/lib/weston/libilmControl.so.2.2.0
lib, /usr/lib/weston/libilmInput.so.2.2.0
lib, /usr/lib/weston/libweston-6.so.0
lib, /usr/lib/weston/libweston-desktop-6.so.0
lib, /usr/lib/weston/simple-weston-client
lib, /usr/lib/weston/spring-tool
lib, /usr/lib/weston/wayland-backend.so
lib, /usr/lib/weston/weston
lib, /usr/lib/weston/weston-calibrator
lib, /usr/lib/weston/weston-clickdot
lib, /usr/lib/weston/weston-cliptest
lib, /usr/lib/weston/weston-debug
lib, /usr/lib/weston/weston-desktop-shell
lib, /usr/lib/weston/weston-dnd
lib, /usr/lib/weston/weston-eventdemo
lib, /usr/lib/weston/weston-flower
lib, /usr/lib/weston/weston-fullscreen
lib, /usr/lib/weston/weston-image
lib, /usr/lib/weston/weston-info
lib, /usr/lib/weston/weston-keyboard
lib, /usr/lib/weston/weston-launch
lib, /usr/lib/weston/weston-multi-resource
lib, /usr/lib/weston/weston-resizor
lib, /usr/lib/weston/weston-scaler
lib, /usr/lib/weston/weston-screenshooter
lib, /usr/lib/weston/weston-simple-dmabuf-egldevice
lib, /usr/lib/weston/weston-simple-egl
[... 112 lines cut ...]
lib, /usr/lib/libnvvulkan-producer.so
lib, /usr/lib/libnvwinsys.so
lib, /usr/lib/libsensors.hal-client.nvs.so
lib, /usr/lib/libsensors_hal.nvs.so
lib, /usr/lib/libsensors.l4t.no_fusion.nvs.so
lib, /usr/lib/libtegrav4l2.so
lib, /usr/lib/libv4l2_nvvidconv.so
lib, /usr/lib/libv4l2_nvvideocodec.so
lib, /usr/lib/nvidia_icd.json
lib, /etc/vulkan/icd.d/nvidia_icd.json
sym, /usr/lib/libdrm_nvdc.so
sym, /usr/lib/aarch64-linux-gnu/libv4l2.so.0.0.999999
sym, /usr/lib/aarch64-linux-gnu/libv4lconvert.so.0.0.999999
sym, /usr/lib/aarch64-linux-gnu/libv4l/plugins/nv/libv4l2_nvvidconv.so
sym, /usr/lib/aarch64-linux-gnu/libv4l/plugins/nv/libv4l2_nvvideocodec.so
sym, /usr/lib/libcuda.so
sym, /usr/lib/libnvbufsurface.so
sym, /usr/lib/libnvbufsurftransform.so
sym, /usr/lib/libnvbuf_utils.so
sym, /usr/lib/libnvid_mapper.so
lib, /usr/share/glvnd/egl_vendor.d/10-nvidia.json
lib, /lib/firmware/tegra21x/nvhost_nvdec020_ns.fw
```

Many things from here is not needed, but they will be ignored if they are missing. I haven't gone into depth of what is needed or not, but the debug log can show you what is missing from the list.

triblex, 2020-02-07:
Btw @paroque28 , the `tensorrt` and `cudnn` recipe installs the libraries in `/usr/lib/` directly, and not `/usr/lib/aarch64-linux-gnu/`, so you have to change these .csv files aswell.

paroque28, 2020-02-07:
Thanks @triblex 
I finally made it work :) Your help was so useful.
```bash
root@85f6ce9ce7d5:/# /usr/bin/cuda-samples/UnifiedMemoryStreams
GPU Device 0: "NVIDIA Tegra X1" with compute capability 5.3

Executing tasks on host / device
Task [0], thread [0] executing on device (363)
Task [1], thread [1] executing on device (791)
Task [3], thread [2] executing on device (414)
Task [2], thread [3] executing on device (753)
Task [4], thread [0] executing on device (762)
Task [5], thread [2] executing on device (756)
Task [6], thread [1] executing on device (952)
Task [7], thread [3] executing on device (763)
Task [8], thread [0] executing on device (942)
Task [9], thread [2] executing on device (853)
Task [10], thread [1] executing on device (463)
Task [11], thread [3] executing on device (375)
Task [12], thread [0] executing on device (102)
Task [13], thread [2] executing on device (937)
Task [14], thread [1] executing on host (64)
Task [15], thread [3] executing on device (538)
Task [16], thread [0] executing on device (922)
Task [17], thread [1] executing on host (86)
Task [18], thread [2] executing on device (298)
Task [19], thread [3] executing on device (693)
Task [20], thread [1] executing on device (686)
Task [21], thread [0] executing on device (673)
Task [22], thread [2] executing on device (758)
Task [23], thread [3] executing on device (911)
Task [25], thread [0] executing on device (899)
Task [24], thread [1] executing on device (272)
Task [26], thread [2] executing on device (918)
Task [27], thread [3] executing on host (64)
Task [28], thread [0] executing on device (697)
Task [29], thread [1] executing on device (321)
Task [30], thread [2] executing on device (923)
Task [31], thread [3] executing on device (591)
Task [32], thread [0] executing on device (248)
Task [33], thread [1] executing on device (948)
Task [34], thread [2] executing on device (853)
Task [35], thread [3] executing on device (570)
Task [36], thread [0] executing on device (680)
Task [37], thread [1] executing on device (165)
Task [38], thread [2] executing on device (225)
Task [39], thread [3] executing on device (970)
All Done!

``

paroque28, 2020-02-07:
https://github.com/madisongh/meta-tegra/pull/248

triblex, 2020-02-24:
There seem to be a problem getting access to the CSI camera through the docker containers. Have you tried this? I'm using the Raspberry Pi camera and running the pipeline: 

```gst-launch-1.0 -e nvarguscamerasrc num-buffers=-1 ! 'video/x-raw(memory:NVMM), width=(int)1280, height=(int)720, format=(string)NV12, framerate=(fraction)60/1' ! nvvidconv flip-method=0 ! nvvidconv ! nvegltransform ! udpsink host=$HOST_IP port=$GST_PORT```

The result is this error message:

```
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
=== gst-launch-1.0[65]: Connection established (7FB09771D0)SCF: Error NotSupported: Failed to load EGL library (in src/services/gl/GLService.cpp, function initializeEGLExportFunctions(), line 190)
SCF: Error NotSupported:  (propagating from src/services/gl/GLService.cpp, function initialize(), line 147)
SCF: Error NotSupported:  (propagating from src/services/gl/GLService.cpp, function startService(), line 46)
SCF: Error NotSupported:  (propagating from src/components/ServiceHost.cpp, function startServices(), line 138)
SCF: Error NotSupported:  (propagating from src/api/CameraDriver.cpp, function initialize(), line 168)
SCF: Error InvalidState: Services are already stopped (in src/components/ServiceHost.cpp, function stopServicesInternal(), line 188)
SCF: Error NotSupported:  (propagating from src/api/CameraDriver.cpp, function getCameraDriver(), line 109)
(Argus) Error NotSupported:  (propagating from src/api/GlobalProcessState.cpp, function createCameraProvider(), line 204)
=== gst-launch-1.0[65]: CameraProvider initialized (0x7fac014ec0)Error generated. /dvs/git/dirty/git-master_linux/multimedia/nvgstreamer/gst-nvarguscamera/gstnvarguscamerasrc.cpp, execute:532 No cameras available
Got EOS from element "pipeline0".
Execution ended after 0:00:00.032214229
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...
=== gst-launch-1.0[65]: CameraProvider destroyed (0x7fac014ec0)(Argus) Error EndOfFile: Unexpected error in reading socket (in src/rpc/socket/client/ClientSocketManager.cpp, function recvThreadCore(), line 266)
(Argus) Error EndOfFile: Receive worker failure, notifying 1 waiting threads (in src/rpc/socket/client/ClientSocketManager.cpp, function recvThreadCore(), line 340)
(Argus) Error InvalidState: Argus client is exiting with 1 outstanding client threads (in src/rpc/socket/client/ClientSocketManager.cpp, function recvThreadCore(), line 357)
(Argus) Error EndOfFile: Receiving thread terminated with error (in src/rpc/socket/client/ClientSocketManager.cpp, function recvThreadWrapper(), line 368)
(Argus) Error EndOfFile: Client thread received an error from socket (in src/rpc/socket/client/ClientSocketManager.cpp, function send(), line 145)
(Argus) Error EndOfFile:  (propagating from src/rpc/socket/client/SocketClientDispatch.cpp, function dispatch(), line 87)
```

Using the flag `-e DISPLAY=0` just gives a new error:

`nvbuf_utils: Could not get EGL display connection`

triblex, 2020-02-27:
The above problem with camera access through the docker can be solved with setting the correct arguments when running the container. note the flag `--ipc=host` and mounting the volume  `-v /tmp/argus_socket:/tmp/argus_socket`. The following line worked for me.

`docker run --net=host --runtime nvidia --rm --ipc=host -v /tmp/.X11-unix/:/tmp/.X11-unix/ -v /tmp/argus_socket:/tmp/argus_socket --cap-add SYS_PTRACE -e DISPLAY=$DISPLAY -it nvcr.io/nvidia/l4t-base:r32.3.1`

paroque28, 2020-02-28:
FYI For some reason, only l4t.csv is generated when I use the wip-container-32.3.1 branch. I will be on top of that

wremie, 2020-02-28:
@paroque28 I'm currently evaluating the suggestions made by @triblex in [#248 ](https://github.com/madisongh/meta-tegra/pull/248#issuecomment-590737495). So far I've been able to verify the following;
```
-sym, ${base_libdir}/aarch64-linux-gnu/libv4l2.so.0.0.999999
-sym, ${base_libdir}/aarch64-linux-gnu/libv4lconvert.so.0.0.999999
-sym, ${base_libdir}/aarch64-linux-gnu/libv4l/plugins/nv/libv4l2_nvvidconv.so
-sym, ${base_libdir}/aarch64-linux-gnu/libv4l/plugins/nv/libv4l2_nvvideocodec.so
+sym, ${libdir}/libv4l2.so.0
+sym, ${libdir}/libv4lconvert.so.0
+lib, ${libdir}/libv4l2_nvvidconv.so
+lib, ${libdir}/libv4l2_nvvideocodec.so
```
Although I'm unable to locate `sym, ${libdir}/libdrm_nvdc.so` among others. Not sure yet if this is because I'm missing a package in my image. The best would be to link all the entries of the .csv file against the packages needed in build config. Do you have any thoughts on that?

wremie, 2020-02-28:
@triblex I do agree with you that there are still some bits missing. Running a Docker container with following command (nvargus-daemon is running);
````
docker run -it --rm --net=host --runtime nvidia -v /tmp/argus_socket:/tmp/argus_socket nvcr.io/nvidia/deepstream-l4t:4.0.2-19.12-base
````
volume mapping the `argus_socket` and running the following gsstreamer pipeline;
```
~# gst-launch-1.0 nvarguscamerasrc ! 'video/x-raw(memory:NVMM), format=NV12, width=1920, height=1080' ! nvv4l2h264enc insert-sps-pps=true ! h264parse ! rtph264pay pt=96 ! udpsink host=127.0.0.1 port=8001 sync=false -e

(gst-plugin-scanner:20): GStreamer-WARNING **: 15:16:54.547: Failed to load plugin '/usr/lib/aarch64-linux-gnu/gstreamer-1.0/deepstream/libnvdsgst_dewarper.so': libnppig.so.10.0: cannot open shared object file: No such file or directory

(gst-plugin-scanner:20): GStreamer-WARNING **: 15:16:54.564: Failed to load plugin '/usr/lib/aarch64-linux-gnu/gstreamer-1.0/deepstream/libnvdsgst_infer.so': libnvparsers.so.6: cannot open shared object file: No such file or directory
WARNING: erroneous pipeline: no element "nvarguscamerasrc"
```
still complains about missing libraries. Both are present in host at:
```
/usr/lib/libnvparsers.so.6
/usr/local/cuda-10.0/lib/libnppig.so.10.0
```
But not present in `l4t.csv`. Last but not least it seems that pipeline is not able to communicate with ARGUS api according to missing `nvarguscamerasrc`.

All of the above is not very satisfying but wanted to share before start of the weekend.

paroque28, 2020-02-28:
Oh I see it now, it's because you need to add each package-container-csv to the IMAGE_INSTALL. I wil see if there's an automated way of doing this with yocto

wremie, 2020-02-29:
@paroque28 just stumbled upon the issue created by @triblex in [nvidia-docker](https://github.com/NVIDIA/nvidia-docker/issues/1202#issue-570584835). I think you appreciate the post as I'm experiencing similar issues.

triblex, 2020-03-02:
@wremie 

I found that linking or copying all the files from the path `/usr/lib/gstreamer1.0/` to `/usr/lib/aarch64-linux-gnu/gstreamer1.0/` fixes the missing nvarguscamerasrc issue (along some other possible issues). 

Even if all the same files are present on the host as on the official NVIDIA SDK-Manager Ubuntu image and are correctly listed in the l4t.csv file to be mounted in the docker image, there are still problems. The paths that are changed from the original l4t.csv, needs to be linked or copied to the "correct" places inside the docker image or somehow change where the docker image is looking for them.

That and then gstreamer probably needs to be upgraded to 1.16 in the docker image or downgraded to 1.14 on the host.

triblex, 2020-03-03:
I've downgraded gstreamer to 1.14. And then I've attempted to link all the files to places where the official nvidia docker image assumes they're located.

I ran this pipieline as before (works outside docker container):
```
gst-launch-1.0 -e nvarguscamerasrc num-buffers=-1 ! 'video/x-raw(memory:NVMM), width=(int)1280, height=(int)720, format=(string)NV12, framerate=(fraction)60/1' ! nvvidconv flip-method=0 ! nvvidconv ! nvegltransform ! udpsink host=$HOST_IP port=$GST_PORT
```

I got to this error:

```
Setting pipeline to PAUSED ...
[ 2415.341857] (NULL device *): nvhost_channelctl: invalid cmd 0x80685600
[ 2415.350034] (NULL device *): nvhost_channelctl: invalid cmd 0x80685600
Failed to query video capabilities: Inap[ 2415.357153] (NULL device *): nvhost_channelctl: invalid cmd 0x80685600
propriate ioctl for device
libv4l2: error getting capabilities: Inappropriate ioctl for device
ERROR: Pipeline doesn't want to pause.
ERROR: from element /GstPipeline:pipeline0/nvv4l2h264enc:nvv4l2h264enc0: Error getting capabilities for device '/dev/nvhost-msenc': It isn't a v4l2 driver. Check if it is a v4l1 driver.
Additional debug info:
v4l2_calls.c(98): gst_v4l2_get_capabilities (): /GstPipeline:pipeline0/nvv4l2h264enc:nvv4l2h264enc0:
system error: Inappropriate ioctl for device
Setting pipeline to NULL ...
Freeing pipeline ...
```

wremie, 2020-03-03:
@triblex thanks for the work. If i remember correctly you mount the device node as `-v /dev/video0:/dev/video0 `. 

However, according to [deepstream-l4t](https://ngc.nvidia.com/catalog/containers/nvidia:deepstream-l4t) CSI should be mounted as `-v /tmp/argus_socket:/tmp/argus_socket`

Maybe this is worth trying?

triblex, 2020-03-03:
@wremie I've actually tried running the docker with both. I am using this docker run command:

`docker run --net=host --runtime nvidia --rm --ipc=host -v /tmp/.X11-unix/:/tmp/.X11-unix/ -v /tmp/argus_socket:/tmp/argus_socket --device=/dev/video0:/dev/video0 --cap-add SYS_PTRACE -e DISPLAY=$DISPLAY -it nvcr.io/nvidia/l4t-base:r32.3.1`

closed 2020-03-21 by madisongh