OE4T/meta-tegra #232: how to build u-boot-bup-payload?

danielwangksu, 2020-01-14:
Hi there, I used to be able to build bup-payload in U-boot work directory, but it has gone. I tried to add `u-boot-payload` into `IMAGE_INSTALL` under core-image-minimal.bb, but Yocto reports "do_populate_sdk error". What is the correct way to enable the bup-payload build?

madisongh, 2020-01-14:
See [this recipe](https://github.com/madisongh/test-distro/blob/master/layers/meta-testdistro/recipes-bsp/tegra-binaries/tegra-bup-payload_1.0.bb) for an example.

danielwangksu, 2020-01-14:
Thank you @madisongh  should I add u-boot-bup-payload as as RDEPENDS of core-image-minimal image?

madisongh, 2020-01-14:
You can't directly add u-boot-bup-payload, since it doesn't install anything in the rootfs. You have to create another recipe (like the one I linked to) to install the deployed payload into the rootfs, and add that recipe/package to your image.

danielwangksu, 2020-01-14:
Got it thanks you so much!

closed 2020-01-14 by danielwangksu