OE4T/meta-tegra #233: Compilation with cuda Unkown __WORDSIZE

Austriker, 2020-01-15:
Hi,

I am having an issue while compiling my app with the meta-tegra for a TX2.

Here is the config :
```sh
Parsing of 1679 .bb files complete (1678 cached, 1 parsed). 2407 targets, 110 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies
NOTE: preferred version 10.0% of cuda-binaries-native not available (for item cuda-binaries-native)
NOTE: versions of cuda-binaries-native available: 6.5-53

Build Configuration:
BB_VERSION           = "1.42.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-poky-linux"
MACHINE              = "jetson-tx2"
DISTRO               = "poky"
DISTRO_VERSION       = "2.7.2"
TUNE_FEATURES        = "aarch64 armv8a crc"
TARGET_FPU           = ""
meta                 
meta-poky            
meta-yocto-bsp       = "warrior:d0f73121551dc98f6924cd77952bf9ebf5ef3dd7"
meta-tegra           
contrib              = "warrior-l4t-r32.2:bd8023e171b67a437443b8ef03f76aa6315be3ad"
meta-mender-core     = "warrior:9650b37762443f01cf694deb1e00417751a2a719"
meta-mender-tegra    = "warrior:e51eaf11461755339a42431216339184b03fd894"
meta-qt5             = "HEAD:e8ee8bc088936129ca0d1335fc09c4e9e2a4aaa6"
meta-tegra-app  = "tegra-app-warrior:db94ee1e450307012fe33377dccabef7e4fba302"
meta-oe              = "warrior:a24acf94d48d635eca668ea34598c6e5c857e3f8"

Initialising tasks: 100% |#######################################| Time: 0:00:06
```

Here is the error : 
```sh
| /home/ubuntu/tegra/build/tmp/work/armv8a_tegra186-poky-linux/cudaapp/4.3.2-r0/recipe-sysroot/usr/include/bits/endian.h:46:2: error: #error "Unknown __WORDSIZE detected"
|  #error "Unknown __WORDSIZE detected"
|   ^~~~~
| In file included from /home/ubuntu/tegra/build/tmp/work/armv8a_tegra186-poky-linux/cudaapp/4.3.2-r0/recipe-sysroot/usr/include/sys/types.h:244:0,
|                  from /home/ubuntu/tegra/build/tmp/work/armv8a_tegra186-poky-linux/cudaapp/4.3.2-r0/recipe-sysroot/usr/include/stdlib.h:394,
|                  from /usr/include/c++/7/cstdlib:75,
|                  from /usr/include/c++/7/stdlib.h:36,
|                  from /home/ubuntu/tegra/build/tmp/work/armv8a_tegra186-poky-linux/cudaapp/4.3.2-r0/recipe-sysroot/usr/local/cuda-10.0/include/crt/common_functions.h:129,
|                  from /home/ubuntu/tegra/build/tmp/work/armv8a_tegra186-poky-linux/cudaapp/4.3.2-r0/recipe-sysroot/usr/local/cuda-10.0/include/cuda_runtime.h:120,
|                  from <command-line>:0:
| /home/ubuntu/tegra/build/tmp/work/armv8a_tegra186-poky-linux/cudaapp/4.3.2-r0/recipe-sysroot/usr/include/bits/pthreadtypes.h:23:2: error: #error "__WORDSIZE is not defined"
|  #error "__WORDSIZE is not defined"
|   ^~~~~
| /home/ubuntu/tegra/build/tmp/work/armv8a_tegra186-poky-linux/cudaapp/4.3.2-r0/recipe-sysroot/usr/include/bits/pthreadtypes.h:46:2: error: #error "Unknown __WORDSIZE detected"
|  #error "Unknown __WORDSIZE detected"
|   ^~~~~
| In file included from /home/ubuntu/tegra/build/tmp/work/armv8a_tegra186-poky-linux/cudaapp/4.3.2-r0/recipe-sysroot/usr/include/math.h:233:0,
|                  from /usr/include/c++/7/cmath:45,
|                  from /usr/include/c++/7/math.h:36,
|                  from /home/ubuntu/tegra/build/tmp/work/armv8a_tegra186-poky-linux/cudaapp/4.3.2-r0/recipe-sysroot/usr/local/cuda-10.0/include/crt/math_functions.h:8859,
|                  from /home/ubuntu/tegra/build/tmp/work/armv8a_tegra186-poky-linux/cudaapp/4.3.2-r0/recipe-sysroot/usr/local/cuda-10.0/include/crt/common_functions.h:267,
|                  from /home/ubuntu/tegra/build/tmp/work/armv8a_tegra186-poky-linux/cudaapp/4.3.2-r0/recipe-sysroot/usr/local/cuda-10.0/include/cuda_runtime.h:120,
|                  from <command-line>:0:
| /home/ubuntu/tegra/build/tmp/work/armv8a_tegra186-poky-linux/cudaapp/4.3.2-r0/recipe-sysroot/usr/include/bits/fp-fast.h:23:2: error: #error "__WORDSIZE is not defined"
|  #error "__WORDSIZE is not defined"
|   ^~~~~
| /home/ubuntu/tegra/build/tmp/work/armv8a_tegra186-poky-linux/cudaapp/4.3.2-r0/recipe-sysroot/usr/include/bits/fp-fast.h:46:2: error: #error "Unknown __WORDSIZE detected"
|  #error "Unknown __WORDSIZE detected"
|   ^~~~~
```

I just upgraded my stack from Thud to Warrior.

I am struggling to find where the word size is defined.

Best regards,

madisongh, 2020-01-15:
It should get defined in the header `<bits/wordsize.h>`, but it looks like something's not right with your setup.  These messages, for instance:
```
NOTE: preferred version 10.0% of cuda-binaries-native not available (for item cuda-binaries-native)
NOTE: versions of cuda-binaries-native available: 6.5-53
```
Did you use the NVIDIA SDK manager to download the JetPack 4.2.1 content, and set NVIDIA_DEVNET_MIRROR to point to it, as mentioned in the [README](https://github.com/madisongh/meta-tegra/blob/warrior-l4t-r32.2/README)?  The CUDA binaries packages downloaded should be version 10.0.326-1 for JetPack 4.2.1.

I'm also seeing host-side `/usr/include` paths in the error messages, which could indicate a problem with how the C++ compiler is being invoked - missing `--sysroot` or `-I` options, perhaps.

Austriker, 2020-01-15:
Hi,

Yes I used the sdk manager and I have this in the folder : 
```sh
ls -la
total 5790420
drwxrwxr-x  2 ubuntu ubuntu       4096 nov.   3 16:05 .
drwxr-xr-x 18 ubuntu ubuntu       4096 janv. 15 11:34 ..
-rw-rw-r--  1 ubuntu ubuntu  261275024 nov.   3 16:01 2019_06_23_2352-26660552-NVIDIA_Nomad_2019.2.19174.2352_Release_External-L4T.linux.deb
-rw-rw-r--  1 ubuntu ubuntu         46 nov.   3 15:58 cuda.csv
-rw-rw-r--  1 ubuntu ubuntu  394475774 nov.   3 16:02 cuda-repo-cross-aarch64-10-0-local-10.0.326_1.0-1_all.deb
-rw-rw-r--  1 ubuntu ubuntu 1000333292 nov.   3 16:04 cuda-repo-l4t-10-0-local-10.0.326_1.0-1_arm64.deb
-rw-rw-r--  1 ubuntu ubuntu 1582667320 nov.   3 16:05 cuda-repo-ubuntu1804-10-0-local-10.0.326-410.108_1.0-1_amd64.deb
-rw-rw-r--  1 ubuntu ubuntu         96 nov.   3 15:58 cudnn.csv
-rw-rw-r--  1 ubuntu ubuntu   25538902 nov.   3 15:59 devtools_docs.zip
-rw-rw-r--  1 ubuntu ubuntu      16106 nov.   3 15:58 graphsurgeon-tf_5.1.6-1+cuda10.0_arm64.deb
-rw-rw-r--  1 ubuntu ubuntu  130960016 nov.   3 15:58 Jetson_Linux_R32.2.0_aarch64.tbz2
-rw-rw-r--  1 ubuntu ubuntu  159174288 nov.   3 15:58 libcudnn7_7.5.0.56-1+cuda10.0_arm64.deb
-rw-rw-r--  1 ubuntu ubuntu  148029812 nov.   3 15:58 libcudnn7-dev_7.5.0.56-1+cuda10.0_arm64.deb
-rw-rw-r--  1 ubuntu ubuntu    5983100 nov.   3 15:57 libcudnn7-doc_7.5.0.56-1+cuda10.0_arm64.deb
-rw-rw-r--  1 ubuntu ubuntu      42744 nov.   3 15:58 libnvidia-container0_0.9.0~beta.1_arm64.deb
-rw-rw-r--  1 ubuntu ubuntu      12596 nov.   3 15:58 libnvidia-container-tools_0.9.0~beta.1_arm64.deb
-rw-rw-r--  1 ubuntu ubuntu   37725708 nov.   3 15:57 libnvinfer5_5.1.6-1+cuda10.0_arm64.deb
-rw-rw-r--  1 ubuntu ubuntu   39350886 nov.   3 15:58 libnvinfer-dev_5.1.6-1+cuda10.0_arm64.deb
-rw-rw-r--  1 ubuntu ubuntu  458082356 nov.   3 16:01 libnvinfer-samples_5.1.6-1+cuda10.0_all.deb
-rw-rw-r--  1 ubuntu ubuntu    8882498 nov.   3 15:58 libopencv_3.3.1-2-g31ccdfe11_arm64.deb
-rw-rw-r--  1 ubuntu ubuntu   27931360 nov.   3 15:58 libopencv_3.3.1-2-gb3f86dcd5_amd64.deb
-rw-rw-r--  1 ubuntu ubuntu    1103544 nov.   3 15:58 libopencv-dev_3.3.1-2-g31ccdfe11_arm64.deb
-rw-rw-r--  1 ubuntu ubuntu    1118974 nov.   3 15:58 libopencv-dev_3.3.1-2-gb3f86dcd5_amd64.deb
-rw-rw-r--  1 ubuntu ubuntu    1559580 nov.   3 15:58 libopencv-python_3.3.1-2-g31ccdfe11_arm64.deb
-rw-rw-r--  1 ubuntu ubuntu    1653492 nov.   3 15:58 libopencv-python_3.3.1-2-gb3f86dcd5_amd64.deb
-rw-rw-r--  1 ubuntu ubuntu     185418 nov.   3 15:58 libopencv-samples_3.3.1-2-g31ccdfe11_arm64.deb
-rw-rw-r--  1 ubuntu ubuntu     184036 nov.   3 15:58 libopencv-samples_3.3.1-2-gb3f86dcd5_amd64.deb
-rw-rw-r--  1 ubuntu ubuntu   66561472 nov.   3 16:01 libvisionworks-repo_1.6.0.500n_amd64.deb
-rw-rw-r--  1 ubuntu ubuntu   60780704 nov.   3 16:02 libvisionworks-repo_1.6.0.500n_arm64.deb
-rw-rw-r--  1 ubuntu ubuntu   20597586 nov.   3 16:02 libvisionworks-sfm-repo_0.90.4_amd64.deb
-rw-rw-r--  1 ubuntu ubuntu   20216850 nov.   3 16:02 libvisionworks-sfm-repo_0.90.4_arm64.deb
-rw-rw-r--  1 ubuntu ubuntu   13113238 nov.   3 16:02 libvisionworks-tracking-repo_0.88.2_amd64.deb
-rw-rw-r--  1 ubuntu ubuntu   13107966 nov.   3 16:02 libvisionworks-tracking-repo_0.88.2_arm64.deb
-rw-rw-r--  1 ubuntu ubuntu  112960388 nov.   3 16:03 NsightSystems-linux-public-2019.4.1.10-a76094a.deb
-rw-rw-r--  1 ubuntu ubuntu    1564964 nov.   3 15:58 nvidia-container-runtime_0.9.0_beta.1+docker18.09.2-1_arm64.deb
-rw-rw-r--  1 ubuntu ubuntu     501188 nov.   3 15:58 nvidia-container-runtime-hook_0.9.0_beta.1-1_arm64.deb
-rw-rw-r--  1 ubuntu ubuntu       2876 nov.   3 15:58 nvidia-docker2_0.9.0_beta.1+docker18.09.2-1_arm64.deb
-rw-rw-r--  1 ubuntu ubuntu    5350765 nov.   3 16:02 NVIDIA_VisionWorks_References.zip
-rw-rw-r--  1 ubuntu ubuntu     309810 nov.   3 15:58 python3-libnvinfer_5.1.6-1+cuda10.0_arm64.deb
-rw-rw-r--  1 ubuntu ubuntu       3624 nov.   3 15:58 python3-libnvinfer-dev_5.1.6-1+cuda10.0_arm64.deb
-rw-rw-r--  1 ubuntu ubuntu     312310 nov.   3 15:58 python-libnvinfer_5.1.6-1+cuda10.0_arm64.deb
-rw-rw-r--  1 ubuntu ubuntu       3616 nov.   3 15:58 python-libnvinfer-dev_5.1.6-1+cuda10.0_arm64.deb
-rw-rw-r--  1 ubuntu ubuntu     211967 nov.   3 16:05 sdkml3_jetpack_l4t_421_b107_rev2.json
-rw-rw-r--  1 ubuntu ubuntu 1255371587 nov.   3 16:05 Tegra_Linux_Sample-Root-Filesystem_R32.2.0_aarch64.tbz2
-rw-rw-r--  1 ubuntu ubuntu   70301782 nov.   3 16:03 Tegra_Multimedia_API_R32.2.0_aarch64.tbz2
-rw-rw-r--  1 ubuntu ubuntu    1636904 nov.   3 15:58 tensorrt_5.1.6.1-1+cuda10.0_arm64.deb
-rw-rw-r--  1 ubuntu ubuntu        691 nov.   3 15:58 tensorrt.csv
-rw-rw-r--  1 ubuntu ubuntu      37554 nov.   3 15:58 uff-converter-tf_5.1.6-1+cuda10.0_arm64.deb
-rw-rw-r--  1 ubuntu ubuntu        287 nov.   3 15:58 visionworks.csv
```
I always had this warning but I don't understand how to get rid of it.

And here is the compile line : 
```sh
| make[1]: Leaving directory '/home/ubuntu/tegra/build/tmp/work/armv8a_tegra186-poky-linux/app/4.3.2-r0/build/src/plugins/plugin_inference_gt'
| cd src/plugins/plugin_inference_cuda/ && ( test -e Makefile || /home/ubuntu/tegra/build/tmp/work/armv8a_tegra186-poky-linux/app/4.3.2-r0/recipe-sysroot-native/usr/bin/qmake -o Makefile /home/ubuntu/tegra/build/tmp/work/armv8a_tegra186-poky-linux/app/4.3.2-r0/git/src/plugins/plugin_inference_cuda/plugin_inference_cuda.pro CUDA_ARCH=sm_62 CONFIG+=release CONFIG+=agrifac EXTRA_NVCCFLAGS+=-I/home/ubuntu/tegra/build/tmp/work/armv8a_tegra186-poky-linux/app/4.3.2-r0/recipe-sysroo [... cut ...]
|                  from /home/ubuntu/tegra/build/tmp/work/armv8a_tegra186-poky-linux/app/4.3.2-r0/recipe-sysroot/usr/local/cuda-10.0/include/cuda_runtime.h:83,
|                  from <command-line>:0:
| /home/ubuntu/tegra/build/tmp/work/armv8a_tegra186-poky-linux/app/4.3.2-r0/recipe-sysroot/usr/include/bits/wordsize.h:23:2: error: #error "__WORDSIZE is not defined"
|  #error "__WORDSIZE is not defined"
|   ^~~~~
| /home/ubuntu/tegra/build/tmp/work/armv8a_tegra186-poky-linux/app/4.3.2-r0/recipe-sysroot/usr/include/bits/wordsize.h:46:2: error: #error "Unknown __WORDSIZE detected"
|  #error "Unknown __WORDSIZE detected"
|   ^~~~~
```

madisongh, 2020-01-15:
>I always had this warning but I don't understand how to get rid of it.

Oh yeah, I see it, too, now that I'm looking... it isn't a real issue, just annoying.  I'll fix it up when I get a chance.

What I do see in the compile line, though, is that `/usr/local/cuda/bin/nvcc` is being run, rather than the `nvcc` in your build workspace.  I also don't see a `-ccbin` option being passed on the command line to tell it to use `aarch64-poky-linux-gcc` as the C++ compiler.  That's probably why `bits.h` is generating the errors... if you look in that file, it only handles ARM and MIPS cases, and  `nvcc` is probably using your host's native C++ compiler.

I don't know anything about `qmake`, unfortunately, but perhaps there's something in that tool, or the build file it's generating the makefile from, that needs a change, so it generates the right commands for cross-compiling?

Austriker, 2020-01-16:
Ok I see. I will try to fix. Do you have a sample project for compiling a cuda app with make or Cmake in Yocto ? It will help me a lot.

With qmake you can do the almost the same as Cmake. But I am struggling with which info I should pass to Qmake.

madisongh, 2020-01-16:
I have a very simple one that I build with cmake on a regular basis.  The CMakeLists.txt file is [here](https://github.com/madisongh/gpu-burn/blob/master/CMakeLists.txt), and the recipe is [here](https://github.com/madisongh/test-distro/blob/master/layers/meta-testdistro/recipes-test/tegra-tests/gpu-burn_git.bb).

>Regarding the warning I can gladly do a PR but on which branch should I do it ? master ?

Thanks, that would be great.  Yes, master first would be best, if you can do that.

closed 2020-02-17 by Austriker