OE4T/meta-tegra #236: Weston fails to load tegra-udrm

build2c, 2020-01-18:
I am building a Weston based distro using Zeus and L4T 32.2.3.  I have been trying to get an HDMI display working. It displays boot logs fine but never loads the Wayland desktop.

In the logs, there is an error about failing to load tegra-udrm. The log says that it tried searching /usr/lib/dri for tegra-udrm. I have confirmed that nothing in /usr/lib/dri contains this name.

Any suggestions on what might be missing and why?

madisongh, 2020-01-19:
I think it was working with earlier 32.2.x releases, but I just confirmed that tegra_udrm does interfere with weston on 32.2.3, so I've removed it from the TX2 machine configuration for now.

build2c, 2020-01-19:
> You received this message because you are subscribed to the Google Groups "IT Services" group.
> To unsubscribe from this group and stop receiving emails from it, send an email to it-services+unsubscribe@build2c.org.
> To view this discussion on the web visit https://groups.google.com/a/build2c.org/d/msgid/it-services/madisongh/meta-tegra/issues/236/576014060%40github.com.

I am not an expert on DRM. What is the impact of not having this?

madisongh, 2020-01-19:
In terms of getting wayland/weston to work, there are already patches in place to deal with the graphics initialization on the Tegras and the lack of a full DRM emulation there.   The udrm driver was just getting in the way.

kekiefer, 2020-01-19:
I can say that the mechanisms to use the system libdrm (falling back from libdrm_nvdc) are working in general, as I'm still successful in using a custom drm driver. I'll take a look at tegra-udrm tomorrow and see if anything is apparent there.

build2c, 2020-01-19:
> To unsubscribe from this group and stop receiving emails from it, send an email to it-services+unsubscribe@build2c.org <mailto:it-services+unsubscribe@build2c.org>.

Any chance you can provide a diff for removing it?

madisongh, 2020-01-20:
It's [this commit](https://github.com/madisongh/meta-tegra/commit/6d79639f701d5cfcf0d9dc4c33539aea2b1b9055) that I pushed to the zeus branch.

kekiefer, 2020-01-20:
It looks like I won't be able to add much except to say that it is also my recollection that it was working on 32.2.x at some point.

 I don't have a 32.2.3 build handy to try, but at least I can confirm that on 32.3.1 this is working ok

BTW, there are no longer differences between platforms here -- the driver is shipped by default with jetson-xavier. It might make sense to unify the modprobe.d configurations between tx2 and xavier. Also, I'd have to go back and look to be sure, but I think some patches to weston can be dropped if tegra-udrm is present.

Without tegra-udrm:

```
root@jetson-xavier:~# cat /var/run/user/0/weston.log                                                        
Date: 2020-01-20 UTC
[17:48:03.745] weston 7.0.0
               https://wayland.freedesktop.org
               Bug reports to: https://gitlab.freedesktop.org/wayland/weston/issues/
               Build: yocto-3.0.1-11-ga34a5fb3eb
[17:48:03.745] Command line: /usr/bin/weston --log=/run/user/0/weston.log --use-egldevice
[17:48:03.745] OS: Linux, 4.9.140-l4t-r32.3.1+g47e7e1cb0b49, #1 SMP PREEMPT Mon Jan 20 17:10:44 UTC 2020, aarch4
[17:48:03.748] Using config file '/etc/xdg/weston/weston.ini'
[17:48:03.751] Output repaint window is 7 ms maximum.
[17:48:03.752] Loading module '/usr/lib/libweston-7/drm-backend.so'
[17:48:03.769] initializing drm backend
[17:48:03.778] logind: session control granted
[17:48:03.780] using drm-nvdc
...
```

With tegra-udrm:

```
root@jetson-xavier:~# cat /var/run/user/0/weston.log                                                        
Date: 2020-01-20 UTC
[17:46:46.296] weston 7.0.0
               https://wayland.freedesktop.org
               Bug reports to: https://gitlab.freedesktop.org/wayland/weston/issues/
               Build: yocto-3.0.1-11-ga34a5fb3eb
[17:46:46.296] Command line: /usr/bin/weston --log=/run/user/0/weston.log --use-egldevice
[17:46:46.297] OS: Linux, 4.9.140-l4t-r32.3.1+g47e7e1cb0b49, #1 SMP PREEMPT Mon Jan 20 17:10:44 UTC 2020, aarch4
[17:46:46.300] Using config file '/etc/xdg/weston/weston.ini'
[17:46:46.303] Output repaint window is 7 ms maximum.
[17:46:46.304] Loading module '/usr/lib/libweston-7/drm-backend.so'
[17:46:46.322] initializing drm backend
[17:46:46.332] logind: session control granted
[17:46:46.336] using /dev/dri/card0
...
```

kekiefer, 2020-01-20:
Actually, here's one thing about tegra-udrm that could be a problem. The qt5 eglfs_kms_egldevice backend is broken against tegra-udrm, at least in QT 5.12.5 (from warrior). I'm going to try to upgrade to zeus now (I've been putting this off since I had problems with my application and the newere qt version) and see what's up there.

kekiefer, 2020-01-20:
tegra-udrm doesn't work with eglfs_kms_egldevice on zeus either. What's more, it looks like tegra-udrm is still not loaded by default in Jetpack 4.3 on Xavier, and when I do load it, it breaks all nvidia's egldevice graphics_demos there.

I guess the question I have would be, is including tegra-udrm for TX2 machines breaking their ability to run these examples or qt? Other than that, I won't advocate enabling tegra-udrm on Xavier just yet.

madisongh, 2020-01-20:
Bummer.  I was hoping it would be more ready for prime time in 32.3.1, and that it would help eliminate the need for hacking weston as much as I've had to.  Sounds like it might not be.

build2c, 2020-01-22:
A possibly related issue that I'm seeing after making these changes. I can now get the desktop showing after boot, but many programs cannot use the display (including gst-launch-1.0 and argus_camera). These are reporting:

nvbuf_utils: Could not get EGL display connection

Any ideas?

madisongh, 2020-01-22:
Did you unset the `DISPLAY` environment variable before trying the programs?  If you don't, most programs will try to use X, and that won't work.

build2c, 2020-01-22:
Will try. Never occurred to me as every display related search result says to set the variable. :)

build2c, 2020-01-23:
Hmm. It now says it is failing to get EGL configs. Does my distro need to add anything special?

madisongh, 2020-01-23:
Not that I know of.  I build `core-image-weston` for my testing, then have some scripts for setting up gstreamer capture and display pipelines that more or less follow the examples that NV gives in their Accelerated Gstreamer documentation (which I guess is now included as a chapter in the L4T BSP documentation, rather than a separate doc).  The recipe for the tests and scripts I use are [here](https://github.com/madisongh/test-distro/tree/zeus/layers/meta-testdistro/recipes-test/tegra-tests).

madisongh, 2020-01-25:
@kekiefer I've verified that weston works fine on all platforms with tegra_udrm in 32.3.1, and I've been able to drop the extra patches I had to add for bypassing weston's normal DRM/KMS calls.  I have pushed changes to master to enable tegra_udrm modeset by default.

kekiefer, 2020-01-25:
I was also able to get Weston working with tegra-udrm. It looks like the trouble is with any client that uses the `drmOpen` helper function. Calls to `drmOpen`, either when asking explicitly for tegra-udrm or implicitly as _the default egldevice whenever the module is loaded_, will end up handled by libdrm_nvdc, and that is what is broken now (it worked previously). The fact that tegra-udrm becomes the default is the rub.

It just so happens that Weston opens the DRM device without using this helper and that is what makes it work -- but I don't think this is too common. So it's not so much that tegra-udrm is broken but that libdrm_nvdc is broken when tegra-udrm is loaded.

Programs that use `drmOpen` include QT's eglfs_kms_egldevice backend and NVIDIA's test programs from the jetpack release.

For testing QT, a decent smoke test is to include meta-qt5 and add the qt5-opengles2-test recipe to your image. To select the right backend, set the environment `QT_QPA_EGLFS_INTEGRATION=eglfs_kms_egldevice` and `QT_QPA_PLATFORM=eglfs`.

NVIDIA's test programs are documented here:
https://docs.nvidia.com/jetson/l4t/index.html#page/Tegra%2520Linux%2520Driver%2520Package%2520Development%2520Guide%2Fopengl_egl_test_app.html%23

The sources for these examples are in the tegra-binaries tarball. It might be worth putting together a recipe to build these.

madisongh, 2020-01-25:
OK, thanks for the pointers.  Maybe I can move the modprobe config for tegra_udrm to another recipe and have weston RDEPEND on that, so it gets loaded only when weston is in the mix.

kekiefer, 2020-01-25:
I think that sounds like a good plan!

madisongh, 2020-01-26:
Done.  I also added a recipe for building the graphics demos and can see the egldevice flavor of each demo failing to open the DRM device when tegra_udrm is loaded.

closed 2020-03-04 by madisongh