OE4T/meta-tegra #237: Wrong checksum on master Branch

paroque28, 2020-01-20:
ERROR: cboot-32.2.3-r0 do_fetch: Fetcher failure for URL: 'https://developer.download.nvidia.com/embedded/L4T/r32-3-1_Release_v1.0/Sources/T210/public_sources.tbz2;downloadfilename=Tegra210-public_sources-32.2.3.tbz2'. Checksum mismatch!
File: '/wr/sandbox/prodrigu/projects/lts17_jetson_nano/build/downloads/Tegra210-public_sources-32.2.3.tbz2' has md5 checksum 778c2f4a7241f0859e6f71122cbe5fca when 96e592ef7f0bb94b900b98d6bf014872 was expected
File: '/wr/sandbox/prodrigu/projects/lts17_jetson_nano/build/downloads/Tegra210-public_sources-32.2.3.tbz2' has sha256 checksum 201f8f55b6e02421d463b5dd17861e02027fe8a3939ee2a02fb2720f8e4446fd when 676b4c04ad2cca0ed691215b53f8528bd14d09b47c73528a67fd63068369a67b was expected
If this change is expected (e.g. you have upgraded to a new version without updating the checksums) then you can use these lines within the recipe:
SRC_URI[md5sum] = "778c2f4a7241f0859e6f71122cbe5fca"
SRC_URI[sha256sum] = "201f8f55b6e02421d463b5dd17861e02027fe8a3939ee2a02fb2720f8e4446fd"
Otherwise you should retry the download and/or check with upstream to determine if the file has become corrupted or otherwise unexpectedly modified.

ERROR: cboot-32.2.3-r0 do_fetch: Fetcher failure for URL: 'https://developer.download.nvidia.com/embedded/L4T/r32-3-1_Release_v1.0/Sources/T210/public_sources.tbz2;downloadfilename=Tegra210-public_sources-32.2.3.tbz2'. Unable to fetch URL from any source.
ERROR: cboot-32.2.3-r0 do_fetch: Function failed: base_do_fetch
ERROR: Logfile of failure stored in: /wr/sandbox/prodrigu/projects/lts17_jetson_nano/build/tmp-glibc/work/jetson_nano_qspi_sd-wrs-linux/cboot/32.2.3-r0/temp/log.do_fetch.100348
ERROR: Task (/wr/sandbox/prodrigu/projects/lts17_jetson_nano/layers/meta-tegra/recipes-bsp/cboot/cboot_32.2.3.bb:do_fetch) failed with exit code '1'

madisongh, 2020-01-20:
That recipe is only for Xavier builds - NV has only provided source for cboot for that platform.  I've just pushed a fix to make sure it's not visible when building for other platforms.

paroque28, 2020-01-20:
Thank you, Madison :)

closed 2020-01-20 by paroque28